esphome/issues #6507: Inkplate 5V2 screen is flipped on X axis

Leicas, 2024-11-30:
### The problem

When running a sample, I have a consistent Mirror/Flip on the X axis (and could not find a way to fix it).

I confirmed with sample code on arduino that it’s only hapenning with home assistant.

### Which version of ESPHome has the issue?

2024.11.2

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Board

inkplate

### Component causing the issue

inkplate6

### Example YAML snippet

```yaml
esphome:
  name: eink-plate
  friendly_name: eink_plate

esp32:
  board: esp-wrover-kit
  framework:
    type: arduino

# Example minimal configuration entry
pca6416a:
  - id: pca6416a_hub
    address: 0x20

switch:
  - platform: template
    name: "Inkplate Greyscale mode"
    lambda: return id(inkplate_display).get_greyscale();
    turn_on_action:
      - lambda: id(inkplate_display).set_greyscale(true);
    turn_off_action:
      - lambda: id(inkplate_display).set_greyscale(false);

  - platform: template
    name: "Inkplate Partial Updating"
    lambda: return id(inkplate_display).get_partial_updating();
    turn_on_action:
      - lambda: id(inkplate_display).set_partial_updating(true);
    turn_off_action:
      - lambda: id(inkplate_display).set_partial_updating(false);

  - platform: gpio
    id: battery_read_mosfet
    pin:
      pca6416a: pca6416a_hub
      number: 9

sensor:
  - platform: adc
    id: battery_voltage
    update_interval: never
    attenuation: 12db
    pin: 35
  - platform: template
    name: "Inkplate Battery Voltage"
    lambda: |-
      id(battery_read_mosfet).turn_on();
      delay(5);
      float adc = id(battery_voltage).sample();
      id(battery_read_mosfet).turn_off();
      return adc;
    filters:
      - multiply: 2
    unit_of_measurement: 'V'

i2c:

time:
  - platform: sntp
    id: esptime

font:
  - file: "gfonts://Roboto"
    id: helvetica_96
    size: 96
  - file: "gfonts://Roboto"
    id: helvetica_48
    size: 48

binary_sensor:
  - platform: status
    name: "Inkplate Status"
    id: system_status

http_request:
  verify_ssl: False

display:
- platform: inkplate6
  id: inkplate_display
  greyscale: false
  partial_updating: true
  update_interval: 60s
  model: inkplate_5_v2
  full_update_every: 20
  rotation: 180

  ckv_pin: 32
  sph_pin: 33
  gmod_pin:
    pca6416a: pca6416a_hub
    number: 1
  gpio0_enable_pin:
    pca6416a: pca6416a_hub
    number: 8
  oe_pin:
    pca6416a: pca6416a_hub
    number: 0
  spv_pin:
    pca6416a: pca6416a_hub
    number: 2
  powerup_pin:
    pca6416a: pca6416a_hub
    number: 4
  wakeup_pin:
    pca6416a: pca6416a_hub
    number: 3
  vcom_pin:
    pca6416a: pca6416a_hub
    number: 5
  lambda: |-
    it.fill(COLOR_ON);

    it.print(100, 100, id(helvetica_48), COLOR_OFF, TextAlign::TOP_LEFT, "ESPHome");

    it.strftime(400, 300, id(helvetica_48), COLOR_OFF, TextAlign::CENTER, "%Y-%m-%d", id(esptime).now());
    it.strftime(400, 400, id(helvetica_96), COLOR_OFF, TextAlign::CENTER, "%H:%M", id(esptime).now());

    if (id(system_status).state) {
      it.print(700, 100, id(helvetica_48), COLOR_OFF, TextAlign::TOP_RIGHT, "Online");
    } else {
      it.print(700, 100, id(helvetica_48), COLOR_OFF, TextAlign::TOP_RIGHT, "Offline");
    }

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: ""

ota:
  - platform: esphome
    password: ""

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Eink-Plate Fallback Hotspot"
    password: ""

captive_portal:
```

iamfrommars81, 2025-02-28:
Experiencing the same issue on the same device.

Leicas, 2025-03-01:
Fully solved with
https://github.com/esphome/esphome/pull/7904#issuecomment-2591163614

closed 2025-05-29 by jesserockz