esphome/issues #6509: LVGL error during compilation

maurofain, 2024-12-02:
### The problem

I get a blocking error after upgrade to 2024.11.1 
src/esphome/components/lvgl/lvgl_esphome.h:231:27: error: 'image_type_to_width_stride' is not a member of 'esphome::image'
It seems that in lvgl image.h the function [ inline int image_type_to_width_stride(int width, ImageType type) { return (width * image_type_to_bpp(type) + 7u) / 8u; } ] is not available. 

### Which version of ESPHome has the issue?

> 2024.11.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.11.2

### What platform are you using?

ESP32-IDF

### Board

sunton-esp32-8048s070

### Component causing the issue

lvgl

### Example YAML snippet

```yaml
Compiling the yaml there are nonerrors emerged in it but complitantion stops with this message:
src/esphome/components/lvgl/lvgl_esphome.h:231:27: error: 'image_type_to_width_stride' is not a member of 'esphome::image'

No more messages
```

maurofain, 2024-12-02:
Compilation is OK rolling back esphome to 2024.10.3

clydebarrow, 2024-12-02:
Can't help without seeing your yaml, but if you are using `external_components` that is quite possibly your problem.

ESP32andmore, 2024-12-03:
Clive, Same issue here. Compiled fine up to 2024.11.1 using your external component source: github://clydebarrow/esphome@lvgl_original
My Yaml is here: https://1drv.ms/u/c/e8a42f33d34bf114/EcGAi4YLyfpGh2VsR7L3TeMBSJ0hg6KnLUVeT04p1lXIww?e=cEy9vm if you would be kind enough to look at it.  Got the esphome release compiling without this error after dome minor naming mods but issue with the .bin booting. Thanks

clydebarrow, 2024-12-03:
> using your external component source

That's your problem - don't use that. It's merged into ESPHome release now (well, months ago actually.)

closed 2024-12-03 by clydebarrow

ESP32andmore, 2024-12-03:
Clive, like I said I did attempt to use the merged with some slight mods for naming changes in the merged release. There was no reason to move over to the merged until now.  That executable failed entirely:
[23:18:47]ESP-ROM:esp32s3-20210327
[23:18:47]Build:Mar 27 2021
[23:18:47]rst:0x15 (USB_UART_CHIP_RESET),boot:0x8 (SPI_FAST_FLASH_BOOT)
[23:18:47]Saved PC:0x40378c15
[23:18:47]SPIWP:0xee
[23:18:47]mode:DIO, clock div:1
[23:18:47]load:0x3fce3808,len:0x43c
[23:18:47]load:0x403c9700,len:0xbec
[23:18:47]load:0x403cc700,len:0x2a3c
[23:18:47]entry 0x403c98d8
[23:18:47]ESP-ROM:esp32s3-20210327
[23:18:47]Build:Mar 27 2021
[23:18:47]rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
[23:18:47]Saved PC:0x420e5e2e
[23:18:47]SPIWP:0xee
[23:18:47]mode:DIO, clock div:1
[23:18:47]load:0x3fce3808,len:0x43c
[23:18:47]load:0x403c9700,len:0xbec
[23:18:47]load:0x403cc700,len:0x2a3c
[23:18:47]entry 0x403c98d8
[23:18:47]ESP-ROM:esp32s3-20210327

So I stuck with between a rock and a hard place. Your external throwing the compilation error and the merged release failing ungracefully. Plus the yaml is like 3200 lines and I have no clue what is causing the runtime failure.  I thought the path of least resistance would be to check your repo and saw the post.  I wrote this back in June before the merge and there had been no reason until now to recompile it or modify to the merged release in August.   BTW: very impressive work you did with the LVGL component.

clydebarrow, 2024-12-03:
Well, the reason to go with the release is that it's the only supported version. If there's a problem with the release version, open a new issue. A couple of suggestions - the log you copied above is just from the bootloader, even with a crash I would expect more - make sure you are using the right serial port for logging - it defaults to USB_SERIAL_JTAG on S3, but some boards use UART0 (or both, on separate USB connectors.) Lambdas are prime suspects for run-time exceptions.

maurofain, 2024-12-03:
This is the yaml : https://pastebin.com/C4qrMNnh

this is the external ref:
external_components:
  - source:
      type: git
      url: https://github.com/clydebarrow/esphome
      ref: fd15094c0860df23d532881df36cfd16c7da1091 #previous commit - wont be needed in the future      
    components: [ lvgl ]

Is it possible to stop automatic update of ESPhome?

maurofain, 2024-12-04:
Thank you sirs, I removed thr external component and now, after same changes in the code since the version of lvgl as external component was quite old and some syntax now is different, it compiles.