esphome/issues #6514: bme68x_bsec2_i2c throws exception with ESP8266 + BME688

chrisbdaemon, 2024-12-03:
### The problem

Trying to use BME688 throws an exception and logs a stack trace.

Using ESP8266 Adafruit Huzzah board with BME688 breakout board.

### Which version of ESPHome has the issue?

2024.11.2

### What type of installation are you using?

pip

### What platform are you using?

ESP8266

### Board

huzzah

### Component causing the issue

bme68x_bsec2_i2c

### YAML Config

```yaml
esphome:
  name: office-air
  friendly_name: office-air

esp8266:
  board: huzzah

# Enable logging
logger:

captive_portal:

i2c:
  sda: GPIO4
  scl: GPIO5
  scan: true

bme68x_bsec2_i2c:
  address: 0x77
  model: BME688
  operating_age: 4d
```

### Anything in the logs that might be useful for us?

```txt
[00:27:05]--------------- CUT HERE FOR EXCEPTION DECODER ---------------
[00:27:05]
[00:27:05]Exception (3):
WARNING Exception type: LoadStoreError: Processor internal physical address or data error during load or store
[00:27:05]epc1=0x402030d2 epc2=0x00000000 epc3=0x00000000 excvaddr=0x40272a20 depc=0x00000000
WARNING Decoded 0x402030d2: rdqkshvtubwagfme
[00:27:05]
[00:27:05]>>>stack>>>
WARNING Found stack trace! Trying to decode it
[00:27:05]
[00:27:05]ctx: cont
[00:27:05]sp: 3ffff5b0 end: 3fffffd0 offset: 0150
[00:27:05]3ffff700:  feefeffe feefeffe feefeffe feefeffe  
[00:27:05]3ffff710:  feefeffe feefeffe feefeffe feefeffe  
[00:27:05]3ffff720:  feefeffe feefeffe feefeffe feefeffe  
[... 157 lines cut ...]
WARNING Decoded 0x4020b1f0: bme68x_soft_reset
[00:27:06]3fffff00:  c282e0b6 42fa0000 432a0000 3fff2388  
[00:27:06]3fffff10:  3ffe915e 3fff2fdc 3fff22ec 4020b2b4  
WARNING Decoded 0x4020b2b4: bme68x_init
[00:27:06]3fffff20:  000366e4 d7068d34 234b0058 1e1eff28  
[00:27:06]3fffff30:  fb1e0000 3e1ef39f 2d0033a1 eb9c7814  
[00:27:06]3fffff40:  bcde9d65 16aa2d12 4022034b 40201038  
WARNING Decoded 0x4022034b: esphome::wifi::WiFiComponent::print_connect_params_()
WARNING Decoded 0x40201038: bsec_init_m
[00:27:06]3fffff50:  3fff0744 00000002 3fff22dc 4021a7f9  
WARNING Decoded 0x4021a7f9: esphome::bme68x_bsec2::BME68xBSEC2Component::setup()
[00:27:06]3fffff60:  3fffff61 3fffff60 00000000 3fff22c0  
[00:27:06]3fffff70:  3fff22dc 00000003 3fff0740 40224549  
WARNING Decoded 0x40224549: esphome::Application::setup()
[00:27:06]3fffff80:  0000000b 0000000b 3fff32fc feefeffe  
[00:27:06]3fffff90:  00000010 3fff081c 3fff22dc feefeffe  
[00:27:06]3fffffa0:  feefeffe feefeffe feefeffe 3fff0c98  
[00:27:06]3fffffb0:  3fffdad0 00000000 3fff0c6c 4023459c  
WARNING Decoded 0x4023459c: loop_wrapper() at core_esp8266_main.cpp
[00:27:07]3fffffc0:  feefeffe feefeffe 3fffdab0 40100471  
WARNING Decoded 0x40100471: cont_wrapper
[00:27:07]<<<stack<<<
[00:27:07]
[00:27:07]--------------- CUT HERE FOR EXCEPTION DECODER ---------------
```

chrisbdaemon, 2024-12-03:
@neffs @kbx81

If you could take a look at this I would greatly appreciate it. You are listed as the code owners.

kbx81, 2024-12-03:
I highly recommend **not** using an 8266, but if you must, see [this comment](https://github.com/esphome/esphome/pull/4585#issuecomment-2282656538).

chrisbdaemon, 2024-12-03:
Understood, I had an ESP8266 within arms reach but I'll go digging for an ESP32.

Closing, will open new issue if it happens with ESP32.

closed 2024-12-03 by chrisbdaemon