esphome/issues #6516: Two of my ESPHome devices keep disconnecting from HA API

notsphinx, 2024-12-03:
### The problem

I have 103 ESPHome devices on my home network. Two of the devices disconnect from Home Assistant API. They are connected fine on WiFi and I can manage them using the web interface, but none of their related automations work. I can re-add the device to HA after which they stay connected for about 2 minutes and then drop off from HA even though they are reachable on the network. This is always specific to two of these devices (different types, so different yaml files) and this problem persists across device reboots and HA reboots. 

I have multiple devices of each of these types, so not specific to one particular device type. The WiFi signal strength according to these devices is about -57 or -58 dBm. I do not believe this is WiFi related since that could cause some degree of randomness. 

### Which version of ESPHome has the issue?

2024.11.2

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

2024.11.3

### What platform are you using?

BK72XX

### Board

Cree BR30 Light bulb (generic-bk7231t-qfn32-tuya)

### Component causing the issue

api

### YAML Config

```yaml
substitutions:
  name: cree-br30
  friendly_name: Cree BR30

esphome:
  name: ${name}
  name_add_mac_suffix: true

bk72xx:
  board: generic-bk7231t-qfn32-tuya
  framework:
    version: recommended

preferences:
  flash_write_interval: 1min

<<: !include common/network-lite.yaml

# Sync time with Home Assistant.
time:
  - platform: homeassistant
    id: homeassistant_time

button:
- platform: restart
  id: restart_button
  name: Restart
- platform: safe_mode
  id: safe_mode_restart_button
  name: Safe Mode Restart

debug:
  update_interval: 60s

# Individual on/off sensors
binary_sensor:
# Reports if this device is Connected or not
- platform: status
  name: Status

# Individual data sensors
sensor:
# Reports the WiFi signal strength/RSSI in dB, see https://esphome.io/components/sensor/wifi_signal.html
- platform: wifi_signal
  name: Signal
  update_interval: 60s
```

### Anything in the logs that might be useful for us?

```txt
Snapshot of logs between the connect and disconnect events:

[15:38:43][W][api.connection:092]: : Socket operation failed: BAD_INDICATOR errno=9
[15:38:43][D][api:103]: Accepted 172.x.x.x
[15:38:43][W][api.connection:092]: : Socket operation failed: HANDSHAKESTATE_READ_FAILED errno=9
[15:38:47][D][api:103]: Accepted 172.x.x.x
[15:38:47][W][component:237]: Component api took a long time for an operation (126 ms).
[15:38:47][W][component:238]: Components should block for at most 30 ms.
[15:38:47][D][api.connection:199]: aioesphomeapi (172.x.x.x) requested disconnected
[15:38:47][D][api:103]: Accepted 172.x.x.x
[15:38:47][W][component:237]: Component api took a long time for an operation (122 ms).
[15:38:47][W][component:238]: Components should block for at most 30 ms.
[15:38:47][D][api.connection:1446]: Home Assistant 2024.11.3 (172.x.x.x): Connected successfully
[15:38:47][D][time:050]: Synchronized time: 2024-12-02 15:38:47
[15:38:54][D][sensor:093]: 'Signal': Sending state -59.00000 dBm with 0 decimals of accuracy
[15:39:00][W][component:237]: Component wifi took a long time for an operation (116 ms).
[15:39:00][W][component:238]: Components should block for at most 30 ms.
[15:39:28][D][sensor:093]: 'Uptime': Sending state 1594.68616 minutes with 0 decimals of accuracy
[15:39:41][D][sensor:093]: 'Signal Percent': Sending state 84.00000 Signal % with 0 decimals of accuracy
[15:39:54][D][sensor:093]: 'Signal': Sending state -59.00000 dBm with 0 decimals of accuracy
[15:40:28][D][sensor:093]: 'Uptime': Sending state 1595.68616 minutes with 0 decimals of accuracy
[15:40:41][D][sensor:093]: 'Signal Percent': Sending state 84.00000 Signal % with 0 decimals of accuracy
[15:40:54][D][sensor:093]: 'Signal': Sending state -58.00000 dBm with 0 decimals of accuracy
[15:41:28][D][sensor:093]: 'Uptime': Sending state 1596.68616 minutes with 0 decimals of accuracy
[15:41:41][D][sensor:093]: 'Signal Percent': Sending state 84.00000 Signal % with 0 decimals of accuracy
[15:41:54][D][sensor:093]: 'Signal': Sending state -60.00000 dBm with 0 decimals of accuracy
[15:42:28][D][sensor:093]: 'Uptime': Sending state 1597.68616 minutes with 0 decimals of accuracy
[15:42:41][D][sensor:093]: 'Signal Percent': Sending state 82.00000 Signal % with 0 decimals of accuracy
[15:42:54][D][sensor:093]: 'Signal': Sending state -59.00000 dBm with 0 decimals of accuracy
[15:43:02][W][component:237]: Component wifi_info.text_sensor took a long time for an operation (54 ms).
[15:43:02][W][component:238]: Components should block for at most 30 ms.
[15:43:28][D][sensor:093]: 'Uptime': Sending state 1598.68616 minutes with 0 decimals of accuracy
[15:43:41][D][sensor:093]: 'Signal Percent': Sending state 82.00000 Signal % with 0 decimals of accuracy
[15:43:54][D][sensor:093]: 'Signal': Sending state -60.00000 dBm with 0 decimals of accuracy
[15:44:28][D][sensor:093]: 'Uptime': Sending state 1599.68616 minutes with 0 decimals of accuracy
[15:44:41][D][sensor:093]: 'Signal Percent': Sending state 82.00000 Signal % with 0 decimals of accuracy
[15:44:54][D][sensor:093]: 'Signal': Sending state -58.00000 dBm with 0 decimals of accuracy
[15:45:28][D][sensor:093]: 'Uptime': Sending state 1600.68616 minutes with 0 decimals of accuracy
[15:45:41][D][sensor:093]: 'Signal Percent': Sending state 86.00000 Signal % with 0 decimals of accuracy
[15:45:54][D][sensor:093]: 'Signal': Sending state -56.00000 dBm with 0 decimals of accuracy
[15:46:28][D][sensor:093]: 'Uptime': Sending state 1601.68616 minutes with 0 decimals of accuracy
[15:46:41][D][sensor:093]: 'Signal Percent': Sending state 88.00000 Signal % with 0 decimals of accuracy
[15:46:54][D][sensor:093]: 'Signal': Sending state -55.00000 dBm with 0 decimals of accuracy
[15:47:09][D][time:050]: Synchronized time: 2024-12-02 15:47:09
[15:47:09][D][time:050]: Synchronized time: 2024-12-02 15:47:09
[15:47:23][D][api.connection:199]: Home Assistant 2024.11.3 (172.x.x.x) requested disconnected

There are no logs at the default log level on HA when the disconnect happens.
```

ssieb, 2024-12-03:
The disconnect is coming from the HA side, not the device.  There are no logs on HA about this?  Check both the general ones and the device specific ones.

How are you getting these logs with the mac suffix enabled?

notsphinx, 2024-12-03:
Mystery partially resolved:

I looked in the HA logs with the device name instead of the device IP address. It looks like HA is attempting to connect to an IP address that is different than what the DHCP server assigned to the device. I tested by configuring the DHCP server to assign the address that HA was trying to connect to and the ESPHome device's connection to HA API is stable.

Where is the IP address of the device cached in HA? And why doesn't it auto update if the IP address changes?

ssieb, 2024-12-04:
Did you add the device by name or IP address?

spudje, 2024-12-04:
Would these be related? https://github.com/esphome/issues/issues/6283

notsphinx, 2024-12-04:
> Did you add the device by name or IP address?

Added using IP address. 

I suspect this may have something to do with MDNS discovery and the IP of the esphome device changing. I had a DHCP server crash and brought up another one which changed the IP address of the device. HA seems to either discover the old address or has it cached somewhere.

notsphinx, 2024-12-04:
> Would these be related? #6283

Symptoms seem similar. My esphome device is stable on WiFi - have uptimes running into days, reachable over webui, etc. HA logs seem to indicate that HA may be attempting to reach the device over a different IP address a few minutes after the API connection is established and then drops the connection. The current thing I am trying to chase down is either MDNS discovery or the hostname->IP address of the device is cached somewhere and resurfaces every now and then. Once HA drops its API connection, it never gets re-established until I add the device manually (I have always used the IP address to add the device).

notsphinx, 2024-12-04:
Does this mean that the better practice is to add the device using it's hostname?

randybb, 2024-12-05:
IP is fine, if you have a fixed IP on your device (just not flexible). If you can reliably resolve device IP from mDNS/DNS, then using its domain name is better.
I am using exclusively only domain names, my esphome devices are in a separate VLAN and mDNS is doing very good job (even with IPv6 addresses), so no connection issues. And yes, all devices (including switches, APs, printers. fixed IP only for devices which needs to be NATed) have a dynamic address from DHCP server.

notsphinx, 2024-12-10:
The two devices have been stable on the network ever since I nailed the IP address for them (in fact, for all ESPHome devices) in the DHCP server using static mappings. The issue may have happened because I had a DHCP server crash and brought another one up when the IP addresses may have changed. Can close this from my end. Thanks

closed 2024-12-10 by notsphinx