esphome/issues #6517: LVGL Spinbox triggers on_value when updated by lvgl.spinbox.update

uglide, 2024-12-03:
### The problem

I'm attempting to follow the documentation for Spinbox here https://esphome.io/cookbook/lvgl#climate-control

**The problem** with the spinbox is that it triggers on_value when it's programmatically updated by a sensor trigger. Because of that, the code gets stuck in an endless loop. 

**The correct behavior:** do not trigger  `on_value` when updating spinbox programmatically via `lvgl.spinbox.update`

### Which version of ESPHome has the issue?

2024.11.2

### What type of installation are you using?

pip

### What platform are you using?

ESP32-IDF

### Board

esp32-s3-devkitc-1

### Component causing the issue

lvgl

### YAML Config

```yaml
sensor:
  - platform: wifi_signal
    # name: "WiFi Signal Sensor"
    internal: True
    id: wifi_signal_sensor
    update_interval: 1s
  - platform: homeassistant
    id: igor_thermostat_temp
    entity_id: climate.sonoff_trvzb_thermostat
    attribute: temperature
  - platform: homeassistant
    id: igor_thermostat_curr_temp
    entity_id: climate.sonoff_trvzb_thermostat
    attribute: current_temperature
    on_value:
      - lvgl.spinbox.update:
          id: spinbox_id
          value: !lambda return x;
      - lvgl.label.update:
          id: lbl_igor_thermostat
          text:
            format: "Igor Office: %.0f°C"
            args: [ 'id(igor_thermostat_curr_temp).get_state()' ]

lvgl:
  on_idle:
    timeout: !lambda "return (id(display_timeout).state * 1000);"
    then:
      - logger.log: "LVGL is idle"
      - light.turn_off: display_backlight
      - lvgl.pause:
  displays:
    - main_display
  touchscreens:
    - main_touchscreen
  pages:
    - id: main_page
      widgets:
        # Igor Thermostat
        - obj:
            align: BOTTOM_MID
            y: -50
            layout:
              type: FLEX
              flex_flow: ROW
              flex_align_cross: CENTER
            width: SIZE_CONTENT
            height: SIZE_CONTENT
            widgets:
              - label:
                  id: lbl_igor_thermostat
                  align: LEFT_MID
                  text_align: LEFT
                  text_color: 0x000000
              - button:
                  id: spin_down
                  on_click:
                    - lvgl.spinbox.decrement: spinbox_id
                  widgets:
                    - label:
                        text: "-"
              - spinbox:
                  id: spinbox_id
                  align: CENTER
                  text_align: CENTER
                  width: 50
                  range_from: 15
                  range_to: 22
                  step: 1
                  rollover: false
                  digits: 3
                  decimal_places: 0

              - button:
                  id: spin_up
                  on_click:
                    - lvgl.spinbox.increment: spinbox_id
                    - logger.log:
                        format: "Spinbox value is %f"
                        args: [ !lambda "return id(spinbox_id).value;" ]
                    - homeassistant.action:
                        action: climate.set_temperature
                        data:
                          temperature: !lambda return x;
                          hvac_mode: heat
                          target.area_id: igor_s_office
                    - delay: 2s
                    - lvgl.spinbox.update:
                        id: spinbox_id
                        value: !lambda "return id(igor_thermostat_temp).get_state();"
                  widgets:
                    - label:
                        text: "+"
        # Igor Thermostat End
        - obj:
            align: TOP_MID
            styles: header_footer
            widgets:
              - label:
                  text: "Climate control"
                  align: CENTER
                  text_align: CENTER
                  text_color: 0xFFFFFF

  style_definitions:
    - id: header_footer
      bg_color: 0x2F8CD8
      bg_grad_color: 0x005782
      bg_grad_dir: VER
      bg_opa: COVER
      border_opa: TRANSP
      radius: 0
      pad_all: 0
      pad_row: 0
      pad_column: 0
      border_color: 0x0077b3
      text_color: 0xFFFFFF
      width: 100%
      height: 30
  top_layer:
    widgets:
      - label:
          text: "\uF1EB"
          id: lbl_hastatus
          hidden: true
          align: top_right
          x: -2
          y: 7
          text_align: right
          text_color: 0xFFFFFF
      - buttonmatrix:
          align: bottom_mid
          styles: header_footer
          pad_all: 0
          outline_width: 0
          id: top_layer
          items:
            styles: header_footer
          rows:
            - buttons:
                - id: page_prev
                  text: "\uF053"
                  on_press:
                    then:
                      lvgl.page.previous:
                - id: page_home
                  text: "\uF015"
                  on_press:
                    then:
                      lvgl.page.show: main_page
                - id: page_next
                  text: "\uF054"
                  on_press:
                    then:
                      lvgl.page.next:
```

clydebarrow, 2024-12-04:
I don't see any use of `on_value` for the spinbox in your yaml.

clydebarrow, 2024-12-04:
So what's the point of this issue? If there is an actual problem, you need to post the yaml that demonstrates the problem, not how you worked-around it.

uglide, 2024-12-05:
Sorry for the confusion. I just copied and pasted the example from https://esphome.io/cookbook/lvgl#climate-control now to make it less confusing. If you try to create a config based on the example from the LVGL cookbook, you will run into the issue of endless loop:
1. the sensor receives value from HA
2. Updates the spinbox
3. This update triggers `on_update` action
4. Spinbox runs the defined actions to update the remote entity in HA and the process repeats

uglide, 2024-12-05:
BTW, don't get me wrong. I really appreciate your hard work and enjoyed building a 7" sensor control panel for my HA based on the LVGL component in ESPHome. I just want to report that the example from the cookbook doesn't work and less experienced users might struggle with it.

closed 2024-12-10 by jesserockz

piposant, 2025-01-16:
Could you explain the solution? I have a problem

clydebarrow, 2025-01-16:
There is now (in the dev branch) another event `on_change` that is triggered user interaction only.

piposant, 2025-01-16:
So how would the code that is in the official documentation with the on_change event be so that it works?

sensor:
  - platform: homeassistant
    id: room_thermostat
    entity_id: climate.room_thermostat
    attribute: temperature
    on_value:
      - lvgl.spinbox.update:
          id: spinbox_id
          value: !lambda return x;

lvgl:
    ...
    pages:
      - id: thermostat_control
        widgets:
          - obj:
              align: BOTTOM_MID
              y: -50
              layout:
                type: FLEX
                flex_flow: ROW
                flex_align_cross: CENTER
              width: SIZE_CONTENT
              height: SIZE_CONTENT
              widgets:
                - button:
                    id: spin_down
                    on_click:
                      - lvgl.spinbox.decrement: spinbox_id
                    widgets:
                      - label:
                           text: "-"
                - spinbox:
                    id: spinbox_id
                    align: CENTER
                    text_align: CENTER
                    width: 50
                    range_from: 15
                    range_to: 35
                    step: 0.5
                    rollover: false
                    digits: 3
                    decimal_places: 1
                    on_value:
                      then:
                        - homeassistant.action:
                            action: climate.set_temperature
                            data:
                              temperature: !lambda return x;
                              entity_id: climate.room_thermostat
                - button:
                    id: spin_up
                    on_click:
                      - lvgl.spinbox.increment: spinbox_id
                    widgets:
                      - label:
                          text: "+"

clydebarrow, 2025-01-16:
Firstly, when including YAML or other code, surround it with backticks as per [the Github docs](https://docs.github.com/en/get-started/writing-on-github/getting-started-with-writing-and-formatting-on-github/basic-writing-and-formatting-syntax#quoting-code) - or just select the code and use the "code" button above (`<>`).

If you replace `on_value` in the widget config with `on_change`, it will not be triggered when the widget is updated externally.

piposant, 2025-01-17:
First of all, sorry for forgetting the "<>" in the previous update.
I put the on_change in the widget and the same thing keeps happening. The thermostat goes down to 15º and seems to get stuck in a loop.

piposant, 2025-01-17:
I've already solved the problem, in the sensor part I had to put: <value: !lambda return x * 10;> otherwise the spinbox wouldn't correctly take the value. But now I have another problem. When loading the esp32 and pressing "+" or "-" it doesn't modify the value. I have to put the focus on one of the digits so if pressing "+" or "-" does modify the value.