esphome/issues #6520: Cannot compile ESP-IDF components using non-root docker container (wrong cache folder)

r-xyz, 2024-12-04:
### The problem

When running in docker with non-root user, it is impossible to compile ESP-IDF components.

As a test, I tried to compile a slightly simplified version of m5stack.atom-echo-voice-assistant.

Notice, at the end of attached logs, how esphome tries to create folders under `/.cache/` instead of `/cache`.

Manually creating and setting permissions to `/.cache/` solves the problem.

Proposed fix: change cache folder for EspressIf to  `/cache/Espressif/`.

PS: it also affect at least 2024.10.x.

### Which version of ESPHome has the issue?

2024.11.x

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

N/A

### What platform are you using?

ESP32-IDF

### Board

m5stack-atom

### YAML Config

```yaml
esphome:
  name: test
  friendly_name: test

esp32:
  board: m5stack-atom
  framework:
    type: esp-idf

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "REDACTED"

ota:
  - platform: esphome
    password: "REDACTED"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Test Fallback Hotspot"
    password: "NvyaCbQBjf74"

http_request:

# dashboard_import:
[... 309 lines cut ...]
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /cache/platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-g++
-- Check for working CXX compiler: /cache/platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32
Dependencies lock doesn't exist, solving dependencies.
Updating lock file at /config/.esphome/build/test/dependencies.lock
-- Configuring incomplete, errors occurred!
See also "/config/.esphome/build/test/.pioenvs/test/CMakeFiles/CMakeOutput.log".

fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
CMake Error at /cache/platformio/packages/framework-espidf/tools/cmake/build.cmake:463 (message):
  ERROR: Failed to create cache directory: /.cache/Espressif/ComponentManager

Call Stack (most recent call first):
  /cache/platformio/packages/framework-espidf/tools/cmake/project.cmake:476 (idf_build_process)
  CMakeLists.txt:3 (project)

========================== [FAILED] Took 7.27 seconds ==========================
```

FeatherKing, 2025-01-29:
still a problem in 2024.12.4

Tank5050, 2025-03-04:
This appears to be fixed as of 2025.2.1 (maybe earlier). Its been working well for me.