esphome/issues #6522: local SVG images (non-mdi)

V1pr, 2024-12-05:
### The problem

ESPHome fails to build if image is SVG (pillow and cariosvg installed).

Build error:
```
  File "/usr/local/bin/esphome", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/usr/local/lib/python3.12/site-packages/esphome/__main__.py", line 1036, in main
    return run_esphome(sys.argv)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/esphome/__main__.py", line 1023, in run_esphome
    rc = POST_CONFIG_ACTIONS[args.command](args, config)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/esphome/__main__.py", line 503, in command_run
    exit_code = write_cpp(config)
                ^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/esphome/__main__.py", line 212, in write_cpp
    generate_cpp_contents(config)
  File "/usr/local/lib/python3.12/site-packages/esphome/__main__.py", line 224, in generate_cpp_contents
    CORE.flush_tasks()
  File "/usr/local/lib/python3.12/site-packages/esphome/core/__init__.py", line 684, in flush_tasks
    self.event_loop.flush_tasks()
  File "/usr/local/lib/python3.12/site-packages/esphome/coroutine.py", line 246, in flush_tasks
    next(task.iterator)
  File "/usr/local/lib/python3.12/site-packages/esphome/__main__.py", line 204, in wrapped
    await coro(conf)
  File "/usr/local/lib/python3.12/site-packages/esphome/components/image/__init__.py", line 290, in to_code
    image = Image.open(io.BytesIO(file_contents))
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/PIL/Image.py", line 3498, in open
    raise UnidentifiedImageError(msg)
PIL.UnidentifiedImageError: cannot identify image file <_io.BytesIO object at 0x7f86794039c0>

```

### Which version of ESPHome has the issue?

2024.11.2

### What type of installation are you using?

pip

### What platform are you using?

ESP32-IDF

### Board

Guition display board (ESP-S3)

### Component causing the issue

image

### YAML Config

```yaml
substitutions:
  name: "guition-esp32-s3-4848s040"
  friendly_name: "Guition480-basic"
  device_description: "Guition ESP32-S3-4848S040 480*480 Smart Screen"
  project_name: "Guition.ESP32_S3_4848S040"
  project_version: "1.0.0"

esphome:
  name: "${name}"
  friendly_name: "${friendly_name}"
  #name_add_mac_suffix: true
  project:
    name: "${project_name}"
    version: "${project_version}"
  platformio_options:
    board_build.flash_mode: dio

esp32:
  board: esp32-s3-devkitc-1
  variant: esp32s3
  flash_size: 16MB
  framework:
    type: esp-idf
    sdkconfig_options:
      COMPILER_OPTIMIZATION_SIZE: y
      CONFIG_ESP32S3_DEFAULT_CPU_FREQ_240: "y"
      CONFIG_ESP32S3_DATA_CACHE_64KB: "y"
      CONFIG_ESP32S3_DATA_CACHE_LINE_64B: "y"
      CONFIG_SPIRAM_FETCH_INSTRUCTIONS: y
      CONFIG_SPIRAM_RODATA: y

psram:
  mode: octal
  speed: 80MHz

logger:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

web_server:
  port: 80

#
#
# Images
#
#
image:
  - file: "/usr/src/esphome-configs-main/assets/forecast-icons/000.svg"
    id: idokep_fc_icon_000
    resize: 80x80

#
#-------------------------------------------
# LVGL Buttons
#-------------------------------------------
#
lvgl:
  displays:
    - my_display
  touchscreens:
    - touchscreen_id: my_touchscreen
  on_idle:
    - timeout: 10s
      then:
        - logger.log: idle timeout
        - if:
            condition:
              lvgl.is_idle:
                timeout: 5s
            then:
              - logger.log: LVGL is idle
    - timeout: 15s
      then:
        - logger.log: idle 15s timeout

      #- lvgl.pause:
      #- light.turn_off:
      #    id: display_backlight
      #    transition_length: 5s

#-------------------------------------------
# Internal outputs
#-------------------------------------------
output:
  # Backlight LED
  - platform: ledc
    pin: GPIO38
    id: GPIO38
    frequency: 100Hz

    # Built in 240v relay
  - id: internal_relay_1
    platform: gpio
    pin: 40

#-------------------------------------------
# Touchscreen gt911 i2c
#-------------------------------------------
i2c:
  - id: bus_a
    sda: GPIO19
    scl: GPIO45
    #frequency: 100kHz

touchscreen:
  platform: gt911
  transform:
    mirror_x: false
    mirror_y: false
  id: my_touchscreen
  display: my_display

  on_touch:
    - logger.log:
        format: Touch at (%d, %d)
        args: [touch.x, touch.y]
    - lambda: |-
        ESP_LOGI("cal", "x=%d, y=%d, x_raw=%d, y_raw=%0d",
            touch.x,
            touch.y,
            touch.x_raw,
            touch.y_raw
            );

#-------------------------------------------
# Display st7701s spi
#-------------------------------------------
spi:
  - id: lcd_spi
    clk_pin: GPIO48
    mosi_pin: GPIO47

display:
  - platform: st7701s
    id: my_display
    update_interval: never
    auto_clear_enabled: False
    spi_mode: MODE3
    data_rate: 2MHz
    color_order: RGB
    invert_colors: False
    dimensions:
      width: 480
      height: 480
    cs_pin: 39
    de_pin: 18
    hsync_pin: 16
    vsync_pin: 17
    pclk_pin: 21
    pclk_frequency: 12MHz
    pclk_inverted: False
    hsync_pulse_width: 8
    hsync_front_porch: 10
    hsync_back_porch: 20
    vsync_pulse_width: 8
    vsync_front_porch: 10
    vsync_back_porch: 10
    init_sequence:
      - 1
      # Custom sequences are an array, first byte is command, the rest are data.
      - [0xFF, 0x77, 0x01, 0x00, 0x00, 0x10] # CMD2_BKSEL_BK0
      - [0xCD, 0x00] # disable MDT flag
    data_pins:
      red:
        - 11 #r1
        - 12 #r2
        - 13 #r3
        - 14 #r4
        - 0 #r5
      green:
        - 8 #g0
        - 20 #g1
        - 3 #g2
        - 46 #g3
        - 9 #g4
        - 10 #g5
      blue:
        - 4 #b1
        - 5 #b2
        - 6 #b3
        - 7 #b4
        - 15 #b5
```

### Anything in the logs that might be useful for us?

```txt
INFO Generating C++ source...
Traceback (most recent call last):
  File "/usr/local/bin/esphome", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/usr/local/lib/python3.12/site-packages/esphome/__main__.py", line 1036, in main
    return run_esphome(sys.argv)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/esphome/__main__.py", line 1023, in run_esphome
    rc = POST_CONFIG_ACTIONS[args.command](args, config)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/esphome/__main__.py", line 503, in command_run
    exit_code = write_cpp(config)
                ^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/esphome/__main__.py", line 212, in write_cpp
    generate_cpp_contents(config)
  File "/usr/local/lib/python3.12/site-packages/esphome/__main__.py", line 224, in generate_cpp_contents
    CORE.flush_tasks()
  File "/usr/local/lib/python3.12/site-packages/esphome/core/__init__.py", line 684, in flush_tasks
    self.event_loop.flush_tasks()
  File "/usr/local/lib/python3.12/site-packages/esphome/coroutine.py", line 246, in flush_tasks
    next(task.iterator)
  File "/usr/local/lib/python3.12/site-packages/esphome/__main__.py", line 204, in wrapped
    await coro(conf)
  File "/usr/local/lib/python3.12/site-packages/esphome/components/image/__init__.py", line 290, in to_code
    image = Image.open(io.BytesIO(file_contents))
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/PIL/Image.py", line 3498, in open
    raise UnidentifiedImageError(msg)
PIL.UnidentifiedImageError: cannot identify image file <_io.BytesIO object at 0x7f09a6080bd0>
```

### Additional information

SVG content:
```
<?xml version="1.0" encoding="UTF-8"?><svg xmlns="http://www.w3.org/2000/svg" width="1" height="1"/>
```

randybb, 2024-12-05:
I guess there is the same limitation like in https://esphome.io/components/online_image#online-image - only png. But instead of crashing it could show a nice error that only png is supported.

clydebarrow, 2024-12-08:
SVG is supported, but it seems that file isn't being recognised as valid SVG. Please test with this PR. https://github.com/esphome/esphome/pull/7908

clydebarrow, 2024-12-08:
And it seems the reason the file is not recognised is because it does not *end* with `</svg>`

V1pr, 2024-12-09:
> SVG is supported, but it seems that file isn't being recognised as valid SVG. Please test with this PR. [esphome/esphome#7908](https://github.com/esphome/esphome/pull/7908)

@clydebarrow  thanks for the fix, it's working for me! (sidenote: I had to adjust/add some variables, since transparency changed from 'true' to a bunch of options, ```type``` got mandatory)

closed 2025-01-13 by jesserockz