esphome/issues #6523: After updateing to 2024.12.0 cant install ESPHome Device Compiler

GekoCH, 2024-12-06:
### The problem

After I updated my HA to 2024.12.0, Supervisor 2024.11.4, Operating System 14.0 ESPHome Device Compiler was removed.
now I tried to install it via [https://esphome.io/guides/getting_started_hassio.html#installing-esphome-device-compiler](https://esphome.io/guides/getting_started_hassio.html#installing-esphome-device-compiler) but it only says "Add-on not found"

Even when i search for ESPHOME in the Addon menu of HA I can't find it...

Any idea?

### Which version of ESPHome has the issue?

*

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.12.0

### What platform are you using?

ESP8266

GekoCH, 2024-12-06:
ok after loading the backup back and redo the 2024.12 Installation it works now...

jbabio, 2025-02-05:
Hello, Im in:

- Core: 2025.1.4
- Supervisor: 2024.12.3
- Operating System: 14.2
- Frontend: 20250109.2

A remediation prompted that the addon was not available and proposed to delete it. I deleted it cause I thought I had the old HACS addon, but I cannot install the Device compiler again. It always tells that the addon is not available. The ESPHome Integrations are still available, but not the Devide Builder

jbabio, 2025-02-05:
Solved. I was able to install the "ESPHome Device Builder".
 The "Show Addon on MY" button link seems to be wrong. I had to add "/tree/main/esphome" to the end of the link.
Fixed link: [Show Addon on MY](https://my.home-assistant.io/redirect/supervisor_addon/?addon=5c53de3b_esphome&repository_url=https%3A%2F%2Fgithub.com%2Fesphome%2Fhome-assistant-addon%2Ftree%2Fmain%2Fesphome)