esphome/issues #6524: LVGL stack overflow

plplaaa2, 2024-12-06:
### The problem

After updating to esphome 24.11, devices that were good before fall into stack overflow state. 

Even with the same yaml, if you update it, it falls into stack overflow, and the exact cause is not known. 

However, it only occurs on devices using LVGL and especially frequently occurs on waveshare ESP32 S3 LCD 7.

I hope you understand that the entire yaml didn't put everything beyond the 10000 line.

### Which version of ESPHome has the issue?

24.11.1,2,3 

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

24.12

### What platform are you using?

ESP32-IDF

### Board

ESP32 S3

### Component causing the issue

LVGL

### YAML Config

```yaml
esphome:
  name: "${name}"
  comment: "${device_description}"
  platformio_options:
    build_flags: "-DBOARD_HAS_PSRAM"
    board_build.arduino.memory_type: qio_opi
    board_build.flash_mode: dio
    #board_upload.maximum_ram_size: 524288

  on_boot: 
    then:
      - light.turn_on: 
          id: backlight
          brightness: !lambda return id(backlight_brightness).state / 100;
          
esp32:
  board: ${board}
  framework:
    type: ${type}
    sdkconfig_options:
      COMPILER_OPTIMIZATION_SIZE: y
      CONFIG_ESP32S3_DEFAULT_CPU_FREQ_240: "y"
      CONFIG_ESP32S3_DATA_CACHE_64KB: "y"
      CONFIG_ESP32S3_DATA_CACHE_LINE_64B: "y"
      CONFIG_SPIRAM_FETCH_INSTRUCTIONS: "y"
      CONFIG_SPIRAM_RODATA: "y"
      CONFIG_ESP32_S3_BOX_BOARD: "y"
  flash_size: 8MB
  partitions: "default_8MB.csv"
      
psram:
  mode: octal
  speed: 80MHz
[... 783 lines cut ...]
[19:11:06][I][lvgl:000]: flex_update: update 0x3db97124 container 	(in lv_flex.c line #208)
[19:11:06][I][lvgl:000]: flex_update: update 0x3db975f8 container 	(in lv_flex.c line #208)
[19:11:06][I][lvgl:000]: lv_obj_update_layout: Layout update begin 	(in lv_obj_pos.c line #314)
[19:11:06][I][lvgl:000]: flex_update: update 0x3db97124 container 	(in lv_flex.c line #208)
[19:11:06][I][lvgl:000]: flex_update: update 0x3db975f8 container 	(in lv_flex.c line #208)
[19:11:06][I][lvgl:000]: lv_obj_update_layout: Layout update begin 	(in lv_obj_pos.c line #314)
[19:11:06][I][lvgl:000]: lv_obj_update_layout: Layout update begin 	(in lv_obj_pos.c line #314)
[19:11:06]
[19:11:06]***ERROR*** A stack overflow in task loopTask has been detected.
[19:11:06]
[19:11:06]
[19:11:06]Backtrace: 0x40376ea6:0x3fcea240 0x4037cbf5:0x3fcea260 0x4037fcf2:0x3fcea280 0x4037e73d:0x3fcea300 0x4037ccac:0x3fcea330 0x4037cca2:0xa5a5a5a5 |<-CORRUPTED
[19:11:06]
[19:11:06]
[19:11:06]
[19:11:06]
[19:11:06]ELF file SHA256: 285bb87db7ae9f6a
[19:11:06]
[19:11:06]Rebooting...
[19:11:06]ESP-ROM:esp32s3-20210327
[19:11:06]Build:Mar 27 2021
[19:11:06]rst:0xc (RTC_SW_CPU_RST),boot:0x2b (SPI_FAST_FLASH_BOOT)
[19:11:06]Saved PC:0x4037fb1b
Terminal disconnected: NetworkError: The device has been lost.```
```

ESP32andmore, 2024-12-07:
I don't see any LVGL components or widget defined. Is there more YAML?

ESP32andmore, 2024-12-07:
Most likely it is caused by automations for synchronizing between components and widgets that cause recursion.

jzucker2, 2024-12-08:
I would suspect it's related to this https://github.com/esphome/issues/issues/6515

plplaaa2, 2024-12-08:
> I don't see any LVGL components or widget defined. Is there more YAML?

I'm attaching a page because it's too long.

```packages:
  # base
  base: !include lvgl/base-800.yaml
  display: !include lvgl/waveshare-800.yaml
  font: !include lvgl/fonts-800.yaml
  image: !include weather/weather-icon.yaml
  # lvgl  
  style: !include lvgl/styles-800.yaml
  # setting: !include lvgl/setting-800.yaml  
  # sensors  
  naver: !include lvgl/naver-800.yaml
  indoor: !include lvgl/indoor-800.yaml
  outside: !include lvgl/outdoor-800.yaml
  energy: !include lvgl/energy-800.yaml
  # device sensors
  switch: !include lvgl/switch-800.yaml
  aircon: !include lvgl/aircon-800.yaml
  humidfier: !include lvgl/humidifier-800.yaml
  purifier: !include lvgl/purifier-800.yaml
  door: !include lvgl/door-800.yaml  
  blind: !include lvgl/blind-800.yaml
  radio: !include lvgl/radio-800.yaml
  boiler: !include lvgl/boiler-thermostat.yaml
  # pages
  menu-page: !include lvgl/menu-page-800.yaml
  weather-page: !include lvgl/weather-page-800.yaml
  light-page: !include lvgl/light-page-800.yaml
  climate-page: !include lvgl/climate-page-800.yaml
  security-page: !include lvgl/security-page-800.yaml
  energy-page: !include lvgl/energy-page-800.yaml
  setting-page: !include lvgl/setting-page-800.yaml```

```
```
text_sensor:
# 에어컨 텍스트 센서
  - platform: homeassistant
    entity_id: climate.eeokeon
    name: aircon_mode
    id: aircon_mode
    on_value:      
      - lvgl.label.update: 
          id: aircon_mode_text
          text:
            format: "%s"
            args: [ x.c_str() ]
      - if:
          condition:
            lambda: 'return id(aircon_mode).state == "heat_cool";'
          then:
            - lvgl.dropdown.update:
                id: aircon_mode_set
                selected_index: 0
            - lvgl.label.update: 
                id: aircon_mode_text
                text: "#FFFFFF 자동#"
      - if:
          condition:
            lambda: 'return id(aircon_mode).state == "heat";'
          then:
            - lvgl.dropdown.update:
                id: aircon_mode_set
                selected_index: 1
            - lvgl.label.update: 
                id: aircon_mode_text
                text: "#FFA500 난방#"    
[... 390 lines cut ...]
                              then:
                                - homeassistant.action:
                                    action: climate.set_fan_mode
                                    data: 
                                      entity_id: $aircon_id
                                      fan_mode: 'high'
                          - if:
                              condition:
                                - lambda: 'return x == 2;'
                              then:
                                - homeassistant.action:
                                    action: climate.set_fan_mode
                                    data: 
                                      entity_id: $aircon_id
                                      fan_mode: 'medium'
                          - if:
                              condition:
                                - lambda: 'return x == 3;'
                              then:
                                - homeassistant.action:
                                    action: climate.set_fan_mode
                                    data: 
                                      entity_id: $aircon_id
                                      fan_mode: 'low'
 ```

plplaaa2, 2024-12-08:
[https://www.youtube.com/shorts/WfNRwrhKC6Q](url)

This is a video of my device working normally at 10.3.

ESP32andmore, 2024-12-08:
I know from personal experience that with sliders and arcs, the widget updating a component "on_value" when that component also updates the widget, will cause recursion and stack overflows in the current release. But there you did use "on_release" and that is OK.  I see on your dropdown aircon_mode and aircon_fan_mode interaction event trigger is "on_value" and your component updates the widget accordingly.  I personally have not used a dropdown, but comparing that to a silder or arc, I believe it does "on_value" as you select each value. 

(on_value [trigger](https://esphome.io/automations/actions#actions-trigger) is activated only when you select an item from the list. The new selected index is returned in the variable x)

It is possible this causes a stack overflow.  Maybe try using "on_release" to throw the action.

Also what was helpful to me because my issue presented at boot, was using the serial (TXD/RXD) port if its available to you to hook directly to the HA target and it will do a stack decode of the trace info.  It will generally point to the line of YAML that is at issue. Is this crash happening at boot time or while interacting with your device?

closed 2024-12-10 by jesserockz