esphome/issues #6526: As of 2024.11.3 cannot compile anymore

jelmerkk, 2024-12-06:
### The problem

All compiles give the same error:
`xtensa-esp32-elf-g++: fatal error: Killed signal terminated program cc1plus
compilation terminated.`
This happens with own YAML but also with packages such as bluetooth proxy or the voice assistant

### Which version of ESPHome has the issue?

2024.11.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.12.1

### What platform are you using?

ESP8266

### Board

shelly, atom echo, nodemcu

ssieb, 2024-12-07:
You probably need more RAM.  Otherwise, change the addon settings to reduce the number of compile processes.

jelmerkk, 2024-12-07:
Hmm. After HAOS reboot a compile went OK without using more than 60% of the 4GB RAM. 
I'll see if the issue resurfaces, thanks for thinking along!

closed 2024-12-07 by jelmerkk

ssieb, 2024-12-07:
After a reboot, less things are running.  Just make sure you don't end up thrashing the swap.

jelmerkk, 2024-12-07:
Yeah and it's very tempting to solve it permanently with an 8Gb CM5...