esphome/issues #6530: online_image can run forever, never succeeding, erroring or tripping watchdog.

nspitko, 2024-12-08:
### The problem

I'm running a waveshare s3 LCD device that has a relatively simple task: Update the image once per minute from an external source (via online_image), and show the time.

This mostly works! With two caveats:
1) online_image is very slow, and often will not update within the 1 minute window. This is mostly OK for my use case, but seems really weird

2) sometimes, online_image gets "stuck". It doesn't succeed or error, it just continues to run seemingly forever. It's been blocking for a suspiciously consistent amount of time for over two hours now, so it seems unlikely any kind of timeout is working.

```
[15:19:11][W][component:237]: Component online_image took a long time for an operation (4501 ms).
[15:19:11][W][component:238]: Components should block for at most 30 ms.
[15:19:16][W][component:237]: Component online_image took a long time for an operation (4501 ms).
[15:19:16][W][component:238]: Components should block for at most 30 ms.
[15:19:20][W][component:237]: Component online_image took a long time for an operation (4501 ms).
[15:19:20][W][component:238]: Components should block for at most 30 ms.
[15:19:25][W][component:237]: Component online_image took a long time for an operation (4501 ms).
[15:19:25][W][component:238]: Components should block for at most 30 ms.
[15:19:30][W][component:237]: Component online_image took a long time for an operation (4501 ms).
[15:19:34][W][component:238]: Components should block for at most 30 ms.
[15:19:34][W][component:237]: Component online_image took a long time for an operation (4501 ms).
[15:19:34][W][component:238]: Components should block for at most 30 ms.
[15:19:38][W][component:237]: Component online_image took a long time for an operation (4501 ms).
[15:19:39][W][component:238]: Components should block for at most 30 ms.
```

Unsure how to approach debugging this given the intermittent nature of this, and the fact that it's not hitting any timeouts/watchdogs.

### Which version of ESPHome has the issue?

2024.11.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.10.3

### What platform are you using?

ESP32-IDF

### Board

:ESP32-S3-Touch-LCD-7

### Component causing the issue

online_image

### YAML Config

```yaml
esphome:
  name: esp-lcd-7-01
  friendly_name: esp-lcd-7-01
  platformio_options:
    build_flags: "-DBOARD_HAS_PSRAM -DCONFIG_SPIRAM_CACHE_WORKAROUND"
    board_build.arduino.memory_type: qio_opi
    board_build.flash_mode: dio
    board_upload.maximum_ram_size: 524288

esp32:
  board: esp32-s3-devkitc-1
  variant: esp32s3
  framework:
    type: esp-idf
    version: 5.3.1
[... 312 lines cut ...]
debug:

sensor:
  - platform: template
    name: "bgIdx"
    lambda: |-
      return id(bgIdx);
    update_interval: 60s
    device_class: "distance"
  - platform: debug
    free:
      name: "Heap Free"
    block:
      name: "Heap Max Block"
    loop_time:
      name: "Loop Time"
    psram:
      name: "Free PSRAM"
```

### Additional information

The example script cycles through images named "Test (1).png" with an incrementing index. check globals for the number of images to cycle through. These are 800x480 images, but can be smaller obviously.

Note this took several days to happen, so it's not going to be fun/easy to diagnose.

clydebarrow, 2024-12-09:
4.5s is the default `http_request` timeout so it's just timing out.

https://arc.net/l/quote/tagiynmr

nspitko, 2024-12-09:
The documentation should probably clearly state the timeout behavior here if it's just going to silently retry forever. In pretty much every other HTTP lib a timeout is an error state, and even within esphome, socket select timeouts go through `on_error`. 

The inconsistency, combined with an inability to trap the error, seems bad. At a minimum, a `retry_limit` should be added to kick the esp down `on_error` for an image that will never succeed.

nspitko, 2024-12-09:
Ok I tried upping the timeout to 10s, assuming that maybe one of the larger images in the set just didn't quite make it under 4.5s, or wifi quality was affecting it, but it's still falling into this trap. You can see it struggling a bit to hit the 1 image/minute speed for a bit until it just flatlines. I assume it's stuck in the same loop as before, but I can't verify as it seems to be almost entirely unresponsive; I can connect but timeout before I can get through the preamble to get logging data. The device is also showing a couple minutes behind, though it is still updating the time.

Ultimately there's still seems to be two problems here; the issue preventing it from successfully downloading the image, and the bug preventing us from ever aborting out of the failed state.

[image]

clydebarrow, 2024-12-26:
Setting the PSRAM to 120MHz actually results in it being set to 40MHz in my experience. Check your serial log after boot for lines like this:

```
[19:27:15]I (549) esp_psram: Found 8MB PSRAM device
[19:27:15]I (554) esp_psram: Speed: 40MHz
```

That is from an S3 with PSRAM set to 120MHz. Setting it to 80 shows 80MHz in the log as expected.

clydebarrow, 2024-12-26:
Also, I would remove `    build_flags: "-DBOARD_HAS_PSRAM -DCONFIG_SPIRAM_CACHE_WORKAROUND"` from your config. The first one is redundant, the second is not needed except for some old ESP32s, and may cause a performance hit on other chips.