esphome/issues #6532: Extend not working for wifi_info/ethernet_info sensors

lcnittl, 2024-12-09:
### The problem

Extending the `wifi_info`'s and `ethernet_info`'s sensors, such as `ip_address`, is not possible and fails with `text_sensor.unknown` error messages.

In my actual use case, the `on_value` triggers an update of a status LED and display component.

Packaging is used in this case, to define a standard set of sensors for all my ESPHome devices. Since not all of them have displays/status LEDs, only some of them need this extension.

### Which version of ESPHome has the issue?

2024.11.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.12.1

### What platform are you using?

ESP32

### Board

esp32dev

### Component causing the issue

ethernet_info

### YAML Config

```yaml
# /config/esphome/mydevice.yaml
esphome:
  name: mydevice
  friendly_name: My Device

esp32:
  board: esp32dev
  framework:
    type: arduino

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
captive_portal:

packages:
  default_configs: !include
    file: mydevice/default_configs.yaml

text_sensor:
  - platform: wifi_info
    ip_address:
      id: !extend ip_addr
      on_value:
        then: # log is just for MWE, in real case updates LED and display
          - logger.log: New IP address
    mac_address:
      id: !extend mac_addr
      on_value:
        then: # log is just for MWE, in real case updates LED and display
          - logger.log: New MAC address
---
# /config/esphome/mydevice/default_configs.yaml
text_sensor:
  - platform: wifi_info
    ip_address:
      name: IP
      id: ip_addr
      icon: mdi:ip-network-outline
      entity_category: diagnostic
      disabled_by_default: true
    mac_address:
      name: MAC-Address
      id: mac_addr
      icon: mdi:hexadecimal
      entity_category: diagnostic
      disabled_by_default: true
```

### Anything in the logs that might be useful for us?

```txt
INFO ESPHome 2024.11.3
INFO Reading configuration /config/esphome/mydevice.yaml...
Failed config

text_sensor.unknown: [source /config/esphome/mydevice.yaml:381]
  
  Source for extension of ID 'ip_addr' was not found.
  id: !extend ip_addr
  on_value: 
    then: 
      - script.execute: update_connectivity_status
text_sensor.unknown: [source /config/esphome/mydevice.yaml:385]
  
  Source for extension of ID 'mac_addr' was not found.
  id: !extend mac_addr
  on_value: 
    then: 
      - script.execute: update_connectivity_status
```

### Additional information

I also tried adding an `id` to the component itself:

```yaml
text_sensor:
  - platform: wifi_info
    id: wifi_info
    ip_address:
      name: IP
      id: ip_addr
      icon: mdi:ip-network-outline
      entity_category: diagnostic
      disabled_by_default: true
    mac_address:
      name: MAC-Address
      id: mac_addr
      icon: mdi:hexadecimal
      entity_category: diagnostic
      disabled_by_default: true
```

which leads to a different error:

```log
[id] is an invalid option for [text_sensor.wifi_info]. Did you mean [ssid]?
```

closed 2025-04-23 by github-actions[bot]

lcnittl, 2025-04-23:
Still an issue

marshn, 2025-05-31:
I have exactly the same problem with 2025.5