esphome/issues #6534: while/repeat/wait_until will corrupt arguments when run concurrently

afflux, 2024-12-10:
### The problem

The actions are instantiated once per definition but they store the trigger arguments in their class member ([example](https://github.com/esphome/esphome/blob/444e162c92c3148a14493f66d3de256679a6eedf/esphome/core/base_automation.h#L356)), which gets overridden, causing corruption when they are called multiple times.

### Which version of ESPHome has the issue?

2024.12.0-dev

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

n/a

### What platform are you using?

RP2040

### Board

Pico W

### Component causing the issue

core

### YAML Config

```yaml
esphome:
  name: test-2040

rp2040:
  board: rpipicow

wifi:
  ssid: Ulai
  password: !secret wifi_password
  fast_connect: true

  # Optional manual IP
  manual_ip:
    static_ip: 192.168.1.91
    gateway: 192.168.1.1
    subnet: 255.255.255.0
    dns1: 192.168.1.1

logger:
  level: ERROR

ota:
  - platform: esphome
    password: !secret ota_password

api:
  reboot_timeout: 0s

globals:
   - id: global_counter
     type: size_t

time:
  - platform: sntp
    id: sntp_time
    timezone: Etc/UTC
    servers:
      - 192.168.1.1

    on_time:
      - seconds: /10
        then:
          - http_request.get:
              url: !lambda |-
                return "https://httpbin.org/anything?x=" + std::to_string(id(global_counter)++);
              capture_response: true
              max_response_buffer_size: 4kB
              on_response:
                then:
                  - lambda: |-
                      const char *const TAG = "TEST";
                      
                      json::parse_json(*body, [TAG](JsonObject root) -> bool {

                        JsonObject args = root["args"];
                        if (!args) {
                          ESP_LOGE(TAG, "no args");
                          return false;
                        }
                        const char *arg = args["x"];
                        ESP_LOGE(TAG, "got response for request #%s", arg);
                        return true;
                      });

                  - wait_until:
                      condition:
                        lambda: return (id(global_counter) % 5) == 0;

                  - lambda: |-
                      const char *const TAG = "TEST_AFTER_WAIT";
                      
                      json::parse_json(*body, [TAG](JsonObject root) -> bool {

                        JsonObject args = root["args"];
                        if (!args) {
                          ESP_LOGE(TAG, "no args");
                          return false;
                        }
                        const char *arg = args["x"];
                        ESP_LOGE(TAG, "got response for request #%s", arg);
                        return true;
                      });

http_request:
  verify_ssl: false
```

### Anything in the logs that might be useful for us?

```txt
[09:58:41][E][TEST:068]: got response for request #0
[09:58:51][E][TEST:068]: got response for request #1
[09:59:00][E][TEST:068]: got response for request #2
[09:59:10][E][TEST:068]: got response for request #3
[09:59:22][E][TEST:068]: got response for request #4
[09:59:22][E][TEST_AFTER_WAIT:089]: got response for request #4
[09:59:22][E][TEST_AFTER_WAIT:089]: got response for request #3
[09:59:22][E][TEST_AFTER_WAIT:089]: got response for request #3
[09:59:22][E][TEST_AFTER_WAIT:089]: got response for request #3
[09:59:22][E][TEST_AFTER_WAIT:089]: got response for request #3
```

afflux, 2024-12-10:
Same thing applies to `ble_client.connect`, `ble_client.disconnect` and `ble_client.ble_write`

ssieb, 2024-12-10:
The problem is that there's only one instance of the action, so where do you store the variables?
But triggers should not be used for long actions.  Use a script instead.  Those might not have the same issue, but if they do, they will get fixed.  Triggers probably won't.

afflux, 2024-12-10:
Generally, I'd expect the same thing that happens in interpreted languages: one object that represents the code, and one object *per call* that represents the actual arguments, locals, ...

You could store the arguments in a std::queue - that would also solve the problem of non-default-constructible arguments.

And while we're at it, for added clarity, I think you could directly store the bound `std::bind(&WaitUntilAction::play_next_, &this, x...)` in that std::queue.

IMHO with the way it is right now, somebody will someday be bitten by undocumented behavior that is frankly hard to explain to non-devs.

afflux, 2024-12-10:
Regarding your suggestion to use a script: I think `mode: parallel` suffers from the same issue.

ssieb, 2024-12-10:
It's possible and if it does, that will get fixed.  But triggers really only have a single instance and should never end up in parallel.

afflux, 2024-12-10:
Just verified:

```yaml
script:
  - id: handle_response
    mode: parallel
    parameters:
      body: string
    then:
      - lambda:
        |-
          const char *const TAG = "TEST";

          json::parse_json(body, [TAG](JsonObject root) -> bool {

            JsonObject args = root["args"];
            if (!args) {
              ESP_LOGE(TAG, "no args");
              return false;
            }

            const char *arg = args["x"];
            ESP_LOGE(TAG, "got response for request #%s", arg);
            return true;
          });

      - wait_until:
          condition:
            lambda: return (id(global_counter) % 5) == 0;

      - lambda: |-
          const char *const TAG = "TEST_AFTER_WAIT";

          json::parse_json(body, [TAG](JsonObject root) -> bool {

            JsonObject args = root["args"];
            if (!args) {
              ESP_LOGE(TAG, "no args");
              return false;
            }
            const char *arg = args["x"];
            ESP_LOGE(TAG, "got response for request #%s", arg);
            return true;
          });

time:
  - platform: sntp
    id: sntp_time
    timezone: Europe/Berlin
    servers:
      - 192.168.79.1

    on_time:
      - seconds: /10
        then:
          - http_request.get:
              url: !lambda |-

                return "https://httpbin.org/anything?x=" + std::to_string(id(global_counter)++);
              capture_response: true
              max_response_buffer_size: 4kB
              on_response:
                then:
                  - lambda: id(handle_response)->execute(body);

```

Same behavior:

```
[E][TEST:054]: got response for request #0
[E][TEST:054]: got response for request #1
[E][TEST:054]: got response for request #2
[E][TEST:054]: got response for request #3
[E][TEST:054]: got response for request #4
[E][TEST_AFTER_WAIT:073]: got response for request #4
[E][TEST_AFTER_WAIT:073]: got response for request #3
[E][TEST_AFTER_WAIT:073]: got response for request #3
[E][TEST_AFTER_WAIT:073]: got response for request #3
[E][TEST_AFTER_WAIT:073]: got response for request #3
```

I'll put together a PR with my suggestions above.

afflux, 2025-04-10:
still relevant

closed 2025-11-02 by bdraco