esphome/issues #6535: Can't build esp-idf 5.3.1 with IPv6 only (IPv4 turned off)

Gunni, 2024-12-10:
### The problem

When building with IPv4 disabled:

```
esp32:
  board: esp32-c6-devkitc-1
  flash_size: 4MB
  variant: esp32c6
  framework:
    type: esp-idf
    version: 5.3.1
    platform_version: 6.9.0
    sdkconfig_options:
      CONFIG_LWIP_IPV6: y
      CONFIG_LWIP_IPV4: n
      CONFIG_DISABLE_IPV4: y
```

Compilation fails. Compilation output:

<details>
```
INFO ESPHome 2024.11.3
INFO Reading configuration /config/esphome/c6-1.yaml...
WARNING The selected ESP-IDF framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
WARNING This board is unknown. Make sure the chosen chip component is correct.
WARNING The selected ESP-IDF framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
WARNING This board is unknown. Make sure the chosen chip component is correct.
WARNING GPIO9 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
WARNING GPIO15 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Generating C++ source...
INFO Updating https://github.com/espressif/esp-protocols.git@mdns-v1.3.2
INFO Compiling app...
Processing c6-1 (board: esp32-c6-devkitc-1; framework: espidf; platform: platformio/espressif32@6.9.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32C6 160MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.50301.0 (5.3.1) 
 - tool-cmake @ 3.16.4 
 - tool-ninja @ 1.7.1 
 - tool-riscv32-esp-elf-gdb @ 12.1.0+20221002 
 - tool-xtensa-esp-elf-gdb @ 12.1.0+20221002 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-riscv32-esp @ 13.2.0+20240530
Reading CMake configuration...
Generating assembly for certificate bundle...
No dependencies
Compiling .pioenvs/c6-1/src/esphome/components/api/api_connection.cpp.o
Compiling .pioenvs/c6-1/src/esphome/components/api/api_frame_helper.cpp.o
Compiling .pioenvs/c6-1/src/esphome/components/api/api_pb2.cpp.o
Compiling .pioenvs/c6-1/src/esphome/components/api/api_pb2_service.cpp.o
In file included from src/esphome/components/network/util.h:5,
                 from src/esphome/components/api/api_connection.cpp:6:
src/esphome/components/network/ip_address.h:59:23: error: expected ')' before '*' token
   59 |   IPAddress(ip4_addr_t *other_ip) {
      |            ~          ^~
      |                       )
src/esphome/components/network/ip_address.h:112:12: error: expected type-specifier before 'ip4_addr_t'
  112 |   operator ip4_addr_t() const { return *ip_2_ip4(&ip_addr_); }
      |            ^~~~~~~~~~
src/esphome/components/network/ip_address.h: In constructor 'esphome::network::IPAddress::IPAddress(uint8_t, uint8_t, uint8_t, uint8_t)':
src/esphome/components/network/ip_address.h:55:5: error: 'IP_ADDR4' was not declared in this scope; did you mean 'IP_ADDR6'?
   55 |     IP_ADDR4(&ip_addr_, first, second, third, fourth);
      |     ^~~~~~~~
      |     IP_ADDR6
src/esphome/components/network/ip_address.h: In constructor 'esphome::network::IPAddress::IPAddress(ip6_addr_t*)':
src/esphome/components/network/ip_address.h:71:14: error: 'ip_addr_t' {aka 'struct ip6_addr'} has no member named 'type'
   71 |     ip_addr_.type = IPADDR_TYPE_V6;
      |              ^~~~
src/esphome/components/network/ip_address.h: In constructor 'esphome::network::IPAddress::IPAddress(esp_ip6_addr_t*)':
src/esphome/components/network/ip_address.h:78:31: error: 'ip_addr_t' {aka 'struct ip6_addr'} has no member named 'u_addr'; did you mean 'addr'?
   78 |     memcpy((void *) &ip_addr_.u_addr.ip6, (void *) other_ip, sizeof(esp_ip6_addr_t));
      |                               ^~~~~~
      |                               addr
src/esphome/components/network/ip_address.h:79:14: error: 'ip_addr_t' {aka 'struct ip6_addr'} has no member named 'type'
   79 |     ip_addr_.type = IPADDR_TYPE_V6;
      |              ^~~~
src/esphome/components/network/ip_address.h: In member function 'esphome::network::IPAddress::operator esp_ip4_addr_t() const':
src/esphome/components/network/ip_address.h:102:46: error: 'const ip_addr_t' {aka 'const struct ip6_addr'} has no member named 'u_addr'; did you mean 'addr'?
  102 |     memcpy((void *) &tmp, (void *) &ip_addr_.u_addr.ip4, sizeof(esp_ip4_addr_t));
      |                                              ^~~~~~
      |                                              addr
src/esphome/components/network/ip_address.h: In member function 'esphome::network::IPAddress& esphome::network::IPAddress::operator+=(uint8_t)':
src/esphome/components/network/ip_address.h:128:29: error: 'ip_addr_t' {aka 'struct ip6_addr'} has no member named 'u_addr'; did you mean 'addr'?
  128 |       (((u8_t *) (&ip_addr_.u_addr.ip4))[3]) += increase;
      |                             ^~~~~~
      |                             addr
*** [.pioenvs/c6-1/src/esphome/components/api/api_connection.cpp.o] Error 1
========================= [FAILED] Took 20.39 seconds =========================
```
</details>

### Which version of ESPHome has the issue?

2024.11.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.12.1

### What platform are you using?

ESP32

### Board

ESP32-C6

### Component causing the issue

network, api

### YAML Config

```yaml
esphome:
  name: test

esp32:
  board: esp32dev
  framework:
    type: esp-idf
    version: 5.3.1
    platform_version: 6.9.0
    sdkconfig_options:
      CONFIG_LWIP_IPV4: n
      CONFIG_DISABLE_IPV4: y

network:
  enable_ipv6: True

api:
```
```

### Anything in the logs that might be useful for us?

```txt
Never get to the point to install it, the compilation fails.
```

### Additional information

I tried all sorts of different esp-idf versions, always kept getting this:

`UnknownPackageError: Could not find the package with 'platformio/framework-espidf @ ~3.50302.0' requirements for your system 'linux_x86_64'`

If I don't specify

```
version: 5.3.1
platform_version: 6.9.0
```

It builds successfully. But I cannot use that for my ESP32-C6. Right?

Gunni, 2025-04-13:
Not stale.

randybb, 2025-04-13:
@whatthebottleneck because 5.3.2 is not supported by platformio. Sinse esphome supports pioarduino, you can switch to that platform:
```yaml
    version: 5.3.2
    platform_version: 53.03.11
```

In any case I cannot reproduce your error, but on the other side wifi and api are still tight to ipv4 so compilation will fail as ipv4 is expected. And if I remember correctly mdns can use only ipv4 (or it was related to mdns forwarding on ubiquiti?).

Btw, issues should be created only for the default version, when overriding you are on your own. But it doesn't compile with the default one, so in this case it is a valid one, just that part is misleading. - in this case it should be closed as not valid, or rewritten with the right description.

whatthebottleneck, 2025-04-13:
I was able to compile against ESP-IDF 5.4.0 using PlatformIO 6.10.0, it resolved the issues I had with previous versions of ESP-IDF related to I2C. PlatformIO 6.10.0 was already supported in esphome, I simply had to include version 5.4.0 in https://github.com/esphome/esphome/blob/ca78dd44b5382187739ad837fdc47fa195fa88a0/esphome/components/esp32/__init__.py#L261 for ESP-IDF. Thanks for looking into this.