esphome/issues #6536: xtensa-lx106-elf-g++: fatal error:

Pop70, 2024-12-10:
### The problem

Again, the problem is like here: [https://github.com/esphome/issues/issues/3603#top](url).
Again, there is not enough memory on the raspberry pi 3 when compiling. The compile_process_limit: 1 option is used.

### Which version of ESPHome has the issue?

2024.11.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.12.0

### What platform are you using?

ESP8266

### Anything in the logs that might be useful for us?

```txt
INFO ESPHome 2024.11.3
INFO Reading configuration /config/esphome/Pol_v_gostinnoy_zona_1.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing pol-v-gostinnoy-zona-1 (board: esp12e; framework: arduino; platform: platformio/espressif8266@4.2.1)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
Dependency Graph
|-- ESPAsyncTCP-esphome @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 3.2.2
|-- DNSServer @ 1.1.1
|-- ESP8266WiFi @ 1.0
|-- ESP8266mDNS @ 1.2
Compiling .pioenvs/pol-v-gostinnoy-zona-1/src/main.cpp.o
xtensa-lx106-elf-g++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
*** [.pioenvs/pol-v-gostinnoy-zona-1/src/main.cpp.o] Error 1
========================= [FAILED] Took 17.56 seconds =========================
```

### Additional information

I haven't updated my devices for a long time. So, the problem, apparently, did not come with the latest release

randybb, 2024-12-10:
Restart your HA device and try again. And in meantime think about HW upgrade, or just use more powerful device for compilation.

ssieb, 2025-04-10:
You can't run esphome on a pi3.  Install it on a another computer.

closed 2025-04-10 by ssieb