esphome/issues #6539: Device no longer finds Home Assistant and regularly doesn't respond to pings

Rainz3, 2024-12-12:
### The problem

Starting a couple days ago I can no longer get my esp32-box3 to connect to the internet and find home assistant. It doesn't respond to pings as well. I've tried to reset the device as well reinstall the config several times. I've also placed the device near the router just incase it was a signal issue but it still will not connect. I've even set up a new wireless network and tried having it connect to that but same results.

Heres the log https://pastebin.com/aB0LHsC2

### Which version of ESPHome has the issue?

2024.11.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.12.2

### What platform are you using?

ESP32

### Board

esp32-s3-box-3b 

### YAML Config

```yaml
substitutions:
  name: esp32-s3-box-3-2e8adc
  friendly_name: Nabu
  micro_wake_word_model: okay_nabu
packages:
  esphome.voice-assistant: github://esphome/wake-word-voice-assistants/esp32-s3-box-3/esp32-s3-box-3.yaml@main
esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
api:
  encryption:
    key: pC2QdQHXGH9vuGwSlG6hIIRjjucqFjcZRf4kw0trwlU=

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

external_components:
  - source: github://sammcj/esphome-esp-s3-box-3-volume@main
    components: [esp_box_volume]
    refresh: always
  - source: github://pr#5230
    components: esp_adf

esp_box_volume:
  id: speaker_volume

number:
  - name: Volume
    id: volume
    platform: template
    entity_category: config
    min_value: 0
    max_value: 1.0
    step: 0.05
    update_interval: never
    optimistic: true
    restore_value: true
    initial_value: 0.85
    on_value:
      - lambda: id(speaker_volume).set_volume(x);
```

ssieb, 2024-12-12:
The device doesn't connect to HA, HA connects to the device.
Can you ping `esp32-s3-box-3-2e8adc.local` or `192.168.1.180`?

Rainz3, 2024-12-12:
pinging esp32-s3-box-3-2e8adc.local returns with unable to find host. Pinging the ip returns
```
Pinging 192.168.1.180 with 32 bytes of data:
Request timed out.
Request timed out.
Reply from 192.168.1.32: Destination host unreachable.
Reply from 192.168.1.32: Destination host unreachable.
```

closed 2025-04-26 by github-actions[bot]