esphome/issues #6541: Undefined reference to ADCSensor class

szupi-ipuzs, 2024-12-12:
### The problem

I have lamp that detects ambient light via adc. I've recently upgraded to the december beta and the linker now fails as in the log. Looks like the ADCSensor class implementation is not compiled-in.
This is on beken chip, using libretiny @ 1.7.0 (latest official).

### Which version of ESPHome has the issue?

2024.12.0b1

### What type of installation are you using?

Docker

### What platform are you using?

BK72XX

### Board

wb3s

### Component causing the issue

adc

### YAML Config

```yaml
packages:
  wifi: !include ../common/wifi.yaml
  device_base: !include ../common/device_base.yaml
  web_server: !include ../common/web_server.yaml

esphome:
  on_boot:
    then:
      lambda: !lambda |-
        id(output_pirsense).set_level(id(pirduty_pwm).state/100);

bk72xx:
  board: wb3s

logger:

button:
  - platform: template
    id: toggle_light
    name: "Toggle light"
    on_press:
      then:
        if:
          condition:
            - light.is_on: light_led
          then:
            - script.stop: waitThenDIM
            - script.stop: waitThenOff
            - light.turn_off: light_led
            - globals.set:
                id: manual_mode
                value: 'false'
          else:
            - globals.set:
                id: manual_mode
                value: 'true'
            - script.stop: waitThenDIM
            - script.stop: waitThenOff
            - light.turn_on: light_led

globals:
  - id: manual_mode
    type: bool
    restore_value: yes
    initial_value: 'false'
  - id: light_sensitivity
    type: int
    restore_value: no
    initial_value: '0'

script:
  - id: fullOnWhenMotion
    then:
      - logger.log:
          format: "fullOnWhenMotion"
      - light.turn_on:
          id: light_led
          brightness: 100%
  - id: waitThenDIM
    then:
      - logger.log:
          format: "waitThenDIM (%d, %d)"
          args: ["id(full_light_duration_s).state", "id(dim_light_percent).state"]
      - delay: !lambda "return id(full_light_duration_s).state*1000*1000;"
      - if:
          condition:
            - light.is_on: light_led
          then:
            - light.dim_relative:
                id: light_led
                relative_brightness: !lambda "return id(dim_light_percent).state/100;"
            - script.execute: waitThenOff
  - id: waitThenOff
    then:
      - logger.log:
          format: "waitThenOff (%d)"
          args: ["id(dim_light_duration_s).state"]
      - if:
          condition:
            - light.is_on: light_led
          then:
          - delay: !lambda "return id(dim_light_duration_s).state*1000*1000;"
          - light.turn_off: light_led

text_sensor:
  - platform: libretiny
    version:
      name: LibreTiny Version

sensor:
  - platform: adc
    id: ambient
    name: Ambient
    unit_of_measurement: "V"
    state_class: "measurement"
    update_interval: 5s
    pin: P23

binary_sensor:
  - platform: gpio
    id: pir_sensor
    name: Motion
    device_class: motion
    pin:
      number: P6
      mode:
        input: true
    on_press:
      then:
        - if:
            condition:
              - lambda: !lambda return (!id(manual_mode) && (id(ambient).state > id(light_sensitivity)));
            then:
              - script.stop: waitThenDIM
              - script.stop: fullOnWhenMotion
              - script.execute: fullOnWhenMotion
    on_release:
      then:
        - if:
            condition:
              - lambda: !lambda return (!id(manual_mode) && (id(light_led).current_values.is_on()));
            then:
              - logger.log:
                  format: "Executing on_release"
              - script.execute: waitThenDIM

output:
  - platform: libretiny_pwm
    id: output_pirsense
    pin: P26
    inverted: True
  - platform: libretiny_pwm
    id: out_warm
    pin: P9
  - platform: libretiny_pwm
    id: out_cold
    pin: P8

light:
  - platform: cwww
    id: light_led
    name: Light
    restore_mode: RESTORE_AND_OFF
    cold_white_color_temperature: 6500 K
    warm_white_color_temperature: 2700 K
    warm_white: out_warm
    cold_white: out_cold

number:
  - platform: template
    id: full_light_duration_s
    restore_value: yes
    name: Full light duration
    icon: "mdi:clock"
    entity_category: config
    unit_of_measurement: "s"
    min_value: 0
    max_value: 600
    initial_value: 10
    step: 1
    optimistic: true

  - platform: template
    id: dim_light_duration_s
    restore_value: yes
    name: Dim light duration
    icon: "mdi:clock"
    entity_category: config
    unit_of_measurement: "s"
    min_value: 0
    max_value: 60
    initial_value: 10
    step: 1
    optimistic: true

  - platform: template
    id: dim_light_percent
    restore_value: yes
    name: Dim light percent
    icon: "mdi:clock"
    entity_category: config
    unit_of_measurement: "%"
    min_value: 0
    max_value: 100
    initial_value: 50
    step: 1
    optimistic: true

  - platform: template
    id: pirduty_pwm
    restore_value: yes
    initial_value: 50
    name: PIR sensitivity
    icon: "mdi:chip"
    entity_category: config
    min_value: 0
    max_value: 100
    step: 10
    optimistic: true
    on_value:
      then:
        lambda: !lambda |-
          id(output_pirsense).set_level(x/100);

select:
  - platform: template
    id: light_sensitivity_setting
    name: "Ambient light trigger"
    entity_category: config
    restore_value: yes
    options:
     - "Day"
     - "Dusk"
     - "Evenfall"
     - "Evening"
     - "Night"
    initial_option: "Evenfall"
    optimistic: true
    set_action:
      - logger.log:
          format: "Chosen option: %s"
          args: ["x.c_str()"]
    on_value:
      then:
        lambda: !lambda |-
          if (i == 1)
          {
            id(light_sensitivity) = 0.157f;
          }
          else
          if (i == 2)
          {
            id(light_sensitivity) = 1.800f;
          }
          else
          if (i == 3)
          {
            id(light_sensitivity) = 2.100f;
          }
          else
          if (i == 4)
          {
            id(light_sensitivity) = 2.180f;
          }
          else id(light_sensitivity) = 0.0f;
```

### Anything in the logs that might be useful for us?

```txt
/config/.esphome/platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: .pioenvs/lsc-floodlight-1/src/esphome/components/adc/adc_sensor_common.cpp.o:(.rodata._ZTVN7esphome3adc9ADCSensorE+0x14): undefined reference to `esphome::adc::ADCSensor::setup()'
/config/.esphome/platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: .pioenvs/lsc-floodlight-1/src/esphome/components/adc/adc_sensor_common.cpp.o:(.rodata._ZTVN7esphome3adc9ADCSensorE+0x18): undefined reference to `esphome::adc::ADCSensor::dump_config()'
/config/.esphome/platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: .pioenvs/lsc-floodlight-1/src/esphome/components/adc/adc_sensor_common.cpp.o:(.rodata._ZTVN7esphome3adc9ADCSensorE+0x20): undefined reference to `esphome::adc::ADCSensor::sample()'
/config/.esphome/platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: .pioenvs/lsc-floodlight-1/src/esphome/components/adc/adc_sensor_common.cpp.o:(.rodata._ZTVN7esphome3adc9ADCSensorE+0x2c): undefined reference to `non-virtual thunk to esphome::adc::ADCSensor::setup()'
/config/.esphome/platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: .pioenvs/lsc-floodlight-1/src/esphome/components/adc/adc_sensor_common.cpp.o:(.rodata._ZTVN7esphome3adc9ADCSensorE+0x34): undefined reference to `non-virtual thunk to esphome::adc::ADCSensor::dump_config()'
/config/.esphome/platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: .pioenvs/lsc-floodlight-1/src/esphome/components/adc/adc_sensor_common.cpp.o:(.rodata._ZTVN7esphome3adc9ADCSensorE+0x70): undefined reference to `non-virtual thunk to esphome::adc::ADCSensor::sample()'
```

jesserockz, 2024-12-13:
@edwardtfn Looks like you dropped the libretiny code from #7940

https://github.com/esphome/esphome/pull/7940/files#diff-7a18cdb9eb3cf1884a3120ace98e4d0b7acf8dc767eea0d5322329736b2c5755L309-L325

edwardtfn, 2024-12-13:
Oh no!
I will look at that asap.

edwardtfn, 2024-12-13:
@szupi-ipuzs, I cannot test it with a real device, but at least I can compile it now with esphome/esphome#7955.
Could you please give it a try and let me know if it is working properly (not only compiling, but also measuring data properly)?
 ```yaml
external_components:
  - source: github://pr#7955
    refresh: 30s
    components:
      - adc
```

szupi-ipuzs, 2024-12-13:
@edwardtfn I confirm that this helped and the linker was able to build a valid image.
ADC values also seem to make sense. In my case their meaning is the amount of light (the darker it is, the higher the measured voltage) - and I think they match to what I observed before.

Thank you for this quick fix!

closed 2024-12-14 by jesserockz