esphome/issues #6542: Cannot use lambda in light "initial_state" statement

bennydiamond, 2024-12-12:
### The problem

Simplified example (tested) for illustration purpose.

```
light:
  - platform: monochromatic
    id: led_light
    output: led_pwm_output
    name: "White LED"
    default_transition_length: 0s
    initial_state: 
      state: !lambda "return true;"
```

Returns the following error when trying to generate:

> INFO ESPHome 2024.11.3
> INFO Reading configuration /config/esphome/switch-lumieres-portique.yaml...
> INFO Generating C++ source...
> Traceback (most recent call last):
>   File "/usr/local/bin/esphome", line 8, in <module>
>     sys.exit(main())
>              ^^^^^^
>   File "/esphome/esphome/__main__.py", line 1036, in main
>     return run_esphome(sys.argv)
>            ^^^^^^^^^^^^^^^^^^^^^
>   File "/esphome/esphome/__main__.py", line 1023, in run_esphome
>     rc = POST_CONFIG_ACTIONS[args.command](args, config)
>          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
>   File "/esphome/esphome/__main__.py", line 503, in command_run
>     exit_code = write_cpp(config)
>                 ^^^^^^^^^^^^^^^^^
>   File "/esphome/esphome/__main__.py", line 212, in write_cpp
>     generate_cpp_contents(config)
>   File "/esphome/esphome/__main__.py", line 224, in generate_cpp_contents
>     CORE.flush_tasks()
>   File "/esphome/esphome/core/__init__.py", line 684, in flush_tasks
>     self.event_loop.flush_tasks()
>   File "/esphome/esphome/coroutine.py", line 246, in flush_tasks
>     next(task.iterator)
>   File "/esphome/esphome/__main__.py", line 204, in wrapped
>     await coro(conf)
>   File "/esphome/esphome/components/monochromatic/light.py", line 21, in to_code
>     await light.register_light(var, config)
>   File "/esphome/esphome/components/light/__init__.py", line 223, in register_light
>     await setup_light_core_(light_var, output_var, config)
>   File "/esphome/esphome/components/light/__init__.py", line 164, in setup_light_core_
>     initial_state = LightStateRTCState(
>                     ^^^^^^^^^^^^^^^^^^^
>   File "/esphome/esphome/cpp_generator.py", line 762, in __call__
>     call = CallExpression(self.base, *args)
>            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
>   File "/esphome/esphome/cpp_generator.py", line 131, in __init__
>     self.args = ExpressionList(*args)
>                 ^^^^^^^^^^^^^^^^^^^^^
>   File "/esphome/esphome/cpp_generator.py", line 98, in __init__
>     self.args = [safe_exp(arg) for arg in args]
>                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
>   File "/esphome/esphome/cpp_generator.py", line 98, in <listcomp>
>     self.args = [safe_exp(arg) for arg in args]
>                  ^^^^^^^^^^^^^
>   File "/esphome/esphome/cpp_generator.py", line 381, in safe_exp
>     raise ValueError("Object is not an expression", obj)
> ValueError: ('Object is not an expression', Lambda<return true;>)

It might be me but 'initial_state' in light component is templatable according to documentation.
https://esphome.io/components/light/#base-light-configuration

### Which version of ESPHome has the issue?

2024.11.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.12

### What platform are you using?

BK72XX

### Board

CB2S

### Component causing the issue

Monochromatic Light

### YAML Config

```yaml
output:
  - platform: libretiny_pwm
    id: led_pwm_output
    pin:
      number: P24
      inverted: True
    frequency: 1000 Hz
    min_power: 0.13
    max_power: 1.0
    zero_means_zero: True

light:
  - platform: monochromatic
    id: led_light
    output: led_pwm_output
    name: "White LED"
    default_transition_length: 0s
    initial_state: 
      state: !lambda "return true;"
```

git-atlanta, 2025-03-21:
same in 2025.2.0

ssieb, 2025-04-17:
This is a side effect of trying to share code.  You can't use lambdas in initial configuration.  The fix here will be to somehow adjust the validation to make it not accept lambdas.

closed 2026-06-06 by github-actions[bot]