esphome/issues #6543: [Internal Temperature Sensor] Invalid temperatures (53.3°C // 128) are reported

gbInArY, 2024-12-12:
### The problem

I am trying to get the internal temperature of my ESP32 board which contains an ESP32-D0WD-V3 chip using the _Internal Temperature Sensor_ component but I only get "Ignoring invalid temperature (success=0, value=53.3)".

Am I missing something on ESPHome? or is there some issue related?

Digging a little deeper, I looked into the internal code of the _Internal Temperature Sensor_ component and, for this ESP32 variant, it uses _temprature_sens_read()_ call. If I make the same call using Arduino IDE, it works fine.

[image]

### Which version of ESPHome has the issue?

2024.11.3

### What type of installation are you using?

Docker

### What platform are you using?

ESP32

### Board

WeMos D1 Mini ESP32 WiFi + Bluetooth

### Component causing the issue

Internal Temperature Sensor

### YAML Config

```yaml
esphome:
    name: test-internal-temp-sensor
    friendly_name: Test Internal Temperature Sensor

esp32:
    board: mhetesp32minikit
    framework:
        type: arduino

logger:

sensor:
    - platform: internal_temperature
      name: Internal Temperature Sensor
      update_interval: 10s
```

### Anything in the logs that might be useful for us?

```txt
[21:30:10]ets Jul 29 2019 12:21:46
[21:30:10]
[21:30:10]rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[21:30:10]configsip: 0, SPIWP:0xee
[21:30:10]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[21:30:10]mode:DIO, clock div:2
[21:30:10]load:0x3fff0030,len:1184
[21:30:10]load:0x40078000,len:13132
[21:30:10]load:0x40080400,len:3036
[21:30:10]entry 0x400805e4
[21:30:10][I][logger:171]: Log initialized
[21:30:10][I][app:029]: Running through setup()...
[21:30:10][I][app:062]: setup() finished successfully!
[21:30:10][D][internal_temperature:092]: Ignoring invalid temperature (success=0, value=53.3)
[21:30:10][D][sensor:094]: 'Internal Temperature Sensor': Sending state nan °C with 1 decimals of accuracy
[21:30:10][I][app:100]: ESPHome version 2024.11.3 compiled on Dec 12 2024, 21:26:06
[21:30:10][C][logger:185]: Logger:
[... 226 lines cut ...]
[21:57:27][D][internal_temperature:092]: Ignoring invalid temperature (success=0, value=53.3)
[21:57:37][D][sensor:094]: 'Internal Temperature Sensor': Sending state 38.33334 °C with 1 decimals of accuracy
[21:57:47][D][internal_temperature:092]: Ignoring invalid temperature (success=0, value=53.3)
[21:57:57][D][internal_temperature:092]: Ignoring invalid temperature (success=0, value=53.3)
[21:58:07][D][sensor:094]: 'Internal Temperature Sensor': Sending state 38.33334 °C with 1 decimals of accuracy
[21:58:17][D][internal_temperature:092]: Ignoring invalid temperature (success=0, value=53.3)
[21:58:27][D][internal_temperature:092]: Ignoring invalid temperature (success=0, value=53.3)
[21:58:37][D][sensor:094]: 'Internal Temperature Sensor': Sending state 38.33334 °C with 1 decimals of accuracy
[21:58:47][D][internal_temperature:092]: Ignoring invalid temperature (success=0, value=53.3)
[21:58:57][D][internal_temperature:092]: Ignoring invalid temperature (success=0, value=53.3)
[21:59:07][D][internal_temperature:092]: Ignoring invalid temperature (success=0, value=53.3)
[21:59:17][D][internal_temperature:092]: Ignoring invalid temperature (success=0, value=53.3)

[...]

[22:02:47][D][internal_temperature:092]: Ignoring invalid temperature (success=0, value=53.3)
[22:02:57][D][internal_temperature:092]: Ignoring invalid temperature (success=0, value=53.3)
[22:03:07][D][internal_temperature:092]: Ignoring invalid temperature (success=0, value=53.3)
[22:03:17][D][internal_temperature:092]: Ignoring invalid temperature (success=0, value=53.3)
[22:03:27][D][sensor:094]: 'Internal Temperature Sensor': Sending state 37.22222 °C with 1 decimals of accuracy
[22:03:37][D][internal_temperature:092]: Ignoring invalid temperature (success=0, value=53.3)
[22:03:47][D][internal_temperature:092]: Ignoring invalid temperature (success=0, value=53.3)
[22:03:57][D][sensor:094]: 'Internal Temperature Sensor': Sending state 36.66667 °C with 1 decimals of accuracy
[22:04:07][D][internal_temperature:092]: Ignoring invalid temperature (success=0, value=53.3)
```

ssieb, 2024-12-12:
In the arduino test, you didn't have it connected to wifi.

gbInArY, 2024-12-12:
Correct, the "same" as the initial yml example

gbInArY, 2024-12-13:
> In the arduino test, you didn't have it connected to wifi.

Here you have a working example on Arduino where Wifi is connected.

[image]

clydebarrow, 2024-12-16:
The internal temperature sensor is apparently buggy (in the hardware) and frequently reports an ADC value of 128, which corresponds to 53.3C. The ESPHome component filters out those spurious values as mentioned in the docs.  Your Arduino sketch is simply reporting those unreliable values.

ssieb, 2024-12-16:
@clydebarrow that's not it.  The Arduino code is not printing that value, so it's getting the correct value.  Compare the working with non-working output.

clydebarrow, 2024-12-16:
> The Arduino code is not printing that value

Well, I can see it in the screenshot so I would say it is. 128F == 53.3C

[image]

closed 2024-12-17 by ssieb

gbInArY, 2024-12-17:
@ssieb Well, actually here there are 2 parts:
- The temperature sensor does not work on ESPHome when the wifi is not activated, but it does work fine on Arduino. **This is the original issue opened.**

- On the other hand, I noticed that when the Wifi is enabled, the temperature sensor doesn't work well on both ESPHome and Arduino.