esphome/issues #6545: slow_pwm output can toggle more often than specified period

pdw-mb, 2024-12-13:
### The problem

Slow PWM [recalculates its duty cycle every loop](https://github.com/esphome/esphome/blob/dev/esphome/components/slow_pwm/slow_pwm_output.cpp#L43). This means that if the input changes mid-cycle, the output can change. 

If you have an input that updates much more often than the slow PWM period, and the input is a bit noisy, the output can toggle several times each period in response to input changes.

In my case, I had a PID input that was updating every 5s and a PWM period of 5 minutes. When the system was in steady state, I would frequently get the output switching a few times on each PWM cycle due to noise on the temperature sensor. I do already have some smoothing on the PID output, but it seems that this isn't sufficient.

Obviously this should be easy to address with a filter on the sensor, and on reflection, I think the current behaviour is potentially useful, as it allows the system to respond quickly to big changes (such as turning the heating on and off), but it wasn't obvious to me from the documentation.

In fact, the documentation for the [sigma delta output](https://esphome.io/components/output/sigma_delta_output.html) seems to suggest that it won't do this:

> Unlike with [Slow PWM Output](https://esphome.io/components/output/slow_pwm), it is possible to update the output value with each update cycle, not just at the end of a longer period.

Perhaps the documentation for Slow PWM could be expanded with an additional note saying something like:

> Note
>
> The duty cycle will be recalculated whenever the input changes, and the output will be updated to reflect that. This can lead to the output toggling more than once per period particularly if the input is noisy, and has an update interval much shorter than the PWM period. It may be necessary to reduce the input period, or use filters to rate limit and/or smooth the input on order to avoid unwanted toggling of the output.

### Which version of ESPHome has the issue?

2024.6.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.11.6

### What platform are you using?

ESP32

### YAML Config

```yaml
output:
  - platform: slow_pwm
    pin: GPIO26
    id: "heater_pwm"
    period: 300s

sensor:
  # Shelly add-on, uses duty-cycle to perform ADC input
  - platform: duty_cycle
    pin: 3
    id: add_on_adc
    name: "Add-on ADC"
    update_interval: 5s
    unit_of_measurement: "V"
    filters:
      - offset: -2.0
      - multiply: 0.033

  - platform: resistance
    id: resistance_sensor
    sensor: add_on_adc
    configuration: DOWNSTREAM
    resistor: 10kOhm
    name: Resistance Sensor

  - platform: ntc
    sensor: resistance_sensor
    calibration:
      - 11.4kOhm -> 26.6°C
      - 13.17kOhm -> 24°C
      - 16.17kOhm -> 18.8°C
    name: Floor Temperature
    id: floor_temperature

climate:
  - platform: pid
    id: pid_climate
    name: "Floor temperature"
    sensor: floor_temperature
    default_target_temperature: 16°C
    heat_output: heater_pwm
    control_parameters:
      kp: 1.04410
      ki: 0.00194
      kd: 140.43016
      output_averaging_samples: 5      # smooth the output over 5 samples
      derivative_averaging_samples: 10  # smooth the derivative value over 10 samples
    deadband_parameters:
      threshold_high: 0.5°C       # deadband within +/-0.5°C of target_temperature
      threshold_low: -0.5°C
```

lutorm, 2025-05-10:
I agree this seems undesired. I would expect a PWM to change at most twice per period.

goloveychuk, 2025-12-29:
Hey @bdraco 

Sorry for pinging. 
Please confirm it’s a bug, I’ll submit a pr