esphome/issues #6548: Unexpected exception while validating config

michnixweiss, 2024-12-13:
### The problem

I am trying to build a wall panel according to https://github.com/albert-canfield/HA-panel-esphome
But i am receiving errors when validating the config.
Tried from ha addon, also dev and beta, then installed esphome with pip in venv on my laptop with the same result.
I can't assign the errors, so not sure if it is a problem of my config, the compiler or what else.

### Which version of ESPHome has the issue?

2024.11.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.12.3

### What platform are you using?

ESP32

### Board

ESP32-4848S040

### Component causing the issue

I poorly dont know

### YAML Config

```yaml
substitutions:
  lightbulb: "\U000F1802"
  boiler_on: "\U000F0238"
  boiler_off: "\U000F1722"
  radiator: "\U000F0438"
  ceiling_light: "\U000F0769"
  house_temp: "\U000F0F55"
  #weather icons
  cloudy: "\U000F0590"
  cloudyalert: "\U000F0F2F"
  cloudyarrowright: "\U000F0E6E"
  fog: "\U000F0591"
  hail: "\U000F0592"
  hazy: "\U000F0F30"
  hurricane: "\U000F0898"
  lightning: "\U000F0593"
  lightningrainy: "\U000F067E"
  night: "\U000F0594"
  nightpartlycloudy: "\U000F0F31"
  partlycloudy: "\U000F0595"
  partlylightning: "\U000F0F32"
  partlyrainy: "\U000F0F33"
  partlysnowy: "\U000F0F34"
  partlysnowyrainy: "\U000F0F35"
  pouring: "\U000F0596"
  rainy: "\U000F0597"
  snowy: "\U000F0598"
  snowyheavy: "\U000F0F36"
  snowyrainy: "\U000F067F"
  sunny: "\U000F0599"
  sunnyalert: "\U000F0F37"
  sunnyoff: "\U000F14E4"
  sunset: "\U000F059A"
  sunsetdown: "\U000F059B"
  sunsetup: "\U000F059C"
  tornado: "\U000F0F38"
[... 1161 lines cut ...]
          ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/voluptuous/schema_builder.py", line 281, in __call__
    return self._compiled([], data)
           ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/voluptuous/validators.py", line 251, in _run
    return self._exec(self._compiled, value, path)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/voluptuous/validators.py", line 379, in _exec
    v = func(path, v)
        ^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/voluptuous/schema_builder.py", line 855, in validate_callable
    return schema(data)
           ^^^^^^^^^^^^
  File "/esphome/esphome/components/font/__init__.py", line 156, in validate_glyphs
    check_missing_glyphs(extra[CONF_FILE][CONF_PATH], points)
  File "/esphome/esphome/components/font/__init__.py", line 99, in check_missing_glyphs
    font = FONT_CACHE[file]
           ~~~~~~~~~~^^^^^^
  File "/esphome/esphome/components/font/__init__.py", line 56, in __missing__
    res = self[key] = freetype.Face(key)
                      ^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/freetype/__init__.py", line 1192, in __init__
    raise FT_Exception(error)
freetype.ft_errors.FT_Exception: FT_Exception:  (unknown file format)
```

randybb, 2024-12-13:
Remove the external component, as that PR has been merged and a loot of LVGL fixes applied. So it is not compatible with any newer version. There were some changes, so open esphome docs and use it (+ there is a nice LVGL cookbook) instead of outdated configs.

michnixweiss, 2024-12-14:
@randybb thanks for the hint.

clydebarrow, 2024-12-16:
Looks like a corrupted font file.

closed 2024-12-17 by jesserockz