esphome/issues #6549: stale mode value obtained when climate.on_control trigger used to assess climate modes.

bradkeifer, 2024-12-14:
### The problem

I am using the `climate.on_control` trigger to call a script. This script checks the modes of all climate controllers I have defined on my esphome device in order to determine if a switch (for a pump set) should be on or off.
What I am observing, is the mode value being reported for the climate controller that is having it's mode value changed is the old mode value and not the new mode value.
If I use the `climate.on_state` trigger to call the same script, then this problem does not exist. However, using the `climate.on_state` trigger is not an ideal solution because it is triggered more frequently than is necessary for what I am seeking to achieve.

### Which version of ESPHome has the issue?

2024.11.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.12.2

### What platform are you using?

ESP32-IDF

### Board

esp32dev / Kincony KC868-A8

### Component causing the issue

climate

### YAML Config

```yaml
# PID Climate Controllers
climate:
  - platform: pid
    id: "zone_1_controller"
    name: "Zone 1"
    sensor: ds18b20_1
    default_target_temperature: 20°C
    heat_output: zone_1_heat
    control_parameters: 
      kp: 0.49460
      ki: 0.00487
      kd: 12.56301
      output_averaging_samples: 5     # smooth the output over 5 samples
      derivative_averaging_samples: 5 # smooth the derivative value over 5 samples
    deadband_parameters: 
      threshold_high: 0.3°C           # deadband within +/-0.3°C of target_temperature
      threshold_low: -0.3°C
    on_control: 
      then:
        - logger.log: "Zone 1 control input updated!"
        - script.execute: pump_set_control

  - platform: pid
    name: "Zone 2"
    id: zone_2_controller
    sensor: shelly_wall_temp
    default_target_temperature: 21°C
    heat_output: zone_2_heat
    control_parameters: 
      kp: 0.49460
      ki: 0.00487
      kd: 12.56301
      output_averaging_samples: 5     # smooth the output over 5 samples
      derivative_averaging_samples: 5 # smooth the derivative value over 5 samples
    deadband_parameters: 
      threshold_high: 0.3°C           # deadband within +/-0.3°C of target_temperature
      threshold_low: -0.3°C
    on_state: 
      then:
        - logger.log: "Zone 2 state updated!"
        - script.execute: pump_set_control

  - platform: pid
    name: "Zone 3"
    id: zone_3_controller
    sensor: shelly_bht_temp
    default_target_temperature: 26°C
    heat_output: zone_3_heat
    control_parameters: 
      kp: 0.49460
      ki: 0.00487
      kd: 12.56301
      output_averaging_samples: 5     # smooth the output over 5 samples
      derivative_averaging_samples: 5 # smooth the derivative value over 5 samples
    deadband_parameters: 
      threshold_high: 0.3°C           # deadband within +/-0.3°C of target_temperature
      threshold_low: -0.3°C
    on_state: 
      then:
        - logger.log: "Zone 3 state updated!"
        - script.execute: pump_set_control

# Scripts
script:
  - id: pump_set_control
    mode: queued
    then:
      lambda: |-
        ESP_LOGD("main", 
          "pump_set_control script called.\n\t"
          "Zone 1 mode %s\n\t"
          "Zone 2 mode %s\n\t"
          "Zone 3 mode %s",
          (char *)climate_mode_to_string(id(zone_1_controller).mode),
          (char *)climate_mode_to_string(id(zone_2_controller).mode),
          (char *)climate_mode_to_string(id(zone_3_controller).mode));
        if (id(zone_1_controller).mode == CLIMATE_MODE_HEAT ||
            id(zone_2_controller).mode == CLIMATE_MODE_HEAT ||
            id(zone_3_controller).mode == CLIMATE_MODE_HEAT) 
        {
          ESP_LOGI("main", "One or more climate controllers are in %s mode. Pump set must be on.", 
            (char *)climate_mode_to_string(CLIMATE_MODE_HEAT));
          id(pump_set).turn_on();
        }
        else 
        {
          ESP_LOGI("main", "No climate controllers are in %s mode. Pump set must be off.",
            (char *)climate_mode_to_string(CLIMATE_MODE_HEAT));
          id(pump_set).turn_off();
        }
```

### Additional information

From a quick inspection of the source code, I can see that the functions used by the `climate.on_state` and `climate.on_control` callbacks are slightly different. I'm not sure if this helps in finding the root cause of the problem?
```
void Climate::add_on_state_callback(std::function<void(Climate &)> &&callback) {
  this->state_callback_.add(std::move(callback));
}

void Climate::add_on_control_callback(std::function<void(ClimateCall &)> &&callback) {
  this->control_callback_.add(std::move(callback));
}
```
I also see that there is conditional compilation being used when the state is being saved that is dependent on the use of the esp-idf framework (amongst other things). I need to use the esp-idf framework because I am also using the `bluetooth_proxy` component in my overall solution. Again, perhaps this helps to isolate the cause of the issue.
```
void Climate::save_state_() {
#if (defined(USE_ESP_IDF) || (defined(USE_ESP8266) && USE_ARDUINO_VERSION_CODE >= VERSION_CODE(3, 0, 0))) && \
    !defined(CLANG_TIDY)
#pragma GCC diagnostic ignored "-Wclass-memaccess"
#define TEMP_IGNORE_MEMACCESS
#endif
  ClimateDeviceRestoreState state{};
  // initialize as zero to prevent random data on stack triggering erase
  memset(&state, 0, sizeof(ClimateDeviceRestoreState));
#ifdef TEMP_IGNORE_MEMACCESS
#pragma GCC diagnostic pop
#undef TEMP_IGNORE_MEMACCESS
#endif
```

ssieb, 2024-12-14:
`on_control` is called before the state is set, to give you a chance to modify the `ClimateCall` before it is used.  You can look in `x` to see what is getting set.

closed 2024-12-14 by ssieb

bradkeifer, 2024-12-14:
OK, thanks @ssieb. This behaviour is not clear in the documentation. Can I suggest that this is made clear in the documentation please?

ssieb, 2024-12-14:
It kind of does say that, but you can make a PR to make it more clear if you want.

bradkeifer, 2024-12-14:
I don't understand how it says that at all. It says that that the trigger is activated each time a control input of the climate device is updated. It doesn't say that it is activated before the control input is updated.

I'm not sure it's reasonable to expect me to make the documentation changes, because I don't know the code very well and am not an expert in what it is mean't to do. I think the people who write the code, or make changes to the code, are the ones who should be responsible for clear documentation, not the end users. If I was proposing a change to the code, then fair enough, I should also propose a change to the documentation.

ssieb, 2024-12-14:
> It will be invoked prior to the on_state trigger, if both are defined. The ClimateCall control object is available to automations as the reference x that can be changed.
Clearly there must be a difference between `on_control` and `on_state`, so that should at least be a clue to look closer.
Then it says that the control object is provided and that you can change it, which indicates that it must not be applied yet.

bradkeifer, 2024-12-14:
Ok, thanks.