esphome/issues #6550: Incorrect pixel format by canned init_sequence for st7701s

simonmicro, 2024-12-14:
### The problem

Uploading a simple hello-world, with white text on **black** background, yields in a **blue** background and white text.

### Which version of ESPHome has the issue?

2024.11.3

### What type of installation are you using?

Docker

### What platform are you using?

ESP32-IDF

### Board

esp32-s3-devkitc-1

### Component causing the issue

display/st7701s

### YAML Config

```yaml
esp32:
  board: esp32-s3-devkitc-1
  flash_size: 16MB
  variant: esp32s3
  framework:
    type: esp-idf

esphome:
  name: wall-switch-01
  platformio_options:
    board_build.flash_mode: dio # default is qio, which is wrong

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

ota:
  platform: esphome
  password: "XXX"

web_server:
  include_internal: true
  port: 80
  auth:
    username: !secret web_user
    password: !secret web_password

logger:
  level: debug
  baud_rate: 115200

# ===================================================================================

api:
  encryption:
    key: "XXX"

debug:
  update_interval: 5s

# text_sensor:
#   - <<: !include .global_text_sensor.yml

# sensor:
#   - <<: !include .global_sensor.yml

psram:
  mode: octal
  speed: 80MHz

output:
  - platform: ledc
    frequency: 600
    pin: GPIO38
    id: gfx_bl

light:
  - platform: monochromatic
    name: backlight
    output: gfx_bl
    restore_mode: ALWAYS_ON

font:
  - file: "gfonts://Roboto"
    id: roboto_20
    size: 20

spi:
  clk_pin: GPIO48
  mosi_pin: GPIO47

display:
  - platform: st7701s
    auto_clear_enabled: false
    update_interval: 60s # we are mostly static right now
    color_order: BGR
    dimensions:
      width: 480
      height: 480
    hsync_front_porch: 10
    hsync_pulse_width: 8
    hsync_back_porch: 50
    vsync_front_porch: 10
    vsync_pulse_width: 8
    vsync_back_porch: 20
    cs_pin: GPIO39
    de_pin: GPIO18
    hsync_pin: GPIO16
    vsync_pin: GPIO17
    pclk_pin: GPIO21

    data_rate: 80MHz
    pclk_frequency: 8MHz # ok, should be default, but then does not work?

    init_sequence:
      # https://focuslcds.com/wp-content/uploads/Drivers/ST7701.pdf
      # -> https://github.com/arendst/Tasmota/discussions/20527
      # -> https://esphome.io/components/display/st7701s.html
      # -> https://github.com/esphome/esphome/blob/dev/esphome/components/st7701s/init_sequences.py
      # [command, data1, data2, data3, ...]

      - [0x01] # software reset, just to make sure everything is on defaults, #1 also does that but i want to be sure
      - delay 10ms
      - 1 # -> this is somewhat similar to st7701_type1_init_operations from Arduino_GFX
      - delay 5ms
      - [0x3A, 0x50] # set to RGB565, as #1 does not set this

    data_pins:
      red:
        - 11        #r1
        - 12        #r2
        - 13        #r3
        - 14        #r4
        - 00        #r5
      green:
        - 8        #g0
        - 20        #g1
        - 3        #g2
        - 46        #g3
        - 9        #g4
        - 10       #g5
      blue:
        - 4        #b1
        - 5        #b2
        - 6        #b3
        - 7        #b4
        - 15        #b5
    lambda: |-
      it.fill(Color::BLACK);
      it.print(0, 0, id(roboto_20), Color::WHITE, TextAlign::TOP_LEFT, "Hello World!");
```

### Additional information

The issue is part of...

https://github.com/esphome/esphome/blob/7f2ca800c1312f8b864f17949ab35339a55ac4f6/esphome/components/st7701s/init_sequences.py#L1C1-L1C93

It seems the author was unable to consult the datasheet to actually verify what these magic numbers mean :wink: - have a look at https://focuslcds.com/wp-content/uploads/Drivers/ST7701.pdf

So, the last command sets the color mode to RGB666 (https://github.com/esphome/esphome/blob/7f2ca800c1312f8b864f17949ab35339a55ac4f6/esphome/components/st7701s/init_sequences.py#L137), which does not match the default color mode of the graphics engine, which seems to operate on RGB565. If you set the correct value `0x50` in your own `init_sequence`, you actually get the correct colors :grinning:

clydebarrow, 2024-12-16:
For that particular board, use this in your display config:

```
    init_sequence:
      - 1
      - [0x3A, 0x60] # 18 bit pixel format
      - [0xFF, 0x77, 0x01, 0x00, 0x00, 0x10] # CMD2_BKSEL_BK0
      - [0xCD, 0x00] # disable MDT flag
   
```

The correct init sequence depends on much more than just the controller chip - it varies with the board design ( e.g. depending on what pins on the st7701s are wired either 16 bit or 18 bit mode may be required) and the LCD characteristics. Many controller chips also seem to have undocumented registers that require modification for specific LCDs, so the datasheet is just a starting point. The canned sequence is also just a starting point, which happens to be sufficient for some boards.

closed 2024-12-16 by clydebarrow

clydebarrow, 2024-12-16:
Here is a complete YAML.

[wall-switch.yaml.txt](https://github.com/user-attachments/files/18144992/wall-switch.yaml.txt)

simonmicro, 2024-12-23:
@clydebarrow Thank you very much. I thought I catched the underlying issue, but it seems I missed the specific pin configuration. Thanks!

simonmicro, 2024-12-23:
@clydebarrow

...but I still think there is something missing here. I obtained the original firmware sources and was able to determine that the screen was being switched to RGB565 before use. According to your changes, this is no needed, which is strange as the datasheet describes the default as `70h` for register `3A`, so 24-bit pixel format, which would mismatch the RGB565s 16-bit color format of the UI by the ESP.

Also taking a look at table _8.6.1_ it seems you are configuring the pinout incorrectly (18-bit, MDT 0, the default after reset is already `00` - but _only_ on paper), not matching the 16-bit color mode physically connected here (based on the pins documented: 5xR, 6xG, 5xB)?

clydebarrow, 2024-12-23:
Without a schematic you have no idea how the bus is configured. Just use what works.

simonmicro, 2024-12-23:
Hmm, the vendor only provided this [4.0 inches IO pin distribution.xlsx](https://github.com/user-attachments/files/18228160/4.0.inches.IO.pin.distribution.xlsx) as well as the sources for known-good applications. Otherwise, see https://homeding.github.io/boards/esp32s3/panel-4848S040.htm