esphome/issues #6551: Hardware SPI not working on Raspberry Pi Pico W

torndar, 2024-12-14:
### The problem

I have an ILI9xxx [1] display connected and the test screen loads but painfully slowly (multiple seconds) and I saw in the log output that it is using software SPI. I'm happy to help test any changes.

```
[08:59:21][C][spi:064]: SPI bus:
[08:59:21][C][spi:065]:   CLK Pin: GPIO2
[08:59:21][C][spi:066]:   SDI Pin: GPIO3
[08:59:21][C][spi:067]:   SDO Pin: GPIO4
[08:59:21][C][spi:074]:   Using software SPI
```

[1] https://esphome.io/components/display/ili9xxx

### Which version of ESPHome has the issue?

2024.11.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.12.3

### What platform are you using?

RP2040

### Board

rpipicow

### Component causing the issue

spi

### YAML Config

```yaml
esphome:
  name: bedroomenvsensor
  friendly_name: BedroomEnvSensor

rp2040:
  board: rpipicow

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: ""

ota:
  - platform: esphome
    password: ""

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot in case wifi connection fails
  ap:
    ssid: "Bedroomenvsensor"
    password: ""

output:
  - platform: gpio
    pin:
      number: LED
      mode: output
    id: LED

#interval:
#  - interval: 500ms
#    then:
#      - output.turn_on: LED
#      - delay: 250ms
#      - output.turn_off: LED

i2c:
  # "raw" pins not physical pins https://datasheets.raspberrypi.com/picow/PicoW-A4-Pinout.pdf
  sda: GPIO20 
  scl: GPIO21

sensor:
  - platform: bme280_i2c
    temperature:
      name: "BME280 Temperature"
    pressure:
      name: "BME280 Pressure"
    humidity:
      name: "BME280 Humidity"
    address: 0x76

spi:
  - miso_pin: GPIO3
    mosi_pin: GPIO4
    clk_pin: GPIO2
    #interface: hardware

display:
  - platform: ili9xxx
    model: ili9341
    dc_pin: GPIO06
    reset_pin: GPIO05
    invert_colors: false
    show_test_card: true
```

clydebarrow, 2024-12-16:
> [08:59:21][C][spi:065]:   CLK Pin: GPIO2
> [08:59:21][C][spi:066]:   SDI Pin: GPIO3
> [08:59:21][C][spi:067]:   SDO Pin: GPIO4

That pin configuration is not supported by the RP2040. If you swap the SDI and SDO pins it will work. Add `interface: hardware` to the spi config to ensure that you do get hardware

closed 2024-12-16 by clydebarrow

torndar, 2024-12-31:
I finally got around to updating my config but hardware SPI seems to boot loop

If I flip the SPI config to the snippet below and flip the wires around to match it works and shows the test card fine.

```
    miso_pin: GPIO3
    mosi_pin: GPIO4
```

## CONFIG:
```
esphome:
  name: bedroomenvsensor
  friendly_name: BedroomEnvSensor

safe_mode:
  num_attempts: 3

rp2040:
  board: rpipicow

# Enable logging
logger:
  level: "VERBOSE"

# Enable Home Assistant API
api:
  encryption:
    key: "XXXXXX"

ota:
  - platform: esphome
    password: "XXXXXX"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot in case wifi connection fails
  ap:
    ssid: "Bedroomenvsensor"
    password: "XXXXXX"

output:
  - platform: gpio
    pin:
      number: LED
      mode: output
    id: LED

#interval:
#  - interval: 500ms
#    then:
#      - output.turn_on: LED
#      - delay: 250ms
#      - output.turn_off: LED

i2c:
  # "raw" pins not physical pins https://datasheets.raspberrypi.com/picow/PicoW-A4-Pinout.pdf
  sda: GPIO20 
  scl: GPIO21

sensor:
  - platform: bme280_i2c
    temperature:
      id: temperature
      name: "BME280 Temperature"
    pressure:
      name: "BME280 Pressure"
    humidity:
[... 226 lines cut ...]

[V][mdns:118]:   Services:

[V][mdns:120]:   - _http, _tcp, 80

[V][mdns:122]:     TXT: version = 2024.11.3

[C][esphome.ota:073]: Over-The-Air updates:

[C][esphome.ota:074]:   Address: bedroomenvsensor.local:2040

[C][esphome.ota:075]:   Version: 2

[C][esphome.ota:078]:   Password configured

[C][safe_mode:018]: Safe Mode:

[C][safe_mode:019]:   Boot considered successful after 60 seconds

[C][safe_mode:021]:   Invoke after 3 boot attempts

[C][safe_mode:022]:   Remain in safe mode for 300 seconds

[W][safe_mode:031]: SAFE MODE IS ACTIVE
```