esphome/issues #6553: try to allocate too many ble clients

tomaszduda23, 2024-12-14:
### The problem

esphome tries to allocate 4 ble devices and last one fails. It seems that it should be:
2 connections for ble proxy
1 connetion for ble client

```
[V][esp32_ble_client:121]: [0] [] gattc_event_handler: event=0 gattc_if=4
[V][esp32_ble_client:127]: [0] [] gattc registered app id 1
[V][esp32_ble:367]: (BLE) gattc_event [esp_gatt_if: 5] - 0
[V][esp32_ble_client:121]: [1] [] gattc_event_handler: event=0 gattc_if=5
[V][esp32_ble_client:127]: [1] [] gattc registered app id 2
[V][esp32_ble:367]: (BLE) gattc_event [esp_gatt_if: 6] - 0
[V][esp32_ble_client:121]: [2] [] gattc_event_handler: event=0 gattc_if=6
[V][esp32_ble_client:127]: [2] [] gattc registered app id 3
[V][esp32_ble:367]: (BLE) gattc_event [esp_gatt_if: 0] - 0
[V][esp32_ble_client:121]: [3] [] gattc_event_handler: event=0 gattc_if=0
[E][esp32_ble_client:131]: [3] [] gattc app registration failed id=4 code=128
```

### Which version of ESPHome has the issue?

2024.9.2

### What type of installation are you using?

Docker

### What platform are you using?

ESP32

### YAML Config

```yaml
esp32:
  board: esp32doit-devkit-v1
esp32_ble:
  io_capability: keyboard_only

esp32_ble_tracker:
  scan_parameters:
    interval: 1100ms
    window: 1100ms
    active: true

bluetooth_proxy:
  active: true  <- this makes esphome to tires to allocate 4 connections

ble_client:
- mac_address: "yyy"
  id: pvvx_ble_display
  on_passkey_request:
    then:
      - logger.log: "Authenticating with passkey"
      - ble_client.passkey_reply:
          id: pvvx_ble_display
          passkey: xyz

display:
- platform: pvvx_mithermometer
  ble_client_id: pvvx_ble_display
  update_interval: 24h
  validity_period: 0s
  disconnect_delay: 30s
  time_id: homeassistant_time
  id: display_bathroom
```

bdraco, 2025-02-23:
@tomaszduda23 Is this solved by one of the mentioned PRs?

tomaszduda23, 2025-02-23:
Nope. There is error message now https://github.com/esphome/esphome/blob/bfa3254d6c5ab2f555196a3b4f8e245800d85724/esphome/components/esp32_ble_client/ble_client_base.cpp#L85. It still over allocate clients.

bdraco, 2025-02-23:
Does the problem happen if you remove `ble_client:` and only use the proxy ?

tomaszduda23, 2025-02-23:
It works correctly if ble_client is removed. It is hardcoded here https://github.com/esphome/esphome/blob/bfa3254d6c5ab2f555196a3b4f8e245800d85724/esphome/components/bluetooth_proxy/__init__.py#L13. It would be nice to verify it during validation of config.

bdraco, 2025-02-23:
We can also increase CONFIG_BTDM_CTRL_BLE_MAX_CONN
Up to 9

Mine runs just fine with 5

But 1KB of ram used per connection slot

tomaszduda23, 2025-02-23:
Mine works on arduino. You cannot change it there.
I've tried to run `esp-idf` some time ago. There were some problem. I rollbacked to arduino.

bdraco, 2025-02-23:
@tomaszduda23 Can you give https://github.com/esphome/esphome/pull/8303 a try?

closed 2025-04-07 by bdraco