esphome/issues #6554: udp component not broadcasting in wifi:AP mode

ruok911, 2024-12-15:
### The problem

upd component works fine in station mode, but once wifi is in AP mode, no broadcast is sent.  Checked logs (in Verbose mode) and no meaningful information is shown (no error messages are logged)
udp component looks like it's being setuped correctly.

Checked network with Wireshark and no traces of a udp packet sent.
web_server component functions fine without any issues in AP mode.

### Which version of ESPHome has the issue?

2024.11.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.12.3

### What platform are you using?

ESP32

### Board

nodemcu

### Component causing the issue

udp

### YAML Config

```yaml
esphome:
  name: esp32-solarcharge
  friendly_name: ESP32 Solar Charge

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:
  level: VERBOSE
  logs:
  #  udp: DEBUG
    sensor: INFO
    json: INFO

    
# Over the Air Update. 
ota:
  - platform: esphome
    password: =======================================

# Enable Web Server.  This way, we can access the status from a cell phone (through the ap) to see values
web_server:
  port: 80
  local: True
  version: 3
  ota: True
  #include_internal: True
  sorting_groups:
    - id: sort_battery
      name: "Battery Data"
      sorting_weight: -5
    - id: sort_load
[... 1279 lines cut ...]
[V][esp32.preferences:126]: sync: key: 1141067765, len: 4
[D][esp32.preferences:143]: Saving 4 preferences to flash: 2 cached, 2 written, 0 failed
[D][esp32.preferences:114]: Saving 4 preferences to flash...
[V][esp32.preferences:126]: sync: key: 2800489596, len: 4
[V][esp32.preferences:126]: sync: key: 1144802869, len: 4
[V][esp32.preferences:126]: sync: key: 533881015, len: 4
[V][esp32.preferences:126]: sync: key: 1141067765, len: 4
[D][esp32.preferences:143]: Saving 4 preferences to flash: 0 cached, 4 written, 0 failed
[D][esp32.preferences:114]: Saving 4 preferences to flash...
[V][esp32.preferences:137]: NVS data not changed skipping 2800489596  len=4
[V][esp32.preferences:137]: NVS data not changed skipping 1144802869  len=4
[V][esp32.preferences:126]: sync: key: 533881015, len: 4
[V][esp32.preferences:126]: sync: key: 1141067765, len: 4
[D][esp32.preferences:143]: Saving 4 preferences to flash: 2 cached, 2 written, 0 failed
[D][esp32.preferences:114]: Saving 4 preferences to flash...
[V][esp32.preferences:137]: NVS data not changed skipping 2800489596  len=4
[V][esp32.preferences:137]: NVS data not changed skipping 1144802869  len=4
[V][esp32.preferences:126]: sync: key: 533881015, len: 4
[V][esp32.preferences:126]: sync: key: 1141067765, len: 4
[D][esp32.preferences:143]: Saving 4 preferences to flash: 2 cached, 2 written, 0 failed
```

### Additional information

Will try a simpler setup with VERY_VERBOSE later to see if it is still an issue (but udp component works fine in STATION mode).

ruok911, 2024-12-15:
Tried a new setup...  Simplified...  Still no result in Wireshark.  No UDP broadcast.

========================================================================
yaml file:

esphome:
  name: esp32-udp-test
  friendly_name: esp32-udp-test

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:
  level: VERY_VERBOSE

wifi:
  #ssid: !secret wifi_ssid
  #password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esp32-Udp-Test Fallback Hotspot"
    password: "91aKPra3SjzZ"

udp:
  - id: udp_pack1
    update_interval: 60s
    encryption: "TestEncrypt"
    sensors:
      - var_display_uptime

sensor:
  - platform: uptime
    id: var_display_uptime
    name: "Display Uptime Sensor"
    update_interval: 5s

==========================================================================

Serial UART Logs (VERY_VERBOSE):

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13132
load:0x40080400,len:3036
entry 0x400805e4
[     6][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
[I][logger:171]: Log initialized
[I][app:029]: Running through setup()...
[V][app:030]: Sorting components by setup priority...
[VV][scheduler:063]: set_interval(name='', interval=60000, offset=2000)
[VV][scheduler:063]: set_interval(name='update', interval=5000, offset=2044)
[C][wifi:048]: Setting up WiFi...
[C][wifi:061]: Starting WiFi...
[C][wifi:062]:   Local MAC: 58:BF:25:8C:1E:B8
[V][esp32.preferences:059]: nvs_get_blob('88491487'): ESP_ERR_NVS_NOT_FOUND - the key might not be set yet
[V][wifi_esp32:077]: Enabling AP.
[    86][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 0 - WIFI_READY
[V][wifi_esp32:499][arduino_events]: Event: WiFi ready
[C][wifi:263]: Setting up AP...
[C][wifi:265]:   AP SSID: 'Esp32-Udp-Test Fallback Hotspot'
[C][wifi:266]:   AP Password: '91aKPra3SjzZ'
[   183][V][WiFiGeneric.cpp:390] _arduino_event_cb(): AP Started
[   193][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 10 - AP_START
[... 283 lines cut ...]
[V][sensor:043]: 'Display Uptime Sensor': Received new state 357.997009
[D][sensor:094]: 'Display Uptime Sensor': Sending state 357.99701 s with 0 decimals of accuracy
[VV][scheduler:226]: Running interval '' with interval=60000 last_execution=298034 (now=358035)
[VV][scheduler:226]: Running interval 'update' with interval=5000 last_execution=357991 (now=362991)
[V][sensor:043]: 'Display Uptime Sensor': Received new state 362.990997
[D][sensor:094]: 'Display Uptime Sensor': Sending state 362.99100 s with 0 decimals of accuracy
[VV][scheduler:226]: Running interval 'update' with interval=5000 last_execution=362991 (now=367997)
[V][sensor:043]: 'Display Uptime Sensor': Received new state 367.997009
[D][sensor:094]: 'Display Uptime Sensor': Sending state 367.99701 s with 0 decimals of accuracy
[VV][scheduler:226]: Running interval 'update' with interval=5000 last_execution=367991 (now=372991)
[V][sensor:043]: 'Display Uptime Sensor': Received new state 372.990997
[D][sensor:094]: 'Display Uptime Sensor': Sending state 372.99100 s with 0 decimals of accuracy
[VV][scheduler:226]: Running interval 'update' with interval=5000 last_execution=372991 (now=377997)
[V][sensor:043]: 'Display Uptime Sensor': Received new state 377.997009
[D][sensor:094]: 'Display Uptime Sensor': Sending state 377.99701 s with 0 decimals of accuracy
[VV][scheduler:226]: Running interval 'update' with interval=5000 last_execution=377991 (now=382991)
[V][sensor:043]: 'Display Uptime Sensor': Received new state 382.990997
[D][sensor:094]: 'Display Uptime Sensor': Sending state 382.99100 s with 0 decimals of accuracy
[VV][scheduler:226]: Running interval 'update' with interval=5000 last_execution=382991 (now=387997)
[V][sensor:043]: 'Display Uptime Sensor': Received new state 387.997009
[D][sensor:094]: 'Display Uptime Sensor': Sending state 387.99701 s with 0 decimals of accuracy
[VV][scheduler:226]: Running interval 'update' with interval=60000 last_execution=332724 (now=392731)
[VV][scheduler:226]: Running interval 'update' with interval=5000 last_execution=387991 (now=392991)
[V][sensor:043]: 'Display Uptime Sensor': Received new state 392.990997
[D][sensor:094]: 'Display Uptime Sensor': Sending state 392.99100 s with 0 decimals of accuracy

jesserockz, 2024-12-15:
This is sort of by design. The udp component checks `network::is_connected()` before even attempting to send a packet. The return value of this is only true when connected to a station and it is used in many places in the codebase for this exact purpose (i.e that part wont change).

ESPHome was not designed to work in AP mode "permanently" and so these use cases were not taken into consideration. If you want to, you can add it as a feature-request, but it is not a bug (at this time).

Jesse