esphome/issues #6556: Component ILI9xxx/ILI9341 bootloop and panic

pmatil, 2024-12-15:
### The problem

I've been testing my ILI9341 display and ran into this issue. The issue is quite serious as it causes bootloop and eventually safe mode as it panics. My test code is only the one pasted below in addition to the skeleton sketch that ESPHome generates.

### Which version of ESPHome has the issue?

2024.11.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.12.3

### What platform are you using?

ESP32

### Board

az-delivery-devkit-v4 (AZ-Delivery DevKit C v4)

### Component causing the issue

ILI9xxx

### YAML Config

```yaml
spi:
  clk_pin: GPIO18
  mosi_pin: GPIO23

display:
  - platform: ili9xxx
    model: ili9341
    dc_pin: GPIO27
    reset_pin: GPIO13
    invert_colors: false
    show_test_card: true
```

### Anything in the logs that might be useful for us?

```txt
E (441) psram: PSRAM ID read error: 0xffffffff
[I][app:029]: Running through setup()...
[D][spi:035]: Setting up SPI bus...
[D][ili9xxx:034]: Setting up ILI9xxx
[D][spi_device:362]: mode 0, data_rate 40000kHz
[D][ili9xxx:030]: Wrote MADCTL 0x48
[C][esp32_ble:032]: Setting up BLE...
[E][display:017]: Could not allocate buffer for display!
[W][component:237]: Component display took a long time for an operation (183 ms).
[W][component:238]: Components should block for at most 30 ms.
[W][component:157]: Component wifi set Warning flag: scanning for networks
[D][esp32_ble:284]: Enabling BLE...
Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x40091d98  PS      : 0x00060430  A0      : 0x80151223  A1      : 0x3fffab50
A2      : 0x00000000  A3      : 0x00000000  A4      : 0x00000130  A5      : 0x00000000
A6      : 0x3fff8884  A7      : 0x00000013  A8      : 0x80158a2c  A9      : 0x3fffab20
A10     : 0x3ffc8630  A11     : 0x00001110  A12     : 0x3ffc862c  A13     : 0x3ffc3bf4
A14     : 0x00002aa6  A15     : 0x00000000  SAR     : 0x0000001e  EXCCAUSE: 0x0000001d
EXCVADDR: 0x00000000  LBEG    : 0x40091d98  LEND    : 0x40091da3  LCOUNT  : 0x00000012

Backtrace:0x40091d95:0x3fffab500x40151220:0x3fffab60 0x4013309d:0x3fffab80 0x4014a493:0x3fffaba0`
```

clydebarrow, 2024-12-16:
That board isn't particularly suitable for driving the ili9341 as it does not have PSRAM. It is possible, but you will want 

> My test code is only the one pasted below in addition to the skeleton sketch that ESPHome generates.

Not according to the logs. The wizard doesn't add BLE but your logs show it as being enabled. Remove the BLE, remove `captive_portal`. The ESPHome display functions need a full size buffer which is a problem without PSRAM, so you will be better off using LVGL:

```
lvgl:
  buffer_size: 25%
```

In the display config set `update_interval: never` and `auto_clear_enable: false` and remove the `show_test_card` line.

closed 2024-12-16 by clydebarrow

clydebarrow, 2024-12-16:
> The wizard doesn't add any BLE config but apparently BLE is enabled by default

It's not AFAIK and certainly not when I test it. Without seeing your complete YAML I can't comment further. When raising an issue it's better to attach the full YAML rather than making assumptions about what is or isn't important.

> Not really sure why the current version of ili9xxx panics the entire ESPHome

It doesn't, nor is PSRAM a requirement, it's just that without it there is a shortage of RAM to allocate the buffer. As you can see from the log you posted the display *is* working

```
[W][component:237]: Component display took a long time for an operation (183 ms)
```
That's caused by the display drawing the test card in 8 bit mode.

The crash occurs after it tries to enable BLE. Probably by this time the RAM is exhausted and the BLE component doesn't handle this gracefully, but the display component is behaving as expected.

pmatil, 2024-12-17:
The wizard actually adds this for me:

# Sets up Bluetooth LE (Only on ESP32) to allow the user
# to provision wifi credentials to the device.
esp32_improv:
  authorizer: none

So that is what enables BLE. It is not uncommented by me, esphome-web added that. Somehow I missed that statement before. Complete YAML edited in the post above since code here will break the formatting.

clydebarrow, 2024-12-17:
Oh, you used web.esphome.io? I have no idea what the point of that is since it doesn't do anything other than flash a fixed config, and it certainly didn't configure the display. But yes, it does add BLE via the `improv` component.

clydebarrow, 2024-12-17:
For future reference, the crash appears to be a bug in the Arduino BLE stack. Switching the framework type to `esp-idf` results in a graceful handling of the out-of-memory error, so no bootloop.

numo68, 2025-01-08:
@clydebarrow out of curiosity, why no way to force - once default - 8-bit mode? The

```
  if (this->buffer_color_mode_ == BITS_16) {
    this->init_internal_(this->get_buffer_length_() * 2);
    if (this->buffer_ != nullptr) {
      return;
    }
    this->buffer_color_mode_ = BITS_8;
  }
```

might be a good heuristics, but even if one has the PSRAM, there still there might be people needing the last bit of memory for own usage and are willing to sacrifice the performance/color quality in order to get extra 75K. Also, it relies on the assumption that the allocation never succeeds if there is no PSRAM, which is highly probable, but I am not sure that it is guaranteed (generally there is a 320*240*2-sized region in the ESP32's memory map).

The documentation also states "Use of 16 bit colors requires double the amount of RAM as 8 bit, and _may need_ PSRAM to be available" but does not explain how to influence the mode.

Unfortunately the PSRAM is still a pretty rare beast in the boards one is able to easily buy at least here in Europe :(

clydebarrow, 2025-01-08:
I've no idea why it was done the way it is. You can force 8 bit mode using the adaptive image palette mode.

Using the display without PSRAM (or technically without enough RAM of any kind for a full buffer) is usually no problem with LVGL and a 25% buffer size.