esphome/issues #6558: ESPhome will not install on clean Python 3.13.1 on windows

Floriszz, 2024-12-15:
### The problem

When installing ESPhome on a clean Windows under clean Python 3.13.1 install, the install fails on building Wheel.
When I install ESPhome the same way under Python 3.12.8, the installation is successful.

Btw. I install Python under c;\Python , for all users with Elevated rights. (so not the default user directory location)

The full install log from start to end and showing the versions of the components installed is depicted here below
Failing install with Python 3.13.1 (below is successful install under Python 3.12.8)
```
Microsoft Windows [Version 10.0.19045.5247]
(c) Microsoft Corporation. All rights reserved.

C:\WINDOWS\system32>python --version
Python 3.13.1

C:\WINDOWS\system32>pip3 install wheel
Collecting wheel
  Downloading wheel-0.45.1-py3-none-any.whl.metadata (2.3 kB)
Downloading wheel-0.45.1-py3-none-any.whl (72 kB)
Installing collected packages: wheel
Successfully installed wheel-0.45.1

C:\WINDOWS\system32>pip list
Package Version
------- -------
pip     24.3.1
wheel   0.45.1

C:\WINDOWS\system32>pip3 install esphome
Collecting esphome
  Downloading esphome-2024.11.3-py3-none-any.whl.metadata (3.5 kB)
Collecting cryptography==43.0.0 (from esphome)
  Downloading cryptography-43.0.0-cp39-abi3-win_amd64.whl.metadata (5.4 kB)
Collecting voluptuous==0.14.2 (from esphome)
  Downloading voluptuous-0.14.2-py3-none-any.whl.metadata (20 kB)
Collecting PyYAML==6.0.2 (from esphome)
  Downloading PyYAML-6.0.2-cp313-cp313-win_amd64.whl.metadata (2.1 kB)
Collecting paho-mqtt==1.6.1 (from esphome)
  Downloading paho-mqtt-1.6.1.tar.gz (99 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Collecting colorama==0.4.6 (from esphome)
  Downloading colorama-0.4.6-py2.py3-none-any.whl.metadata (17 kB)
Collecting icmplib==3.0.4 (from esphome)
  Downloading icmplib-3.0.4-py3-none-any.whl.metadata (19 kB)
Collecting tornado==6.4 (from esphome)
  Downloading tornado-6.4-cp38-abi3-win_amd64.whl.metadata (2.6 kB)
Collecting tzlocal==5.2 (from esphome)
  Downloading tzlocal-5.2-py3-none-any.whl.metadata (7.8 kB)
Collecting tzdata>=2021.1 (from esphome)
  Downloading tzdata-2024.2-py2.py3-none-any.whl.metadata (1.4 kB)
Collecting pyserial==3.5 (from esphome)
  Downloading pyserial-3.5-py2.py3-none-any.whl.metadata (1.6 kB)
Collecting platformio==6.1.16 (from esphome)
  Downloading platformio-6.1.16.tar.gz (239 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Collecting esptool==4.7.0 (from esphome)
  Downloading esptool-4.7.0.tar.gz (285 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Collecting click==8.1.7 (from esphome)
  Downloading click-8.1.7-py3-none-any.whl.metadata (3.0 kB)
Collecting esphome-dashboard==20241120.0 (from esphome)
  Downloading esphome_dashboard-20241120.0-py3-none-any.whl.metadata (1.0 kB)
Collecting aioesphomeapi==24.6.2 (from esphome)
[... 483 lines cut ...]
  Stored in directory: c:\users\floris\appdata\local\pip\cache\wheels\90\00\90\e060c90e673ce4a40bf6f90fade521db90376fafcb0da83bb7
  Building wheel for paho-mqtt (pyproject.toml) ... done
  Created wheel for paho-mqtt: filename=paho_mqtt-1.6.1-py3-none-any.whl size=65645 sha256=a2ecc8c172634b0e7fe19add64c24cbd3bd0dd9eb81b532a8fa75aa7e44ebe04
  Stored in directory: c:\users\floris\appdata\local\pip\cache\wheels\23\d5\af\1f3cbcc350dec9d8e95040f388e0163d132eff0c9a453db659
  Building wheel for platformio (pyproject.toml) ... done
  Created wheel for platformio: filename=platformio-6.1.16-py3-none-any.whl size=419976 sha256=c7e53577752d9b4f2069a80692e8171cf9cb8dff2216cd7db995f9c8622a6dbc
  Stored in directory: c:\users\floris\appdata\local\pip\cache\wheels\a5\0c\1f\d7c6093aa64a15e2c4ebb12b6bdc6af94a031e3ab8b6e6ef24
Successfully built aioesphomeapi esptool paho-mqtt platformio
Installing collected packages: unicodedata2, reedsolo, pyserial, pyelftools, puremagic, paho-mqtt, intelhex, ifaddr, esphome-dashboard, bottle, bitarray, appdirs, zeroconf, voluptuous, urllib3, tzdata, typing_extensions, tornado, tabulate, sniffio, six, setuptools, semantic_version, ruamel.yaml.clib, PyYAML, pyparsing, pycparser, protobuf, pillow, packaging, openstep-plist, kconfiglib, idna, icmplib, h11, freetype-py, FontTools, colorama, charset-normalizer, certifi, bitstring, attrs, async-interrupt, argcomplete, ajsonrpc, aiohappyeyeballs, wsproto, tzlocal, ruamel.yaml, requests, marshmallow, gflanguages, fs, ecdsa, click, cffi, anyio, uvicorn, starlette, cryptography, ufoLib2, platformio, noiseprotocol, esptool, defcon, chacha20poly1305-reuseable, glyphsLib, aioesphomeapi, glyphsets, esphome
Successfully installed FontTools-4.55.3 PyYAML-6.0.2 aioesphomeapi-24.6.2 aiohappyeyeballs-2.4.4 ajsonrpc-1.2.0 anyio-4.7.0 appdirs-1.4.4 argcomplete-3.5.2 async-interrupt-1.2.0 attrs-24.2.0 bitarray-2.9.3 bitstring-4.2.3 bottle-0.13.2 certifi-2024.12.14 cffi-1.17.1 chacha20poly1305-reuseable-0.13.2 charset-normalizer-3.4.0 click-8.1.7 colorama-0.4.6 cryptography-43.0.0 defcon-0.10.3 ecdsa-0.19.0 esphome-2024.11.3 esphome-dashboard-20241120.0 esptool-4.7.0 freetype-py-2.5.1 fs-2.4.16 gflanguages-0.7.1 glyphsLib-6.9.5 glyphsets-1.0.0 h11-0.14.0 icmplib-3.0.4 idna-3.10 ifaddr-0.2.0 intelhex-2.3.0 kconfiglib-13.7.1 marshmallow-3.23.1 noiseprotocol-0.3.1 openstep-plist-0.4.0 packaging-24.2 paho-mqtt-1.6.1 pillow-10.4.0 platformio-6.1.16 protobuf-3.20.3 puremagic-1.27 pycparser-2.22 pyelftools-0.31 pyparsing-3.2.0 pyserial-3.5 reedsolo-1.7.0 requests-2.32.3 ruamel.yaml-0.18.6 ruamel.yaml.clib-0.2.12 semantic_version-2.10.0 setuptools-75.6.0 six-1.17.0 sniffio-1.3.1 starlette-0.39.2 tabulate-0.9.0 tornado-6.4 typing_extensions-4.12.2 tzdata-2024.2 tzlocal-5.2 ufoLib2-0.17.0 unicodedata2-15.1.0 urllib3-2.2.3 uvicorn-0.30.6 voluptuous-0.14.2 wsproto-1.2.0 zeroconf-0.132.2

C:\WINDOWS\system32>
```

### Which version of ESPHome has the issue?

2024.11.3

### What type of installation are you using?

pip

### What platform are you using?

ESP32

randybb, 2024-12-15:
And when you install what is asked of you?
_error: Microsoft Visual C++ 14.0 or greater is required. Get it with "Microsoft C++ Build Tools": https://visualstudio.microsoft.com/visual-cpp-build-tools/_

Floriszz, 2024-12-16:
I had a look at the release changes for Python 3.13.1(compared with 3.12) before, but could not find that "Microsoft Visual C++ 14.0 or greater" is a new requirement;
[https://docs.python.org/3/whatsnew/3.13.html](https://docs.python.org/3/whatsnew/3.13.html)
As you can see in the install logs above, with 3.12 this is not a problem.

randybb, 2024-12-16:
wheel needs that, just read the logs....

Floriszz, 2024-12-16:
See the logs of the install under 3.12. The last lines of my post. Building wheel works perfectly. I don't see "Microsoft Visual C++ 14.0 or greater" being installed on the system.

randybb, 2024-12-16:
the issue is about 3.13, isn't it? and this requirement mentioned in that log

closed 2024-12-16 by ssieb

tobiloeb, 2024-12-22:
Same here, installing c++ build tools didn't helped for me.
Thanks for the hint with python version 3.12, that worked.

MichaelEFlip, 2024-12-28:
Same issue here. Python 3.12 helped

Arnaud1977, 2025-01-05:
Same issue here.

ssieb, 2025-01-06:
@frspp this issue is about installing on Windows, not Mac.  You need to follow the instructions in the error message.

ssieb, 2025-01-06:
I'm running esphome just fine on Linux with 3.13.  Did you even try the instructions from the message?

Floriszz, 2025-01-06:
The issue is on Windows. On Linux is a different world.😜👍

ssieb, 2025-02-14:
> Linux with Python 3.13.1 here, failing as well.

I'm using esphome on Linux with Python 3.13.1 and no issues.  Come ask on discord if you need help.