esphome/issues #6560: [modbus_controller] `offline_skip_updates` works only once when a server goes `offline`

Sleeper85, 2024-12-15:
### The problem

Hi @0x3333

When a server goes `offline`, with `offline_skip_updates` set to `5`, 5 requests are skipped and then an indefinite number of requests are sent to the server regardless of the `offline_skip_updates` setting.

I would expect that after skipping 5 updates, one request would be sent (actually 4 with **max_cmd_retries** set to 3) and then 5 requests would be skipped again if the server continues to not respond so as not to pollute the bus.

The RS485 bus:
```
R120ohm---(server 2)-----(server 1)-----(client)-----(server 3)-----(server 241)---R120ohm
```

Attached are the modbus logs.

[logs_register_V2_device_3_offline_2024.12.15.txt](https://github.com/user-attachments/files/18142329/logs_register_V2_device_3_offline_2024.12.15.txt)

Thank you for your help and the time you spend improving this component.

### Which version of ESPHome has the issue?

2024.11.3

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32-IDF

### Board

esp32doit-devkit-v1

### Component causing the issue

modbus_controller

### YAML Config

```yaml
modbus:
  - uart_id: uart_esp_1
    id: modbus_client
    send_wait_time: 250ms

modbus_controller:
  - id: modbus_controller
    modbus_id: modbus_client
    setup_priority: -10
    address: 3
    update_interval: 5s
    command_throttle: 0ms
    max_cmd_retries: 3
    offline_skip_updates: 5
```

### Anything in the logs that might be useful for us?

```txt
[23:00:53][W][modbus_controller:027]: Modbus device=3 set offline
[23:00:56][V][modbus_controller:199]: Range : 1 Size: 20 (4) skip: 5
[23:01:01][V][modbus_controller:199]: Range : 1 Size: 20 (4) skip: 4
[23:01:06][V][modbus_controller:199]: Range : 1 Size: 20 (4) skip: 3
[23:01:11][V][modbus_controller:199]: Range : 1 Size: 20 (4) skip: 2
[23:01:16][V][modbus_controller:199]: Range : 1 Size: 20 (4) skip: 1
[23:01:21][V][modbus_controller:199]: Range : 1 Size: 20 (4) skip: 0
[23:01:26][V][modbus_controller:199]: Range : 1 Size: 20 (4) skip: 0
[23:01:31][V][modbus_controller:199]: Range : 1 Size: 20 (4) skip: 0
[23:01:36][V][modbus_controller:199]: Range : 1 Size: 20 (4) skip: 0
[23:01:41][V][modbus_controller:199]: Range : 1 Size: 20 (4) skip: 0
...
[23:10:41][V][modbus_controller:199]: Range : 1 Size: 20 (4) skip: 0
```

Anry777, 2025-01-24:
i have the same issue.  Skip works only once

exciton, 2025-06-25:
My PR doesn't specifically change the behaviour of skip_updates, but it significantly helps avoid errors on the bus which may cause the problem in the first place. If you have many devices and see CRC errors or timeout errors then my PR will probably fix it.

exciton, 2025-12-15:
PR #11781 should fix this behaviour properly. Please test and provide feedback if you get a chance.