esphome/issues #6561: LilyGo T-Display S3 AMOLED Touch version broken with change of component from qspi_amoled to qspi_dbi

razorback16, 2024-12-16:
### The problem

I had the following configuration with esphome v2024.10.3
```yaml

esphome:
  name: cgm-display-amoled
  friendly_name: CGM Display Amoled
  platformio_options:
    build_unflags: -Werror=all
    board_build.flash_mode: dio

esp32:
  board: esp32-s3-devkitc-1
  variant: esp32s3
  framework:
    type: esp-idf

spi:
  id: quad_spi
  type: quad
  clk_pin: 47
  data_pins: [18, 7, 48, 5]

i2c:
  sda: 3
  scl: 2

touchscreen:
  - platform: cst816
    id: my_touchscreen
    interrupt_pin: 
      number: 21
    transform:
      mirror_x: true
      swap_xy: false      
    on_touch:
      - lambda: |-
          ESP_LOGI("cal", "x=%d, y=%d, x_raw=%d, y_raw=%d",
            touch.x,
            touch.y,
            touch.x_raw,
            touch.y_raw
          );
      - logger.log:
          format: Touch %d at (%d, %d)
          args: [touch.id, touch.x, touch.y]
      - display.page.show_next: my_display
 
display:
  - platform: qspi_amoled
    id: my_display
    model: RM67162
    data_rate: 80MHz
    dimensions:
      height: 240
      width: 536
    transform:
      mirror_x: true
      mirror_y: false
      swap_xy: true
    color_order: rgb
    brightness: 255
    cs_pin: 6
    reset_pin: 17
    enable_pin: 38
```

When I moved to to v2024.11.3, I was told to change `qspi_amoled` to `qspi_dbi` when compiling, but after I did, I see that the text appear once for a moment, and then the text area becomes blurred, reverting back to esphome v2024.10.3 makes it work again.

### Which version of ESPHome has the issue?

v2024.11.3

### What type of installation are you using?

pip

### What platform are you using?

ESP32

### Board

LilyGo T-Display S3 AMOLED Touch

### Component causing the issue

qspi_dbi

razorback16, 2024-12-17:
Here is the full code that doesn't work
```yaml
substitutions:
  disp_name: CGM Display Amoled

esphome:
  name: cgm-display-amoled
  friendly_name: CGM Display Amoled
  platformio_options:
    build_unflags: -Werror=all
    board_build.flash_mode: dio

esp32:
  board: esp32-s3-devkitc-1
  variant: esp32s3
  framework:
    type: esp-idf

logger:
  level: info

api:

ota:
  platform: esphome

web_server:
  port: 80

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  ap:
    ssid: "CDM Display Amoled"

spi:
  id: quad_spi
  type: quad
  clk_pin: 47
  data_pins: [18, 7, 48, 5]

i2c:
  sda: 3
  scl: 2

touchscreen:
  - platform: cst816
    id: my_touchscreen
    interrupt_pin: 
      number: 21
    transform:
      mirror_x: true
      swap_xy: false      
    on_touch:
      - lambda: |-
          ESP_LOGI("cal", "x=%d, y=%d, x_raw=%d, y_raw=%d",
            touch.x,
            touch.y,
            touch.x_raw,
            touch.y_raw
          );
      - logger.log:
          format: Touch %d at (%d, %d)
          args: [touch.id, touch.x, touch.y]
      - display.page.show_next: my_display

color:
  - id: my_black
    red: 0%
    green: 0%
    blue: 0%
  - id: my_red
    hex: fc0000
  - id: my_green
    hex: 00fc4c
  - id: my_yellow
    hex: e0e637
  - id: my_orange
    hex: ff7728
  - id: my_white
    hex: ffffff

font:
  - file: "gfonts://Arimo"
    id: font_arimo
    size: 144

graph:
  - id: blood_glucose_graph
    duration: 5min
    x_grid: 5s
    y_grid: 50.0
    max_value: 400
    min_value: 50 
    width: 500
    height: 200
    traces:
      - sensor: blood_sugar
        line_type: SOLID
        continuous: true
        line_thickness: 3
        color: my_green

display:
  - platform: qspi_dbi
    id: my_display
    model: RM67162
    data_rate: 80MHz
    dimensions:
      height: 240
      width: 536
    transform:
      mirror_x: true
      mirror_y: false
      swap_xy: true
    color_order: rgb
    brightness: 255
    cs_pin: 6
    reset_pin: 17
    enable_pin: 38
    pages:
      - id: page1
        lambda: |-          
          if (id(blood_sugar).has_state()) {
            if (id(blood_sugar).state < 55.0) {
              it.printf(268, 100, id(font_arimo), id(my_red), TextAlign::CENTER, "%.0f", id(blood_sugar).state);
            }
            if ((id(blood_sugar).state >= 55.0) && (id(blood_sugar).state < 70.0)) {
              it.printf(268, 100, id(font_arimo), id(my_yellow), TextAlign::CENTER, "%.0f", id(blood_sugar).state);
            }
            if ((id(blood_sugar).state >= 70.0) && (id(blood_sugar).state < 180.0)) {
              it.printf(268, 100, id(font_arimo), id(my_green), TextAlign::CENTER, "%.0f", id(blood_sugar).state);
            }
            if ((id(blood_sugar).state >= 180.0) && (id(blood_sugar).state < 250.0)) {
              it.printf(268, 100, id(font_arimo), id(my_yellow), TextAlign::CENTER, "%.0f", id(blood_sugar).state);
            }
            if (id(blood_sugar).state >= 250.0) {
              it.printf(268, 100, id(font_arimo), id(my_orange), TextAlign::CENTER, "%.0f", id(blood_sugar).state);
            }
          }
      - id: page2
        lambda: |-
          it.fill(my_black);  // Clear before drawing
          it.graph(10, 20, id(blood_glucose_graph), my_yellow);

psram:
  mode: octal

switch:
  - platform: restart
    name: ${disp_name} Restart  

sensor:
  - platform: template
    name: "Dummy Blood Sugar"
    id: blood_sugar
    unit_of_measurement: mg/dL
    lambda: |-
      return 25 + rand() % 300;
    update_interval: 5s

  - platform: wifi_signal
    name: ${disp_name} WiFi Status
    update_interval: 5s

captive_portal:
```
Here is how it looks
[image]

clydebarrow, 2024-12-18:
Ok, I have reproduced the problem. A work-around is to add this to your display config:
```
    draw_from_origin: true
```

Alternatively, if used with LVGL rather than the display lambda calls it works fine, and is a lot faster.

I'll work on a proper fix.

closed 2024-12-19 by jesserockz