esphome/issues #6562: ens160 does not deliver values all time

crosenbe, 2024-12-16:
### The problem

ens160 started to deliver values but after a reboot it stopped working. Only AHT21 continued to provide sensor values. ENS160 just print the log message

 `ENS160 readings unavailable - Normal Operation but readings not ready`

### Which version of ESPHome has the issue?

2024.11.2

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

n/a

### What platform are you using?

ESP32

### Board

nodemcu

### Component causing the issue

i2c

### YAML Config

```yaml
esphome:
  name: esp-air01
  friendly_name: ESP Air01

esp32:
  board: esp32dev
  framework:
    type: arduino

i2c:
  sda: GPIO21
  scl: GPIO22
  frequency: 100
  scan: true
  id: i2c_a

sensor:
  - platform: ens160_i2c
    eco2:
      name: "ENS160 CO2"
    tvoc:
      name: "ENS160 TVOC"
    aqi:
      id: ens160_air_quality_index
      name: "ENS160 AQI"
    update_interval: 60s
    address: 0x53
    compensation:
      temperature: id_temperature_sensor
      humidity: id_humidity_sensor
  - platform: aht10
    variant: AHT20
    temperature: 
      name: "ATH21 Temperature"
      id: id_temperature_sensor
    humidity: 
      name: "ATH21 Humidity"
      id: id_humidity_sensor
    update_interval: 30s
    address: 0x38

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "xxx"

ota:
  - platform: esphome
    password: "xxx"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
```

### Anything in the logs that might be useful for us?

```txt
[09:27:50][I][app:100]: ESPHome version 2024.11.2 compiled on Dec 12 2024, 18:59:39
[09:27:50][C][logger:185]: Logger:
[09:27:50][C][logger:186]:   Level: DEBUG
[09:27:50][C][logger:188]:   Log Baud Rate: 115200
[09:27:50][C][logger:189]:   Hardware UART: UART0
[09:27:50][C][i2c.arduino:071]: I2C Bus:
[09:27:50][C][i2c.arduino:072]:   SDA Pin: GPIO21
[09:27:50][C][i2c.arduino:073]:   SCL Pin: GPIO22
[09:27:50][C][i2c.arduino:074]:   Frequency: 100 Hz
[09:27:50][C][i2c.arduino:086]:   Recovery: bus successfully recovered
[09:27:50][I][i2c.arduino:096]: Results from i2c bus scan:
[09:27:50][I][i2c.arduino:102]: Found i2c device at address 0x38
[09:27:50][I][i2c.arduino:102]: Found i2c device at address 0x53
[09:27:50][C][ens160:278]: ENS160:
[09:27:50][D][ens160:300]: Setup successful
[09:27:50][I][ens160:304]: Firmware Version: 5.4.6
[09:27:50][C][ens160:306]:   Update Interval: 60.0s
[09:27:50][C][ens160:307]:   CO2 Sensor: 'ENS160 CO2'
[09:27:50][C][ens160:307]:     Device Class: 'carbon_dioxide'
[09:27:50][C][ens160:307]:     State Class: 'measurement'
[09:27:50][C][ens160:307]:     Unit of Measurement: 'ppm'
[09:27:50][C][ens160:307]:     Accuracy Decimals: 0
[09:27:50][C][ens160:307]:     Icon: 'mdi:molecule-co2'
[09:27:50][C][ens160:308]:   TVOC Sensor: 'ENS160 TVOC'
[09:27:50][C][ens160:308]:     Device Class: 'volatile_organic_compounds_parts'
[09:27:50][C][ens160:308]:     State Class: 'measurement'
[09:27:50][C][ens160:308]:     Unit of Measurement: 'ppb'
[09:27:50][C][ens160:308]:     Accuracy Decimals: 0
[09:27:50][C][ens160:308]:     Icon: 'mdi:radiator'
[09:27:50][C][ens160:309]:   AQI Sensor: 'ENS160 AQI'
[09:27:50][C][ens160:309]:     Device Class: 'aqi'
[09:27:50][C][ens160:309]:     State Class: 'measurement'
[09:27:50][C][ens160:309]:     Unit of Measurement: ''
[09:27:50][C][ens160:309]:     Accuracy Decimals: 0
[09:27:50][C][ens160:309]:     Icon: 'mdi:chemical-weapon'
[09:27:50][C][ens160:312]:     Temperature Compensation: 'ATH21 Temperature'
[09:27:50][C][ens160:312]:     Device Class: 'temperature'
[09:27:50][C][ens160:312]:     State Class: 'measurement'
[09:27:50][C][ens160:312]:     Unit of Measurement: '°C'
[09:27:50][C][ens160:312]:     Accuracy Decimals: 2
[09:27:50][C][ens160:313]:     Humidity Compensation: 'ATH21 Humidity'
[09:27:50][C][ens160:313]:     Device Class: 'humidity'
[09:27:50][C][ens160:313]:     State Class: 'measurement'
[09:27:50][C][ens160:313]:     Unit of Measurement: '%'
[09:27:50][C][ens160:313]:     Accuracy Decimals: 2
[09:27:50][C][ens160_i2c.sensor:028]:   Address: 0x53
[09:27:50][C][aht10:162]: AHT10:
[09:27:50][C][aht10:163]:   Address: 0x38
[09:27:50][C][aht10:167]:   Temperature 'ATH21 Temperature'
[09:27:50][C][aht10:167]:     Device Class: 'temperature'
[09:27:50][C][aht10:167]:     State Class: 'measurement'
[09:27:50][C][aht10:167]:     Unit of Measurement: '°C'
[09:27:50][C][aht10:167]:     Accuracy Decimals: 2
[09:27:50][C][aht10:168]:   Humidity 'ATH21 Humidity'
[09:27:50][C][aht10:168]:     Device Class: 'humidity'
[09:27:50][C][aht10:168]:     State Class: 'measurement'
[09:27:50][C][aht10:168]:     Unit of Measurement: '%'
[09:27:50][C][aht10:168]:     Accuracy Decimals: 2

[09:27:58][D][sensor:094]: 'ATH21 Temperature': Sending state 19.75269 °C with 2 decimals of accuracy
[09:27:58][D][sensor:094]: 'ATH21 Humidity': Sending state 37.29239 % with 2 decimals of accuracy
[09:28:01][D][ens160:172]: ENS160 readings unavailable - Normal Operation but readings not ready
[09:28:28][D][sensor:094]: 'ATH21 Temperature': Sending state 19.73534 °C with 2 decimals of accuracy
[09:28:28][D][sensor:094]: 'ATH21 Humidity': Sending state 37.31184 % with 2 decimals of accuracy
[09:28:58][D][sensor:094]: 'ATH21 Temperature': Sending state 19.76395 °C with 2 decimals of accuracy
[09:28:58][D][sensor:094]: 'ATH21 Humidity': Sending state 37.33768 % with 2 decimals of accuracy
[09:29:01][D][ens160:172]: ENS160 readings unavailable - Normal Operation but readings not ready
```

crosenbe, 2024-12-18:
Yes, that's true, i already mentioned the sensors delivered values but stopped and even if i wait more than 24 hours i do not get values. Maybe it's a hardware issue.

Bjk8kds, 2024-12-18:
> Yes, that's true, i already mentioned the sensors delivered values but stopped and even if i wait more than 24 hours i do not get values. Maybe it's a hardware issue.

Oh sorry, my bad.
Have you tried adding a capacitor like the [comments here](https://community.home-assistant.io/t/issue-with-ens160-aht21-ens160-readings-unavailable/697522/2) suggest?
I had (pretty much) similar problem on my combo board and this step saved me.

crosenbe, 2024-12-18:
Good hint, thanks a lot. I will give it a try.

crosenbe, 2024-12-20:
Just figured out the electrical characteristics, it seems the peak supply current is maybe the issue.

`IDD_PK Peak Supply 65(<5ms) mA`

Based on the voltage and the peak current and the max duration of less 5ms a capacitor of 180uF should have enough capacity. I don't know, if 47uF or 100uF will cover all potential issues. Anyone have more experience?

Bjk8kds, 2024-12-20:
> Good hint, thanks a lot. I will give it a try.

How did it go? Did it work?

> Based on the voltage and the peak current and the max duration of less 5ms a capacitor of 180uF should have enough capacity. I don't know, if 47uF or 100uF will cover all potential issues. Anyone have more experience?

Just read my comment above, mine is also based on his suggestion of using 47 uF capacitor.

Unfortunately one of my boards is a bit damaged/error (?)
It shows a value of 0 ppb if the air quality is quite clean, but will increase to around 60-100 if the air is not clean, aka very insensitive compared to the original (state). I don't know if the quality of the board is bad so it's easily damaged.
But previously I did connect the 3v3 pin to the 3v3 pin on the ESP32, I once read somewhere that you shouldn't use the 3v3 pin (on the board) because it's not good or something like that.
Now I connect the 3v3 pin from the ESP32 to the 5v pin on the board (another board), just to be safe (I used the 3v3 pin on the ESP32 because I ran out of space to connect to the 5v pin).
I'll let you know if the board breaks again.

crosenbe, 2024-12-20:
> How did it go? Did it work?

I didn't even start because i have no capacitor and need to get one first.

> I once read somewhere that you shouldn't use the 3v3 pin (on the board) because it's not good or something like that. Now I connect the 3v3 pin from the ESP32 to the 5v pin on the board (another [...]

I also connected the board with the 3,3V Pin but i changed to VIN instead. I used VIN of my ESP which provides a higher voltage. The VIN of the ENS160 can take 3,3v - 5v it seems. Works in general, but still no update of the sensor values.

crosenbe, 2024-12-20:
> `IDD_PK Peak Supply 65(<5ms) mA`
> 
> Based on the voltage and the peak current and the max duration of less 5ms a capacitor of 180uF should have enough capacity. I don't know, if 47uF or 100uF will cover all potential issues. Anyone have more experience?

I think i have only to calculate the delta between peak of 65mA and regular of 24mA. This would result in 

41mA * 5ms / 1,8V = 113,9uF

crosenbe, 2024-12-20:
Capacitory 100 uF is added, but still no improvement. Maybe it's really a hardware issue. Maybe it's better to split ens160 and temperatuir into two boards or use a different sensor in general.

```
[16:22:57][V][ens160:157]: Status: ENS160 STATAS bit    0x0
[16:22:57][V][ens160:159]: Status: ENS160 STATER bit    0x0
[16:22:57][V][ens160:160]: Status: ENS160 VALIDITY FLAG 0x00
[16:22:57][V][ens160:162]: Status: ENS160 NEWDAT bit    0x0
[16:22:57][V][ens160:164]: Status: ENS160 NEWGPR bit    0x0
[16:22:57][D][ens160:172]: ENS160 readings unavailable - Normal Operation but readings not ready
```

Bjk8kds, 2024-12-21:
> Capacitory 100 uF is added, but still no improvement. Maybe it's really a hardware issue. Maybe it's better to split ens160 and temperatuir into two boards or use a different sensor in general.
> 
Unfortunately, it seems like the board is broken.
I've read on some forums that their boards are not working/broken. It seems like the quality is not good.
It's better to buy a board with a known brand, such as Fermion, Adafruit, DFrobot, etc. even though the price can be twice as much, and does not include a temperature/humidity sensor.

Mine, the temperature/humidity sensor is only as a reference for the ESN160, of course with offset settings.
Fortunately have 2 of those boards as spares. But I don't want to get my hopes up. If it breaks again, I'll definitely switch to a known brand.