esphome/issues #6565: Latest version won't update

boby51, 2024-12-18:
### The problem

Failed to perform the action update/install. Error compiling konnected-f33bcc.yaml; Try again in ESPHome dashboard for more information.

### Which version of ESPHome has the issue?

 2024.12.0 

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2024.12.4

### What platform are you using?

ESP32

### Board

konnected

ssieb, 2024-12-18:
Did you try what it said?  Compiling in the dashboard will let you see the compile logs to find out why it's failing.

boby51, 2024-12-18:
INFO ESPHome 2024.12.0
INFO Reading configuration /config/esphome/konnected-f33bcc.yaml...
Failed config

esp32_improv: [source /data/packages/8735b6a9/packages/wifi-esp32.yaml:9]
  
  Must be none.
  authorizer: False

ssieb, 2024-12-18:
Change `False` to `none`.  When did you last update this device?

ssieb, 2024-12-18:
You're using an external package.  It comes from there.  `packages/wifi-esp32.yaml`
The reason I asked about the last update was because I thought it was related to a change from a year ago, but it only sort of was.  There was a more recent change that caused this.  The docs will get updated and it will get added to the changelog as a breaking change.

boby51, 2024-12-18:
Okay Thanks

ssieb, 2024-12-18:
I'm not sure you understood.  This isn't something that will get "fixed".  You need to contact whoever is providing the config you're using to get them to fix it.  In the meantime, you can try overriding that part of the config.

Where is the config from?

boby51, 2024-12-18:
https://github.com/konnected-io

It has been fixed their end

Sent with Spark

closed 2024-12-20 by ssieb