esphome/issues #6566: heltec wifi kit 32 v2 doesn't compile anymore with esphome 2024.12.0

bemabalu, 2024-12-18:
### The problem

When trying to compile with the new 2024.12.0 Version of ESP Home the compile fails. Seems the board isn't known anymore. Worked fine with last Version of ESPHome.

### Which version of ESPHome has the issue?

2024.12.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.12.4

### What platform are you using?

ESP32

### Board

heltec wifi kit 32 v2

### YAML Config

```yaml
esphome:
  name: display

esp32:
  board: heltec_wifi_kit_32_v2
  framework:
    type: arduino

.....
```

### Anything in the logs that might be useful for us?

```txt
INFO ESPHome 2024.12.0
INFO Reading configuration /config/esphome/display.yaml...
Failed config

esp32: [source /config/esphome/display.yaml:5]
  
  This board is unknown, if you are sure you want to compile with this board selection, override with option 'variant'.
  board: heltec_wifi_kit_32_v2
  framework: 
    type: arduino
```

randybb, 2024-12-18:
There was a breaking change:
Synchronise esp32 boards with platform version 51.03.07 [esphome#7945](https://github.com/esphome/esphome/pull/7945) by [@jesserockz](https://github.com/jesserockz) (breaking-change)
That board profile was [renamed](https://github.com/esphome/esphome/pull/7945/files#diff-927fe5a751cff1b1ef42f66c4da96b7a65183053a92f9e223a93bdd1d2162732L1655-R1761)
heltec_wifi_kit_32_v2 -> heltec_wifi_kit_32_V3

closed 2024-12-18 by randybb

bemabalu, 2024-12-19:
Well, this is hard to believe, as the heltec_wifi_kit_32_v2 uses an ESP32-D0 and the _v3 an ESP32-S3 as seen at [https://heltec.org/project/wifi-kit32-v3/](url). Seems the _v2 is just missing in platform 51.03.07

randybb, 2024-12-19:
I did not check the profile, but this is what it is. Just use the default devkit profile for esp32, these profiles are mostly just a copy of that one.

faninx, 2024-12-22:
I added "variant: esp32", it works.

kfiducia, 2025-02-01:
i needed to make a code change and tried the variant thing, the device no longer works. anyone figure it out?