esphome/issues #6567: ESP32-C6 fails to compile with default versions

syphernl, 2024-12-18:
### The problem

I've been using the ESP32-C6 for some time now with:

```yaml
    version: 5.2.2
    platform_version: 6.7.0
```

After trying a bunch of combinations this was the only that worked the past releases. 
I read in the release notes for 2024.12 that there is now official support for the C6 so I removed the `version` and `platform_version` and updated. 

However, it fails with an error (see below). With the versions added back in they compile fine on 2024.12.0

### Which version of ESPHome has the issue?

2024.12.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.12.4

### What platform are you using?

ESP32-IDF

### Board

esp32-c6-devkitc-1

### Component causing the issue

esp32_ble

### YAML Config

```yaml
esp32:
  board: esp32-c6-devkitc-1
  variant: esp32c6
  flash_size: 8MB
  framework:
    type: esp-idf
    version: 5.2.2
    platform_version: 6.7.0
    sdkconfig_options:
      CONFIG_ESPTOOLPY_FLASHSIZE_8MB: y
      CONFIG_ESPTOOLPY_FLASHSIZE: "8MB"
      # Added for Bermuda
      #CONFIG_BT_BLE_50_FEATURES_SUPPORTED: y
      #CONFIG_BT_BLE_42_FEATURES_SUPPORTED: y
      #CONFIG_ESP_TASK_WDT_TIMEOUT_S: "10"
```

### Anything in the logs that might be useful for us?

```txt
In file included from src/esphome/components/esp32_ble/ble.cpp:6:
src/esphome/components/esp32_ble/const_esp32c6.h:69:1: error: 'const esp_bt_controller_config_t' has no non-static data member named 'enable_uart_hci'
   69 | };
      | ^
Compiling .pioenvs/btproxy-livingroom/src/esphome/components/esp32_ble_client/ble_characteristic.cpp.o
Compiling .pioenvs/btproxy-livingroom/src/esphome/components/esp32_ble_client/ble_client_base.cpp.o
*** [.pioenvs/btproxy-livingroom/src/esphome/components/esp32_ble/ble.cpp.o] Error 1
```

### Additional information

Additional settings that may be relevant:

```yaml
esphome:
  platformio_options:
    board_build.f_flash: 80000000L
    board_build.flash_mode: dio
    board_build.flash_size: 8MB
    board_upload.maximum_size: 8388608
```

ireun, 2024-12-18:
Same here!

```
Compiling .pioenvs/esp32-c6/src/esphome/components/esp32_ble/ble_advertising.cpp.o
In file included from src/esphome/components/esp32_ble/ble.cpp:6:
src/esphome/components/esp32_ble/const_esp32c6.h:69:1: error: 'const esp_bt_controller_config_t' has no non-static data member named 'enable_uart_hci'
   69 | };
      | ^
*** [.pioenvs/esp32-c6/src/esphome/components/esp32_ble/ble.cpp.o] Error 1
========================= [FAILED] Took 78.58 seconds =========================
```

My config:
```yaml
esphome:
  name: esp32-c6
  friendly_name: Sypialnia

esp32:
  board: esp32-c6-devkitc-1
  variant: esp32c6
  flash_size: 16MB
  framework:
    type: esp-idf
    # version: 5.3.0
    # platform_version: 6.8.1
    # sdkconfig_options:
    #   CONFIG_OPENTHREAD_ENABLED: n
    #   CONFIG_ENABLE_WIFI_STATION: y
    #   CONFIG_USE_MINIMAL_MDNS: y
    #   CONFIG_ESPTOOLPY_FLASHSIZE_16MB: y

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "xx"

ota:
  - platform: esphome
    password: "xx"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esp32C6 Fallback Hotspot"
    password: "xx"

captive_portal:

esp32_ble_tracker:

ble_client:
  - mac_address: xx
    id: oclean

i2c:
  scl: 6
  sda: 5
  scan: false

sensor:
  - platform: bme280_i2c
    temperature:
      name: "BME280 Temperature"
      accuracy_decimals: 2
      oversampling: 16x
    pressure:
      name: "BME280 Pressure"
      oversampling: 16x
    humidity:
      name: "BME280 Humidity"
      oversampling: 16x
    address: 0x76
    update_interval: 10s
    iir_filter: 16x
  - platform: ble_client
    type: characteristic
    ble_client_id: oclean
    name: "Oclean Air"
    service_uuid: '180f'
    characteristic_uuid: '2a19'
    device_class: "battery"
    unit_of_measurement: "%"
```

ximex, 2025-02-19:
check this comment: https://github.com/esphome/issues/issues/6639#issuecomment-2669790604

ireun, 2025-02-19:
fixed for me, current config is this:
```yaml
esphome:
  name: "esphome-esp32c6-sypialnia"
  friendly_name: Sypialnia

external_components:
  - source: github://pr#8283
    components: [ i2c ]

esp32:
  board: esp32-c6-devkitc-1
  variant: esp32c6
  framework:
    type: esp-idf
    version: "5.3.1"
    platform_version: 6.9.0
    sdkconfig_options:
      CONFIG_ESPTOOLPY_FLASHSIZE_16MB: y
```

ximex, 2025-02-19:
@ireun `variant` and `sdkconfig_options` shouldn't be necessary any more with `2025.2.0`. this got fixed with the last release

ireun, 2025-02-19:
You were right! Thanks for the tip! Current config as fallows:
```yaml
esphome:
  name: "esphome-esp32c6-sypialnia"
  friendly_name: Sypialnia

external_components:
  - source: github://pr#8283
    components: [ i2c ]

esp32:
  board: esp32-c6-devkitc-1
  flash_size: 16MB
  framework:
    type: esp-idf
    version: "5.3.1"
    platform_version: 6.9.0

i2c:
  scl: 6
  sda: 5
  scan: false
```

itzViking, 2025-04-18:
I was getting the same error when trying to setup the bluetooth proxy
`src/esphome/components/esp32_ble/const_esp32c6.h:69:1: error: 'const esp_bt_controller_config_t' has no non-static data member named 'enable_uart_hci'`

Now works for me with non default version (latest pio as of writing):

```
esp32:
  board: esp32-c6-devkitc-1
  flash_size: 8MB
  framework:
    type: esp-idf
    version: "5.4.0"
    platform_version: espressif32 @ 6.10.0
```

ireun, 2025-06-18:
I've just updated ESPHome to 2025.6.0, and my C6 stopped booting

ximex, 2025-06-18:
@ireun pls post your current config

ximex, 2025-06-18:
any logs? compiling was successfully? does you ESP32-C6 really have 16MB?
could you try one time without `platform: bme280_i2c` and one time without `platform: ble_client`/`ble_client`/`esp32_ble_tracker`

ireun, 2025-06-18:
Huh, seems that it boots without BME / I2C library..