esphome/issues #6569: Flash memory size mismatch detected. Expected 4MB, found 2MB!

Bjk8kds, 2024-12-18:
### The problem

In the previous version, everything was fine, but a warning appeared on Esphome version 2024.12.0, even though my board was correct (it should be) 4 MB.

### Which version of ESPHome has the issue?

2024.12.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

Core 2024.12.4

### What platform are you using?

ESP32-IDF

### Board

Xiao ESP32C3, C3 Supermini, S3 Mini

### Component causing the issue

ESPHome 2024.12.0

### YAML Config

```yaml
esp32:
  board: seeed_xiao_esp32c3 # also lolin_c3_mini and adafruit_feather_esp32s3, etc
  framework:
    type: esp-idf
```

### Anything in the logs that might be useful for us?

```txt
INFO ESPHome 2024.12.0
INFO Reading configuration /config/esphome/esp32c3-bt-proxy.yaml...
WARNING GPIO9 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
WARNING Found and merged multiple configurations for ota platform esphome port(s) [3232]
INFO Generating C++ source...
INFO Updating https://github.com/espressif/esp-protocols.git@mdns-v1.3.2
INFO Compiling app...
Processing esp32c3-bt-proxy (board: seeed_xiao_esp32c3; framework: espidf; platform: https://github.com/pioarduino/platform-espressif32.git#51.03.07)
--------------------------------------------------------------------------------
HARDWARE: ESP32C3 160MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.50105.0 (5.1.5) 
 - tool-cmake @ 3.21.3 
 - tool-esptoolpy @ 4.8.1 
 - tool-mklittlefs @ 3.2.0 
 - tool-ninja @ 1.7.1 
 - tool-riscv32-esp-elf-gdb @ 12.1.0+20221002 
 - tool-xtensa-esp-elf-gdb @ 12.1.0+20221002 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-riscv32-esp @ 12.2.0+20230208
Reading CMake configuration...
Warning! Flash memory size mismatch detected. Expected 4MB, found 2MB!
Please select a proper value in your `sdkconfig.defaults` or via the `menuconfig` target!
Dependency Graph
|-- ArduinoJson @ 6.18.5
```

dwt15, 2024-12-18:
и у меня

dwt15, 2024-12-18:
экран стал подергиваться ( наверное из за этого)

DAVe3283, 2024-12-18:
IMO this is a duplicate of #5404

As a workaround, you can add the following to your project:

```yaml
esp32:
  board: seeed_xiao_esp32c3 # or whatever
  flash_size: 4MB
  framework:
    type: esp-idf
    # Fix flash size by ensuring IDF flash size is also set (was defaulting to 2MB?!?)
    # See https://github.com/esphome/issues/issues/5404
    sdkconfig_options:
      CONFIG_ESPTOOLPY_FLASHSIZE_4MB: y
```

bipsendk, 2024-12-19:
Have the same issue on lolin_s2_mini board

closed 2024-12-20 by kbx81

MorganMLGman, 2025-01-14:
Have the same issue on 2024.12.2
```
esp32:
  board: seeed_xiao_esp32c3
  variant: esp32c3 
  flash_size: 4MB
  framework:
    type: esp-idf
    version: recommended
    sdkconfig_options:
      CONFIG_ESPTOOLPY_FLASHSIZE_4MB: y
```

```
PS C:\Users\cholo\Desktop\ESPHome\GarageLights> python -m esphome --version
Version: 2024.12.2
PS C:\Users\cholo\Desktop\ESPHome\GarageLights> python -m esptool --port COM9 flash_id
esptool.py v4.7.0
Serial port COM9
Connecting...
Detecting chip type... ESP32-C3
Chip is ESP32-C3 (QFN32) (revision v0.4)
Features: WiFi, BLE, Embedded Flash 4MB (XMC)
Crystal is 40MHz
MAC: d8:3b:da:2f:64:88
Uploading stub...
Running stub...
Stub running...
Manufacturer: 20
Device: 4016
Detected flash size: 4MB
Hard resetting via RTS pin...
```

```
Successfully created esp32c3 image.
Archiving .pioenvs\garage_lights\lib699\libImprov.a
Linking .pioenvs\garage_lights\firmware.elf
Retrieving maximum program size .pioenvs\garage_lights\firmware.elf
Checking size .pioenvs\garage_lights\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]  14.3% (used 46772 bytes from 327680 bytes)
Flash: [==========]  96.0% (used 1761844 bytes from 1835008 bytes)
Building .pioenvs\garage_lights\firmware.bin
esptool.py v4.8.1.1
Creating esp32c3 image...
Merged 2 ELF sections
Successfully created esp32c3 image.
```

ssieb, 2025-01-14:
@MorganMLGman there's no issue there.  This issue was about a warning which you are not getting.  If your concern is that you only have 2MB of flash for the program, that's normal.  The other half is for OTA updating.

MorganMLGman, 2025-01-14:
But shouldn't it report 4MB with only 50% used in that case?

kbx81, 2025-01-14:
If it told you that the flash was only 50% used and you tried to upload (in this case/example) a 2.1 MB program and that failed due to "insufficient space", would that not be confusing, as well?

MorganMLGman, 2025-01-15:
No it wouldn't as this case was already well handled by this error message.
[image]

And what if I don't want to use OTA? Would it still report as a 2MB flash even if it is 4MB, or would it report as 120% utilization if I use more than 2MB?

ssieb, 2025-01-15:
That's not an option.  It will always generate a partition table for OTA.