esphome/issues #6575: No readings via Modbus

stawek7902, 2024-12-18:
### The problem

After updating from 2024.11 to version 2024.12, error 705 is returned on the Modbus instead of the sensor value. 

### Which version of ESPHome has the issue?

2024.12.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.12.

### What platform are you using?

ESP32

### Board

Kincony KC868-A2

### Component causing the issue

Modbus controller

### YAML Config

```yaml
uart:
  id: uart_bus
  rx_pin: GPIO35
  tx_pin: GPIO32
  baud_rate: 9600
  data_bits: 8
  parity: EVEN
  stop_bits: 1
  debug:
    direction: BOTH

modbus:
  id: modbus1
  uart_id: uart_bus
  send_wait_time: 800ms

modbus_controller:
  - id: modbus_device1
    address: 1
    modbus_id: modbus1
    command_throttle: 1000ms
.....
sensor:
  - platform: modbus_controller
    modbus_controller_id: modbus_device1
    id: napieciedds661
    name: "P1 napiecie SSR1 BOJLER"
    unit_of_measurement: "V"
    value_type: FP32
    custom_command: [0x1, 0x4, 0x00, 0x00, 0x00, 0x01]
    accuracy_decimals: 1
```

### Anything in the logs that might be useful for us?

```txt
23:05:04
[D]
[uart_debug:114]
>>> 01:04:00:12:00:01:91:CF
23:05:04
[E]
[modbus_controller:705]
not enough data for value
23:05:04
[D]
[modbus_controller.sensor:025]
Sensor new state: 0.00
23:05:04
[D]
[sensor:094]
'P1 moc SSR1 BOJLER': Sending state 0.00000 W with 1 decimals of accuracy
23:05:04
[D]
[uart_debug:114]
<<< 01:04:02:00:00:B9:30
```

### Additional information

Before the firmware change, the readings were correct. No errors or warnings.

stawek7902, 2024-12-19:
The value ...:02:43:6F:... is shown correctly returned by the queried energy meter, but this value does not appear as a new value for the sensor.

stawek7902, 2024-12-19:
It is possible that this part of the code in modbus_controller_8cpp_source is responsible for the error, but I cannot interpret it.
"...
case SensorValueType::FP32:
       if (size >= 4) {
         value = get_data<uint32_t>(data, offset);
         value = [mask_and_shift_by_rightbit]((uint32_t) value, bitmask);
       } else {
         error = true;
       }
..."

stawek7902, 2024-12-20:
After downgrading the firmware to version 2024.11.0 the problem disappeared.

stawek7902, 2024-12-20:
[09:55:51][D][uart_debug:114]: >>> 01:04:00:00:00:01:31:CA
[09:55:51][D][modbus_controller.sensor:025]: Sensor new state: 241.00
[09:55:51][D][sensor:094]: 'P1 napiecie SSR1 BOJLER': Sending state 241.00000 V with 1 decimals of accuracy
[09:55:51][D][uart_debug:114]: <<< 01:04:02:43:71:48:24

randybb, 2024-12-20:
A check to modbus has been added in version 2024.11.2 https://github.com/esphome/esphome/pull/7810
So it is clearly receiving wrong data - it is expecting to get more than it receives - do you have documentation for these modbus calls? Guessing FP32 is too big...

ssieb, 2024-12-20:
Your custom command doesn't match the data type you're telling esphome to expect.  You're only requesting one register, but an FP32 is two registers.  Why do you even need a custom command for this?

closed 2024-12-20 by ssieb

stawek7902, 2024-12-23:
I decided to use a non-standard command because I only need one parameter out of many available. Apparently I chose the wrong method. Thanks for your observations. I'll have to try another solution then. Best regards.

ssieb, 2024-12-23:
Just use the normal settings.  If you need help, come ask on the esphome discord server.