esphome/issues #6576: Climate Toggle Display fails without IR

lacky1991, 2024-12-18:
### The problem

When attempting to toggle the display on or off a Midea air conditioning unit without using an IR remote transmitter, the action fails and an error is logged in the ESPHome logs:

**```
[22:35:47][D][button:010]: 'Midea AC Display Toggle' Pressed.
[22:35:47][W][midea:164]: Action needs remote_transmitter component
```**

### Which version of ESPHome has the issue?

ESPHome 2024.12.0

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP8266

### Board

esp12e

### Component causing the issue

climate uart

### YAML Config

```yaml
# These substitutions allow the end user to override certain values
substitutions:
  name: air-conditioner
  friendly_name: "Air Conditioner"
  
  # Additional substitutions could be added here to make it really easy for the user to override defaults.
  # I am adding a some for example purposes.
  wifi_ap_ssid: "AC-wifi"
  wifi_ap_password: "slwf01pro"
  visual_min_temperature: "17 °C"
  visual_max_temperature: "30 °C"
  visual_temperature_step: "0.5 °C"
  

esphome:
  name: "${name}"
  # Friendly names are used where appropriate in Home Assistant
  friendly_name: "${friendly_name}"
  # Automatically add the mac address to the name
  # so you can use a single firmware for all devices
  name_add_mac_suffix: true

  # This will allow for (future) project identification,
  # configuration and updates.
  project:
    name: SMLIGHT.SLWF-01Pro
    version: "2.1"

  platform: ESP8266
  board: esp12e

# This should point to the public location of this yaml file.
#dashboard_import:
#  package_import_url: github://smlight-dev/slwf-01pro/slwf-01pro-v2.yaml
#  import_full_config: false # or true

wifi:

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "${wifi_ap_ssid}"
    password: "${wifi_ap_password}"

captive_portal:

# Enable logging
logger:
  baud_rate: 0

# Enable Home Assistant API
api:
  services:
    - service: follow_me
      variables:
        temperature: float
      then:
        midea_ac.follow_me:
          temperature: !lambda "return temperature;"
          beeper: false
ota:
  - platform: esphome

uart:
  tx_pin: 12
  rx_pin: 14
  baud_rate: 9600
  
button:
  - platform: factory_reset
    id: factory_reset_btn
    name: Factory reset
# "Display toggle" button
  - platform: template
    name: ${friendly_name} Display Toggle
    id: midea_display_toggle
    icon: mdi:theme-light-dark
    on_press:
      midea_ac.display_toggle:
# "Swing step" button
  - platform: template
    name: ${friendly_name} Swing Step
    id: midea_swing_step
    icon: mdi:tailwind
    on_press:
      midea_ac.swing_step:

climate:
  - platform: midea
    name: ${friendly_name}                            # The value of "friendly_name" will be used here
    id: midea_climate                           
    period: 1s                                        # Optional
    timeout: 2s                                       # Optional
    num_attempts: 3                                   # Optional
    autoconf: false                                   # Autoconfigure most options.
    beeper: true                                      # Beep on commands.
    visual:                                           # Optional. Example of visual settings override.
      min_temperature: "${visual_min_temperature}"    # min: 17
      max_temperature: "${visual_max_temperature}"    # max: 30
      temperature_step: "${visual_temperature_step}"  # min: 0.5
    supported_modes:                                  # All capabilities in this section detected by autoconf.
      - FAN_ONLY                                      # This capability is always used.
      - HEAT_COOL
      - COOL
      - HEAT
      - DRY
    custom_fan_modes:
      - SILENT
      - TURBO
    supported_presets:          # All capabilities in this section detected by autoconf.
      - ECO
      - BOOST
      - SLEEP                   # This capability is always used.
    custom_presets:             # All capabilities in this section detected by autoconf.
      - FREEZE_PROTECTION
    supported_swing_modes:
      - VERTICAL                # This capability is always used.
      - HORIZONTAL
      - BOTH
    outdoor_temperature:        # Optional. Outdoor temperature sensor (may display incorrect values after long inactivity).
      name: ${friendly_name} Outdoor Temperature
    power_usage:                # Optional. Power usage sensor (only for devices that support this feature).
      name: ${friendly_name} Power Usage
    humidity_setpoint:          # Optional. Indoor humidity sensor (only for devices that support this feature).
      name: ${friendly_name} Indoor Humidity

switch:
  - platform: template
    name: ${friendly_name} Beeper
    id: midea_beeper
    icon: mdi:volume-source
    restore_mode: RESTORE_DEFAULT_OFF
    optimistic: true
    turn_on_action:
      midea_ac.beeper_on:
    turn_off_action:
      midea_ac.beeper_off:

sensor:
#  - platform: homeassistant
#    entity_id: ${follow_me_sensor}
#    id: follow_me_sensor
#    filters:
#      - throttle: 10s
#      - heartbeat: 2min
#      - debounce: 1s
#    on_value:
#      midea_ac.follow_me:
#        temperature: !lambda "return x;"
#        beeper: false
  - platform: wifi_signal
    name: ${friendly_name} Wi-Fi Signal
    update_interval: 60s
  - platform: uptime
    name: "Uptime"
    id: uptime_sec
    internal: true
  - platform: template
    name: ${friendly_name} Uptime Days
    lambda: |-
      return (id(uptime_sec).state/60)/60/24;
    icon: mdi:clock-start
    unit_of_measurement: days
    update_interval: 60s
    
# "Power toggle" script
script:
  - id: on_button_click
    then:
      midea_ac.power_toggle:

web_server:
  port: 80
```

### Anything in the logs that might be useful for us?

```txt
[22:35:47][D][button:010]: 'Midea AC Display Toggle' Pressed.
[22:35:47][W][midea:164]: Action needs remote_transmitter component
```
```

### Additional information

im using this device https://smlight.tech/product/slwf-01/

ptparker, 2024-12-28:
Not sure if you figured it out, but you're missing a remote_transmitter section. If you haven't soldered in an IR transmitter, it isn't going to work. 

`
remote_transmitter:
  pin: GPIO5                       # For slwf-01pro-v1.1 dongle
  carrier_duty_percent: 100%        # 50% for IR LED, 100% for direct connect to TSOP IR receiver output.
`

ssieb, 2024-12-28:
The rest of the controls use the uart, but apparently the display toggle needs IR.

closed 2024-12-28 by ssieb