esphome/issues #6578: m5stack.atom-echo-update failed (ESP2024.12.0 & 2024.12.1)

HaraldGithub, 2024-12-19:
### The problem

Update via WLAN doesen't

### Which version of ESPHome has the issue?

2024.12.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.12.4

### What platform are you using?

ESP8266

### YAML Config

```yaml
substitutions:
  name: m5stack-atom-echo-b83c20
  friendly_name: M5-Atom-Echo
packages:
  m5stack.atom-echo-wake-word-voice-assistant: github://esphome/firmware/wake-word-voice-assistant/m5stack-atom-echo.adopted.yaml@main
esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
api:
  encryption:
    key: my_key
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
```

### Anything in the logs that might be useful for us?

```txt
Error-message:

INFO ESPHome 2024.12.1
INFO Reading configuration /config/esphome/m5stack-atom-echo-b83c20.yaml...
INFO Updating https://github.com/esphome/esphome.git@pull/5230/head
INFO Updating https://github.com/jesserockz/esphome-components.git@None
INFO Generating C++ source...
INFO Updating https://github.com/espressif/esp-protocols.git@mdns-v1.3.2
INFO Updating https://github.com/espressif/esp-adf@v2.5
INFO Updating submodules (components/esp-sr, components/esp-adf-libs) for https://github.com/espressif/esp-adf@v2.5
INFO Updating https://github.com/espressif/esp-dsp@v1.2.0
INFO Updating https://github.com/espressif/esp-tflite-micro@v1.3.1
INFO Updating https://github.com/espressif/esp-nn@v1.1.0
INFO Compiling app...
Processing m5stack-atom-echo-b83c20 (board: m5stack-atom; framework: espidf; platform: https://github.com/pioarduino/platform-espressif32.git#51.03.07)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.50105.0 (5.1.5) 
 - tool-cmake @ 3.21.3 
 - tool-esptoolpy @ 4.8.1 
 - tool-mklittlefs @ 3.2.0 
 - tool-ninja @ 1.7.1 
 - tool-riscv32-esp-elf-gdb @ 12.1.0+20221002 
 - tool-xtensa-esp-elf-gdb @ 12.1.0+20221002 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-xtensa-esp32 @ 12.2.0+20230208
Reading CMake configuration...
-- git rev-parse returned 'fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).'
-- Building ESP-IDF components for target esp32
NOTICE: 
Following dependencies have new versions available:
Dependency "espressif/esp-nn": "1.0.2" -> "1.1.0"
Consider running "idf.py update-dependencies" to update your lock file.
Processing 1 dependencies:
[1/1] idf (5.1.5)
-- Configuring incomplete, errors occurred!
See also "/data/build/m5stack-atom-echo-b83c20/.pioenvs/m5stack-atom-echo-b83c20/CMakeFiles/CMakeOutput.log".

fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
CMake Error at /data/cache/platformio/packages/framework-espidf/tools/cmake/build.cmake:266 (message):
  Failed to resolve component 'jsmn'.
Call Stack (most recent call first):
  /data/cache/platformio/packages/framework-espidf/tools/cmake/build.cmake:308 (__build_resolve_and_add_req)
  /data/cache/platformio/packages/framework-espidf/tools/cmake/build.cmake:603 (__build_expand_requirements)
  /data/cache/platformio/packages/framework-espidf/tools/cmake/project.cmake:604 (idf_build_process)
  CMakeLists.txt:3 (project)
========================== [FAILED] Took 3.71 seconds ==========================
```

HaraldGithub, 2024-12-19:
> Same here
> 
> @HaraldGithub just for the protocol: the hardware platform is not ESP8266 as you mentioned. It is ESP32.

Thx, I've corrected the hardware platform (ESP8266 was unlikely pre selected). ;-)

BigBobbas, 2024-12-19:
please see https://github.com/esphome/issues/issues/6573

HaraldGithub, 2024-12-19:
> please see [esphome/wake-word-voice-assistants#42 (comment)](https://github.com/esphome/wake-word-voice-assistants/issues/42#issuecomment-2550985988)

Okay, but as I write above, I'm using the "original"-yaml-file.
How could I fix this issue without using an modified yaml-file?

BigBobbas, 2024-12-19:
@HaraldGithub you just need to add the change to the esp32: block.  Exactly how it is shown in the post linked. or if the device is functioning there is no need to update it all.

HaraldGithub, 2024-12-19:
Device is functioning, so at the moment it's no problem.

Thx for support, but I don't have an esp32:block (as you could see above in my first posting), therefore I asked how to fix it.

BigBobbas, 2024-12-19:
if there is no block you just need to add it into your existing config. @HaraldGithub

HaraldGithub, 2024-12-19:
> if there is no block you just need to add it into your existing config. @HaraldGithub

I‘ve includedthe esp32-part, but Update failed with above mentioned error message, again.

Jojo-A, 2024-12-19:
> > if there is no block you just need to add it into your existing config. @HaraldGithub
> 
> I‘ve includedthe esp32-part, but Update failed with above mentioned error message, again.

Hi,
I had the same problem. I could fix it by cleaning all build files first. After that it compiled correctly :)

AndersRipa, 2024-12-21:
I have the same problem.

sca075, 2024-12-23:
Just for reference "cleaning the building files" did not help.. as mentioned in @Jojo-A  comment.
rolling back to the ESPHOME to 2024.11.x cleaning the build files then update works.

the issue seams here: 
**Failed to resolve component 'jsmn'**
**/data/cache/platformio/packages/framework-espidf/tools/cmake/build.cmake:308 (__build_resolve_and_add_req)**

I guess there is a problem either on the dependencies of the voice assistant or as per recently the platform changed and isn't correctly specified this happens, or.. well here in ESPHome Builder don't know.. I guess you will figure it out and found a solution, also next year would be fine.. no stress please.

complete logs below:  

```
INFO ESPHome 2024.12.2
INFO Reading configuration /config/esphome/esp32-s3-box-3-05ad68.yaml...
INFO Updating https://github.com/esphome/esphome.git@pull/5230/head
INFO Updating https://github.com/jesserockz/esphome-components.git@None
WARNING GPIO0 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
WARNING Found and merged multiple configurations for ota platform esphome port(s) [3232]
INFO Generating C++ source...
INFO Updating https://github.com/espressif/esp-protocols.git@mdns-v1.3.2
INFO Updating https://github.com/jesserockz/esp32-s3-box-3-board.git@main
INFO Updating https://github.com/espressif/esp-adf@v2.5
INFO Updating submodules (components/esp-sr, components/esp-adf-libs) for https://github.com/espressif/esp-adf@v2.5
INFO Updating https://github.com/espressif/esp-dsp@v1.2.0
INFO Updating https://github.com/espressif/esp-tflite-micro@v1.3.1
INFO Updating https://github.com/espressif/esp-nn@v1.1.0
INFO Compiling app...
Processing esp32-s3-box-3-05ad68 (board: esp32s3box; framework: espidf; platform: https://github.com/pioarduino/platform-espressif32.git#51.03.07)
--------------------------------------------------------------------------------
Library Manager: Installing kahrendt/ESPMicroSpeechFeatures @ 1.1.0
INFO Installing kahrendt/ESPMicroSpeechFeatures @ 1.1.0
Unpacking  [####################################]  100%
Library Manager: ESPMicroSpeechFeatures@1.1.0 has been installed!
INFO ESPMicroSpeechFeatures@1.1.0 has been installed!
HARDWARE: ESP32S3 240MHz, 320KB RAM, 16MB Flash
 - framework-espidf @ 3.50105.0 (5.1.5) 
 - tool-cmake @ 3.21.3 
 - tool-esptoolpy @ 4.8.1 
 - tool-mklittlefs @ 3.2.0 
 - tool-ninja @ 1.7.1 
 - tool-riscv32-esp-elf-gdb @ 12.1.0+20221002 
 - tool-xtensa-esp-elf-gdb @ 12.1.0+20221002 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-riscv32-esp @ 12.2.0+20230208 
 - toolchain-xtensa-esp32s3 @ 12.2.0+20230208
Reading CMake configuration...
-- Found Git: /usr/bin/git (found version "2.39.5") 
-- git rev-parse returned 'fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).'
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /data/cache/platformio/packages/toolchain-xtensa-esp32s3/bin/xtensa-esp32s3-elf-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /data/cache/platformio/packages/toolchain-xtensa-esp32s3/bin/xtensa-esp32s3-elf-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /data/cache/platformio/packages/toolchain-xtensa-esp32s3/bin/xtensa-esp32s3-elf-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32s3
Processing 1 dependencies:
[1/1] idf (5.1.5)
-- Configuring incomplete, errors occurred!
See also "/data/build/esp32-s3-box-3-05ad68/.pioenvs/esp32-s3-box-3-05ad68/CMakeFiles/CMakeOutput.log".

fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
CMake Error at /data/cache/platformio/packages/framework-espidf/tools/cmake/build.cmake:266 (message):
  Failed to resolve component 'jsmn'.
Call Stack (most recent call first):
  /data/cache/platformio/packages/framework-espidf/tools/cmake/build.cmake:308 (__build_resolve_and_add_req)
  /data/cache/platformio/packages/framework-espidf/tools/cmake/build.cmake:603 (__build_expand_requirements)
  /data/cache/platformio/packages/framework-espidf/tools/cmake/project.cmake:604 (idf_build_process)
  CMakeLists.txt:3 (project)

========================== [FAILED] Took 6.24 seconds ==========================
```

Thanks and wishing you All the best for the upcoming time.

BigBobbas, 2024-12-23:
@sca075 a link to the solution is in the comments above https://github.com/esphome/wake-word-voice-assistants/issues/42#issuecomment-2550985988

sca075, 2024-12-23:
> @sca075 a link to the solution is in the comments above https://github.com/esphome/wake-word-voice-assistants/issues/42#issuecomment-2550985988

Just testing it.. 
adding the lines of the platform esp32 it looks it looks it entered in an infinite loop..
But I will update with the details later on.

sca075, 2024-12-23:
@BigBobbas 
```
substitutions:
  name: esp32-s3-box-3-05ad68
  friendly_name: ESP32 S3 Box 3 05ad68

packages:
  esphome.voice-assistant: github://esphome/firmware/wake-word-voice-assistant/esp32-s3-box-3.yaml@main
  
esp32:  
  framework:
    type: esp-idf
    version: 4.4.8
    platform_version: 5.4.0
```
yep this + cleaning the build files, fixed the issue with the update (I guess as a work around) anyhow as per the dependencies would need to be updated manually I guess you agree this isn't a permanent solution for the most.

Thanks in any cases, it is an amazing team work :) Again all the best for the upcoming festivities.

BigBobbas, 2024-12-23:
@sca075 glad that sorted this for you 👍 This is being worked on and a permanent solution will be provided in due course. and this is as you mention a work around in the meantime.

all the best and Seasons Greetings :)

GrumpySlav, 2024-12-24:
I have same issue. Using ESP32-s3-box and esp32-s3-box-3/esp32-s3-box-3.yaml 

`INFO ESPHome 2024.12.2
INFO Reading configuration /config/esphome/esphome-web-2e0310.yaml...
INFO Updating https://github.com/esphome/esphome.git@pull/5230/head
INFO Updating https://github.com/jesserockz/esphome-components.git@None
WARNING GPIO0 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Generating C++ source...
INFO Updating https://github.com/espressif/esp-protocols.git@mdns-v1.3.2
INFO Updating https://github.com/jesserockz/esp32-s3-box-3-board.git@main
INFO Updating https://github.com/espressif/esp-adf@v2.5
INFO Updating submodules (components/esp-sr, components/esp-adf-libs) for https://github.com/espressif/esp-adf@v2.5
INFO Updating https://github.com/espressif/esp-dsp@v1.2.0
INFO Updating https://github.com/espressif/esp-tflite-micro@v1.3.1
INFO Updating https://github.com/espressif/esp-nn@v1.1.0
INFO Compiling app...
Processing esp32-wake1 (board: esp32s3box; framework: espidf; platform: https://github.com/pioarduino/platform-espressif32.git#51.03.07)
--------------------------------------------------------------------------------
HARDWARE: ESP32S3 240MHz, 320KB RAM, 16MB Flash
 - framework-espidf @ 3.50105.0 (5.1.5) 
 - tool-cmake @ 3.21.3 
 - tool-esptoolpy @ 4.8.1 
 - tool-mklittlefs @ 3.2.0 
 - tool-ninja @ 1.7.1 
 - tool-riscv32-esp-elf-gdb @ 12.1.0+20221002 
 - tool-xtensa-esp-elf-gdb @ 12.1.0+20221002 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-riscv32-esp @ 12.2.0+20230208 
 - toolchain-xtensa-esp32s3 @ 12.2.0+20230208
Reading CMake configuration...
-- git rev-parse returned 'fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).'
-- Building ESP-IDF components for target esp32s3
Processing 1 dependencies:
[1/1] idf (5.1.5)
-- Configuring incomplete, errors occurred!
See also "/data/build/esp32-wake1/.pioenvs/esp32-wake1/CMakeFiles/CMakeOutput.log".

fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
CMake Error at /data/cache/platformio/packages/framework-espidf/tools/cmake/build.cmake:266 (message):
  Failed to resolve component 'jsmn'.
Call Stack (most recent call first):
  /data/cache/platformio/packages/framework-espidf/tools/cmake/build.cmake:308 (__build_resolve_and_add_req)
  /data/cache/platformio/packages/framework-espidf/tools/cmake/build.cmake:603 (__build_expand_requirements)
  /data/cache/platformio/packages/framework-espidf/tools/cmake/project.cmake:604 (idf_build_process)
  CMakeLists.txt:3 (project)

========================== [FAILED] Took 3.33 seconds ==========================`

do you know how to open CMakeOutput.log in HAOS? I have no directories in /data.

ssieb, 2025-01-18:
Did you read the other comments?