esphome/issues #6579: ESP32dev Update issues 2024.12.1

Chris-126, 2024-12-19:
### The problem

I just installed the latest version of the ESPHome Builder and now receiving errors while updating my ESP32 wirelessly which was running without any issues until then.

After some research I removed the OTA: part, as it seems to be obsolete now, but did not change anything.

Did I miss a change that has to be applied?

### Which version of ESPHome has the issue?

2024.12.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.12.4

### What platform are you using?

ESP32

### Board

ESP32dev

### YAML Config

```yaml
substitutions:
  name: esp32-bluetooth-proxy-fefff8
  friendly_name: ESP Corridor

packages:
  esphome.bluetooth-proxy: github://esphome/firmware/bluetooth-proxy/esp32-generic.yaml@main

esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}

api:
  encryption:
    key: CZvtuYyS39js/fcGgHd5qzYfV2Zx5mkL7AteC8ON1pM=

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

esp32_ble_tracker:
  scan_parameters:
    interval: 1100ms
    window: 1100ms
    active: true

bluetooth_proxy:
  active: true

#ota:
#- platform: esphome
```

### Anything in the logs that might be useful for us?

```txt
INFO ESPHome 2024.12.1
INFO Reading configuration /config/esphome/esp32-bluetooth-proxy-fefff8.yaml...
INFO Generating C++ source...
INFO Updating https://github.com/espressif/esp-protocols.git@mdns-v1.3.2
INFO Compiling app...
Processing esp32-bluetooth-proxy-fefff8 (board: esp32dev; framework: espidf; platform: https://github.com/pioarduino/platform-espressif32.git#51.03.07)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.50105.0 (5.1.5) 
 - tool-cmake @ 3.21.3 
 - tool-esptoolpy @ 4.8.1 
 - tool-mklittlefs @ 3.2.0 
 - tool-ninja @ 1.10.2 
 - tool-riscv32-esp-elf-gdb @ 12.1.0+20221002 
 - tool-xtensa-esp-elf-gdb @ 12.1.0+20221002 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-xtensa-esp32 @ 12.2.0+20230208
Reading CMake configuration...
JSONDecodeError: Expecting value: line 1 column 1 (char 0):
  File "/usr/local/lib/python3.11/dist-packages/platformio/builder/main.py", line 173:
    env.SConscript("$BUILD_SCRIPT")
  File "/data/cache/platformio/packages/tool-scons/scons-local-4.8.1/SCons/Script/SConscript.py", line 620:
    return _SConscript(self.fs, *files, **subst_kw)
  File "/data/cache/platformio/packages/tool-scons/scons-local-4.8.1/SCons/Script/SConscript.py", line 280:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "/data/cache/platformio/platforms/espressif32/builder/main.py", line 363:
    target_elf = env.BuildProgram()
  File "/data/cache/platformio/packages/tool-scons/scons-local-4.8.1/SCons/Util/envs.py", line 252:
    return self.method(*nargs, **kwargs)
  File "/usr/local/lib/python3.11/dist-packages/platformio/builder/tools/piobuild.py", line 61:
    env.ProcessProgramDeps()
  File "/data/cache/platformio/packages/tool-scons/scons-local-4.8.1/SCons/Util/envs.py", line 252:
    return self.method(*nargs, **kwargs)
  File "/usr/local/lib/python3.11/dist-packages/platformio/builder/tools/piobuild.py", line 121:
    env.BuildFrameworks(env.get("PIOFRAMEWORK"))
  File "/data/cache/platformio/packages/tool-scons/scons-local-4.8.1/SCons/Util/envs.py", line 252:
    return self.method(*nargs, **kwargs)
  File "/usr/local/lib/python3.11/dist-packages/platformio/builder/tools/piobuild.py", line 347:
    SConscript(env.GetFrameworkScript(name), exports="env")
  File "/data/cache/platformio/packages/tool-scons/scons-local-4.8.1/SCons/Script/SConscript.py", line 684:
    return method(*args, **kw)
  File "/data/cache/platformio/packages/tool-scons/scons-local-4.8.1/SCons/Script/SConscript.py", line 620:
    return _SConscript(self.fs, *files, **subst_kw)
  File "/data/cache/platformio/packages/tool-scons/scons-local-4.8.1/SCons/Script/SConscript.py", line 280:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "/data/cache/platformio/platforms/espressif32/builder/frameworks/espidf.py", line 1488:
    project_codemodel = get_cmake_code_model(
  File "/data/cache/platformio/platforms/espressif32/builder/frameworks/espidf.py", line 238:
    codemodel = json.load(fp)
  File "/usr/lib/python3.11/json/__init__.py", line 293:
    return loads(fp.read(),
  File "/usr/lib/python3.11/json/__init__.py", line 346:
    return _default_decoder.decode(s)
  File "/usr/lib/python3.11/json/decoder.py", line 337:
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "/usr/lib/python3.11/json/decoder.py", line 355:
    raise JSONDecodeError("Expecting value", s, err.value) from None
========================== [FAILED] Took 4.76 seconds ==========================
```

Mis1ify, 2024-12-20:
I'm having the same problem, but on stable version of the add-on. Reinstalling the add-on doesn't help, cleaning build files neither.

ssieb, 2024-12-20:
I'm curious what research led you to the `ota:` being obsolete.  If you remove that, you won't be able to wirelessly update the device again.
Try re-installing the addon.

Chris-126, 2024-12-20:
There was a message about the OTA: entry being duplicate. And validating the yaml does not show any issue. In fact, the OTA: entry is listed there even while being deactivated in my yaml:

```
INFO ESPHome 2024.12.1
INFO Reading configuration /config/esphome/esp32-bluetooth-proxy-fefff8.yaml...
substitutions:
  name: esp32-bluetooth-proxy-fefff8
  friendly_name: ESP Corridor
esphome:
  name: esp32-bluetooth-proxy-fefff8
  friendly_name: ESP Corridor
  min_version: 2024.6.0
  name_add_mac_suffix: false
  build_path: build/esp32-bluetooth-proxy-fefff8
  area: ''
  platformio_options: {}
  includes: []
  libraries: []
esp32:
  board: esp32dev
  framework:
    version: 5.1.5
    sdkconfig_options: {}
    advanced:
      ignore_efuse_custom_mac: false
    components: []
    platform_version: https://github.com/pioarduino/platform-espressif32.git#51.03.07
    source: pioarduino/framework-espidf@https://github.com/pioarduino/esp-idf/releases/download/v5.1.5/esp-idf-v5.1.5.zip
    type: esp-idf
  flash_size: 4MB
  variant: ESP32
wifi:
  ap:
    ap_timeout: 1min
  domain: .local
  reboot_timeout: 15min
  power_save_mode: LIGHT
  fast_connect: false
  enable_btm: false
  enable_rrm: false
  passive_scan: false
  enable_on_boot: true
  networks:
  - ssid: xyz
    password: xyz
    priority: 0.0
  use_address: esp32-bluetooth-proxy-fefff8.local
api:
  encryption:
    key: xyz
  port: 6053
  password: ''
  reboot_timeout: 15min
logger:
  baud_rate: 115200
  tx_buffer_size: 512
  deassert_rts_dtr: false
  hardware_uart: UART0
  level: DEBUG
  logs: {}
ota:
- platform: esphome
  id: ota_esphome
  version: 2
  port: 3232
esp32_ble_tracker:
  scan_parameters:
    active: true
    interval: 1100ms
    window: 1100ms
    duration: 5min
    continuous: true
bluetooth_proxy:
  active: true
  cache_services: true
  connections:
  - {}
  - {}
  - {}
button:
- platform: safe_mode
  id: button_safe_mode
  name: Safe Mode Boot
  disabled_by_default: false
  icon: mdi:restart-alert
  entity_category: config
  device_class: restart
- platform: factory_reset
  id: factory_reset_btn
  name: Factory reset
  disabled_by_default: false
  icon: mdi:restart-alert
  entity_category: config
  device_class: restart

INFO Configuration is valid!
```

ssieb, 2024-12-20:
Ok, the `ota:` part is included in the package.  Did you try re-installing to fix the compile error?

Chris-126, 2024-12-20:
Just finished... re-installed the AddOn and then some more downloads were done before the compiling and in the end, the update was a success. Thanks for the hint!

closed 2024-12-20 by Chris-126