esphome/issues #6580: Cannot access Home Assitant Sensor value

Tsjippy, 2024-12-19:
### The problem

The log shows that I received a value but when I try to use that value the value is always 0.

### Which version of ESPHome has the issue?

 2024.12.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.12.4

### What platform are you using?

ESP32

### Board

CYD

### Component causing the issue

homeassistant

### YAML Config

```yaml
- platform: homeassistant
    id: illuminance
    name: "Illuminance Sensor From Home Assistant"
    entity_id: sensor.sbmo_003z_ae6a_illuminance
    on_value:
      light.dim_relative:
        id: backlight
        relative_brightness: !lambda |- 
          if(x == 0){
            return 0.01;
          }else{
            ESP_LOGI("cal", "brightness1 = %d", id(illuminance).state); 
            ESP_LOGI("cal", "brightness2 = %d", x); 
            return (id(illuminance).state * 0.05);
          }
```

### Anything in the logs that might be useful for us?

```txt
[14:59:45][W][component:237]: Component display took a long time for an operation (130 ms).
[14:59:45][W][component:238]: Components should block for at most 30 ms.
[14:59:46][W][component:237]: Component display took a long time for an operation (130 ms).
[14:59:46][W][component:238]: Components should block for at most 30 ms.
[14:59:46][D][sensor:094]: 'board_ldr': Sending state 0.07500 V with 2 decimals of accuracy
[14:59:46][D][homeassistant.sensor:024]: 'sensor.sbmo_003z_ae6a_illuminance': Got state 14.00
[14:59:46][D][sensor:094]: 'Illuminance Sensor From Home Assistant': Sending state 14.00000  with 1 decimals of accuracy
[14:59:46][I][cal:396]: brightness1 = 0
[14:59:46][I][cal:397]: brightness2 = 0
```

### Additional information

As you can see in the log I get the value 14, but x is 0

clydebarrow, 2024-12-19:
Sensor states are floats. Use `%f` to format them, not `%d`.

Tsjippy, 2024-12-20:
o so stupid. thanks!

closed 2024-12-20 by Tsjippy