esphome/issues #6581: Can't update S3Box3 to 2024.12.x

GP920, 2024-12-19:
### The problem

S3Box3 will not update to 2024.12.0 or 2024.12.1 .

(It's entirely possible it's something obvious and I'm just missing it. If so, Thanks in advance.)

### Which version of ESPHome has the issue?

2024.12

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.12

### What platform are you using?

ESP32-IDF

### Board

S3Box3

### YAML Config

```yaml
substitutions:
  name: esp32-s3-box-3-5a9674
  friendly_name: 403_s3box3

  micro_wake_word_model: hey_jarvis
packages:
  esphome.voice-assistant: github://esphome/firmware/wake-word-voice-assistant/esp32-s3-box-3.yaml@main
esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
api:
  encryption:
    key:

wifi:
  ssid: !secret
  password: !secret
```

### Anything in the logs that might be useful for us?

```txt
INFO ESPHome 2024.12.1
INFO Reading configuration /config/esphome/esp32-s3-box-3-5a9674.yaml...
INFO Updating https://github.com/esphome/esphome.git@pull/5230/head
INFO Updating https://github.com/jesserockz/esphome-components.git@None
WARNING GPIO0 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Generating C++ source...
INFO Updating https://github.com/espressif/esp-protocols.git@mdns-v1.3.2
INFO Updating https://github.com/jesserockz/esp32-s3-box-3-board.git@main
INFO Updating https://github.com/espressif/esp-adf@v2.5
INFO Updating submodules (components/esp-sr, components/esp-adf-libs) for https://github.com/espressif/esp-adf@v2.5
INFO Updating https://github.com/espressif/esp-dsp@v1.2.0
INFO Updating https://github.com/espressif/esp-tflite-micro@v1.3.1
INFO Updating https://github.com/espressif/esp-nn@v1.1.0
INFO Compiling app...
Processing esp32-s3-box-3-5a9674 (board: esp32s3box; framework: espidf; platform: https://github.com/pioarduino/platform-espressif32.git#51.03.07)
--------------------------------------------------------------------------------
[... 89 lines cut ...]
Reading CMake configuration...
-- git rev-parse returned 'fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).'
-- Building ESP-IDF components for target esp32s3
NOTICE: 
Following dependencies have new versions available:
Dependency "espressif/esp-nn": "1.0.2" -> "1.1.0"
Consider running "idf.py update-dependencies" to update your lock file.
Processing 1 dependencies:
[1/1] idf (5.1.5)
-- Configuring incomplete, errors occurred!
See also "/data/build/esp32-s3-box-3-5a9674/.pioenvs/esp32-s3-box-3-5a9674/CMakeFiles/CMakeOutput.log".

fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
CMake Error at /data/cache/platformio/packages/framework-espidf/tools/cmake/build.cmake:266 (message):
  Failed to resolve component 'jsmn'.
Call Stack (most recent call first):
  /data/cache/platformio/packages/framework-espidf/tools/cmake/build.cmake:308 (__build_resolve_and_add_req)
  /data/cache/platformio/packages/framework-espidf/tools/cmake/build.cmake:603 (__build_expand_requirements)
  /data/cache/platformio/packages/framework-espidf/tools/cmake/project.cmake:604 (idf_build_process)
  CMakeLists.txt:3 (project)

========================= [FAILED] Took 24.73 seconds =========================
```

randybb, 2024-12-19:
duplicate of https://github.com/esphome/wake-word-voice-assistants/issues/42 / https://github.com/esphome/issues/issues/6578 / https://github.com/esphome/issues/issues/6568

for now add this part:
```
esp32:
  framework:
    type: esp-idf
    version: 4.4.5
    platform_version: 5.4.0
```

closed 2024-12-19 by randybb

GP920, 2024-12-19:
Apologies. Adding that to my yaml and installing seems to have worked, but now produces these errors in the log constantly:

[13:02:37][W][component:157]: Component esp_adf.microphone set Warning flag: unspecified
[13:02:37][W][component:170]: Component esp_adf.microphone cleared Warning flag

Thoughts? Thank you!

Mariusthvdb, 2024-12-21:
@randybb shouldnt be closed at all imho, and certainly not as 'Not planned'...

randybb, 2024-12-21:
it is a duplicate