esphome/issues #6582: Microphone Not Capturing Audio on 2024.12.0/1 w/ ESP32 IDF

smashedr, 2024-12-19:
### The problem

I use Voice Assistant with Micro Wake Word on multiple ESP32's with an INMP441 microphone. After upgrading to ESPHome `2024.12.0` and subsequently `2024.12.1` the microphone no longer functions. There are no errors in the logs, and everything looks normal as I enable Micro Wake Word or activate Voice Assistant, however, it never hears anything from the microphone.

This all works just fine on `2024.11.3`, but microphones stop capturing data when upgrading to `2024.12.0`+

Configs source: https://github.com/smashedr/esphome-configs

### Which version of ESPHome has the issue?

2024.12.0 and 2024.12.1

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2024.12.3 and 2024.12.4

### What platform are you using?

ESP32-IDF

### Board

nodemcu-32s and esp32dev

### Component causing the issue

i2s_audio or microphone

### YAML Config

```yaml
#
# Source: https://github.com/smashedr/esphome-configs/blob/master/esptest52.yaml
#

substitutions:
  # https://esphome.io/components/substitutions
  friendly_name: "Test Box 52"
  box_num: "52"
  box_name: "esptest"
  type: "esp32"
  board: "esp32dev"
  comment: "38-pin"

packages:
  # https://esphome.io/components/packages.html
  debug: !include { file: include/debug.yaml }
  common: !include { file: include/common.yaml }
  voice_assistant: !include { file: include/va/voice_assistant.yaml }
  micro_wake_word: !include
    file: include/va/micro_wake_word.yaml
    vars:
      model: models/computer.json
  discord_logs: !include
    file: include/va/discord.yaml
    vars:
      channel_id: !secret discord_va_id

#external_components:
#  - source: github://jesserockz/esphome-components
#    components: [udp_audio]
#
#udp_audio:
#  ip_address: 192.168.1.20
#  port: 6056

[... 254 lines cut ...]
[21:35:35][D][voice_assistant:511]: State changed from STOP_MICROPHONE to IDLE
[21:35:35][W][va:106]: voice_assistant on_error
[21:35:35][W][va:107]: code: stt-no-text-recognized
[21:35:35][W][va:108]: message: No text recognized
[21:35:35][I][va:147]: formatted_message: Error[stt-no-text-recognized]: No text recognized
[21:35:35][D][main:232]: Running script: va_end
[21:35:35][D][main:253]: Running script: turn_off_led
[21:35:35][D][main:266]: Switch micro_wake_toggle is ON: light.turn_on
[21:35:35][D][light:036]: 'Monochromatic Light 52' Setting:
[21:35:35][D][light:051]:   Brightness: 20%
[21:35:35][D][light:085]:   Transition length: 1.0s
[21:35:35][D][light:091]:   Effect: 'None'
[21:35:35][W][component:237]: Component voice_assistant took a long time for an operation (54 ms).
[21:35:35][W][component:238]: Components should block for at most 30 ms.
[21:35:35][D][voice_assistant:635]: Event Type: 2
[21:35:35][D][voice_assistant:727]: Assist Pipeline ended
[21:35:35][I][main:059]: voice_assistant on_end
[21:35:35][D][main:232]: Running script: va_end
```

### Additional information

I have a friend with a ESP32-S3 who does not have this issue, however, I have it on 2 different ESP32 boards...

Again, all works fine on 2024.11.3, let me know if you need more information...

smashedr, 2024-12-20:
I did quite a bit of testing with 2024.12.0 and could not get this to work. I only ran 2 test on 2024.12.1 which should have worked. I circled back to do some more test with 2024.12.1 and all of a sudden its working. Not worth my time to circle back to 2024.12.0 so going to close this unless I find further issues.

smashedr, 2024-12-24:
I just realized my last flash still had `2024.11.3`. Upon re-flashing with `2024.12.2` the microphone no longer captures. I will do a little more testing but not sure what else to do as it works fine on 11.3 but anything 12+ and the microphone just wont capture.

TheGabeMan, 2024-12-27:
Same issue

fabio-garavini, 2025-01-16:
Same issue
I'm using a `Xiao Sense ESP32S3` board

## My Config
```yaml
esphome:
  name: "fabio-bedroom-sensor"
  friendly_name: "Fabio Bedroom Sensor"

esp32:
  board: esp32s3box
  variant: esp32s3
  framework:
    type: esp-idf
    sdkconfig_options:
      CONFIG_ESP32S3_DEFAULT_CPU_FREQ_240: y
      CONFIG_COMPILER_OPTIMIZATION: "Performance"

logger:
  level: DEBUG

api:
  encryption:
    key: !secret fabio_bedroom_sensor_api_key

ota:
  platform: esphome
  password: !secret fabio_bedroom_sensor_ota_password

wifi:
  networks:
    - ssid: !secret wifi_ssid_notte
      password: !secret wifi_password
    - ssid: !secret wifi_ssid_fastweb
      password: !secret wifi_password_fastweb

  ap:
    ssid: "Fabio-Bedroom-Sensor"
    password: !secret fabio_bedroom_sensor_ap_password

captive_portal:

psram:
  mode: octal
  speed: 80MHz

i2s_audio:
  - id: i2s_microphone
    i2s_lrclk_pin: GPIO42

microphone:
  - platform: i2s_audio
    i2s_audio_id: i2s_microphone
    id: echo_microphone
    i2s_din_pin: GPIO41
    adc_type: external
    bits_per_sample: 16bit
    pdm: true
    channel: left

micro_wake_word:
  models:
    - model: github://esphome/micro-wake-word-models/models/v2/hey_jarvis.json@main
  vad:
    model: github://esphome/micro-wake-word-models/models/v2/vad.json@main
  on_wake_word_detected:
    then:
      - voice_assistant.start:
          wake_word: !lambda return wake_word;

voice_assistant:
  id: voiceassistant
  microphone: echo_microphone
  use_wake_word: false #true
  noise_suppression_level: 1
  auto_gain: 31dBFS
  volume_multiplier: 10

  on_end:
    - if:
        condition:
          switch.is_on: use_wake_word
        then:
          - micro_wake_word.start:

  on_error:
    - if:
        condition:
          switch.is_on: use_wake_word
        then:
          - micro_wake_word.stop:
          - micro_wake_word.start:

  on_client_connected:
    - if:
        condition:
          switch.is_on: use_wake_word
        then:
          - micro_wake_word.start:

  on_client_disconnected:
    - if:
        condition:
          switch.is_on: use_wake_word
        then:
          - micro_wake_word.stop:

switch:
  - platform: template
    name: Use wake word
    id: use_wake_word
    optimistic: true
    restore_mode: RESTORE_DEFAULT_ON
    entity_category: config
    on_turn_on:
      - micro_wake_word.start:

    on_turn_off:
      - micro_wake_word.stop:

button:
  - platform: template
    name: "Voice Assistant"
    on_press:
      - if:
          condition:
            - voice_assistant.is_running
          then:
            - voice_assistant.stop:
          else:
            - voice_assistant.start:
```

## Log
```
[15:13:25][D][micro_wake_word:399]: Resetting buffers and probabilities
[15:13:25][D][micro_wake_word:194]: State changed from IDLE to START_MICROPHONE
[15:13:25][D][micro_wake_word:107]: Starting Microphone
[15:13:25][D][micro_wake_word:194]: State changed from START_MICROPHONE to STARTING_MICROPHONE
[15:13:25][D][micro_wake_word:194]: State changed from STARTING_MICROPHONE to DETECTING_WAKE_WORD
[15:13:25][W][component:157]: Component i2s_audio.microphone set Warning flag: unspecified
```

## ESPHome versions

- 2024.11.3      --> WORK
- 2024.12.0b1 --> NOT WORKING
- 2024.12.2      --> NOT WORKING
- 2025.1.0        --> NOT WORKING

All versions above 2024.11.3 give always the same error as soon the microphone is activated

smashedr, 2025-01-20:
I can confirm that this fixes the issue and https://github.com/esphome/esphome/pull/7706 is breaking microphone functionality...

```yaml
esp32:
  ...
  framework:
    type: esp-idf
    version: 4.4.8
    platform_version: 5.4.0
```

Tested with `2024.12.4` as `dev` broke something else unrelated.

TheGabeMan, 2025-01-20:
Could you elaborate on that? I currently have this in the onju-voice.yaml:

```yaml
esp32:
  board: esp32-s3-devkitc-1
  framework:
    type: arduino
```

Do I change that to what you mentioned?
```yaml
esp32:
  board: esp32-s3-devkitc-1
  framework:
    type: esp-idf
    version: 4.4.8
    platform_version: 5.4.0
```

fabio-garavini, 2025-01-21:
> Could you elaborate on that? I currently have this in the onju-voice.yaml:
>
> ```yaml
> esp32:
>   board: esp32-s3-devkitc-1
>   framework:
>     type: arduino
> ```
>
> Do I change that to what you mentioned? [...]

If you are using the arduino framework you are probably experiencing another error. Can you please provide more about your config and logs

TheGabeMan, 2025-01-24:
Sorry it took some time, but I had the device not connected anymore. Found some time today and connected it again and also did the upgrade of esphome builder to 2024.12.4 and it now responded 5 times in a row, so maybe it is resolved.

GSVil, 2025-01-31:
> I can confirm that this fixes the issue and [esphome/esphome#7706](https://github.com/esphome/esphome/pull/7706) is breaking microphone functionality...
> 
> esp32:
>   ...
>   framework:
>     type: esp-idf
>     version: 4.4.8
>     platform_version: 5.4.0
> Tested with `2024.12.4` as `dev` broke something else unrelated.

Thanks, that sorted it out for me.

joshuaboniface, 2025-05-22:
Has there been any further movement on this? The "workaround" of pinning to the old version seems to have stopped working with the latest 2025.5.0 release, and using those pinned versions fail to compile now with a missing dependency:

```
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.40408.0 (4.4.8) 
 - tool-cmake @ 3.16.4 
 - tool-ninja @ 1.7.1 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
ModuleNotFoundError: No module named 'pkg_resources':
  File "/usr/local/lib/python3.12/site-packages/platformio/builder/main.py", line 173:
    env.SConscript("$BUILD_SCRIPT")
  File "/data/cache/platformio/packages/tool-scons/scons-local-4.8.1/SCons/Script/SConscript.py", line 620:
    return _SConscript(self.fs, *files, **subst_kw)
  File "/data/cache/platformio/packages/tool-scons/scons-local-4.8.1/SCons/Script/SConscript.py", line 280:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "/data/cache/platformio/platforms/espressif32@5.4.0/builder/main.py", line 312:
    target_elf = env.BuildProgram()
  File "/data/cache/platformio/packages/tool-scons/scons-local-4.8.1/SCons/Util/envs.py", line 252:
    return self.method(*nargs, **kwargs)
  File "/usr/local/lib/python3.12/site-packages/platformio/builder/tools/piobuild.py", line 62:
    env.ProcessProgramDeps()
  File "/data/cache/platformio/packages/tool-scons/scons-local-4.8.1/SCons/Util/envs.py", line 252:
    return self.method(*nargs, **kwargs)
  File "/usr/local/lib/python3.12/site-packages/platformio/builder/tools/piobuild.py", line 142:
    env.BuildFrameworks(env.get("PIOFRAMEWORK"))
  File "/data/cache/platformio/packages/tool-scons/scons-local-4.8.1/SCons/Util/envs.py", line 252:
    return self.method(*nargs, **kwargs)
  File "/usr/local/lib/python3.12/site-packages/platformio/builder/tools/piobuild.py", line 352:
    SConscript(env.GetFrameworkScript(name), exports="env")
  File "/data/cache/platformio/packages/tool-scons/scons-local-4.8.1/SCons/Script/SConscript.py", line 684:
    return method(*args, **kw)
  File "/data/cache/platformio/packages/tool-scons/scons-local-4.8.1/SCons/Script/SConscript.py", line 620:
    return _SConscript(self.fs, *files, **subst_kw)
  File "/data/cache/platformio/packages/tool-scons/scons-local-4.8.1/SCons/Script/SConscript.py", line 280:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "/data/cache/platformio/platforms/espressif32@5.4.0/builder/frameworks/espidf.py", line 29:
    import pkg_resources
========================== [FAILED] Took 1.10 seconds ==========================
```

Removing the pinned versions allows this to compile but then the lack-of-audio issue returns.

jesserockz, 2025-05-22:
There have been significant changes to the microphone platforms in 2025.5.0. Please make sure you are not overriding any versions and have updated your configs to match the latest documentation and try again.

If it is still not working, please post your full configuration.

joshuaboniface, 2025-05-22:
Here's my relevant configuration parts (this is part of an [extremely large project](https://github.com/joshuaboniface/supersensor2) with many other components. The microphone is an INMP441 and the board is an ESP32-WROOM-32D devkit.

```
esphome:
  on_boot:
    - priority: -600
      then:
        - wait_until:
            api.connected:
        - delay: 5s
        - voice_assistant.start_continuous:

esp32:
  board: esp32dev
  framework:
    type: esp-idf
    sdkconfig_options:
      CONFIG_ESP32_DEFAULT_CPU_FREQ_240: "y"
      CONFIG_ESP32_DEFAULT_CPU_FREQ_MHZ: "240"
      CONFIG_ESP32_DATA_CACHE_64KB: "y"
      CONFIG_ESP32_DATA_CACHE_LINE_64B: "y"
      CONFIG_ESP32S3_DEFAULT_CPU_FREQ_240: "y"
      CONFIG_ESP32S3_DEFAULT_CPU_FREQ_MHZ: "240"
      CONFIG_ESP32S3_DATA_CACHE_64KB: "y"
      CONFIG_ESP32S3_DATA_CACHE_LINE_64B: "y"
      CONFIG_SPIRAM_CACHE_WORKAROUND: "y"
      CONFIG_OPTIMIZATION_LEVEL_RELEASE: "y"

i2s_audio:
  - id: i2s_mic
    i2s_lrclk_pin: GPIO17 # WS
    i2s_bclk_pin: GPIO16  # SCK

microphone:
  - platform: i2s_audio
    id: mic
    adc_type: external
    i2s_audio_id: i2s_mic
    i2s_din_pin: GPIO4 # SD

micro_wake_word:
  microphone: mic
  models:
    - model: hey_jarvis
  on_wake_word_detected:
    then:
      - voice_assistant.start:
          wake_word: !lambda return wake_word;

voice_assistant:
  microphone: mic
  use_wake_word: true	
  noise_suppression_level: 3
  auto_gain: 31dBFS
  volume_multiplier: 8.0
  id: assist
  on_error:
    - logger.log: "voice error"
    - voice_assistant.start_continuous:
  on_end:
    - logger.log: "voice ended"
    - voice_assistant.start_continuous:
  on_wake_word_detected:
    - light.turn_on:
        id: output_led
        brightness: 100%
        red: 0
        green: 0
        blue: 1
  on_listening:
    - light.turn_on:
        id: output_led
        brightness: 100%
        red: 0
        green: 0
        blue: 1
  on_stt_vad_end:
    - light.turn_on:
        id: output_led
        brightness: 75%
        red: 0
        green: 1
        blue: 1
  on_stt_end:
      - light.turn_off:
        id: output_led
  on_tts_start:
    - if:
        condition:
        - lambda: |-
            ESP_LOGI("tts_response", "%s", x.c_str());
            return x.rfind("Sorry", 0) == 0;
        then:
          - logger.log: "Command failed!"
          - light.turn_on:
              id: output_led
              effect: hold
              brightness: 100%
              red: 1
              green: 0
              blue: 0
        else:
          - logger.log: "Command successful!"
          - light.turn_on:
              id: output_led
              effect: hold
              brightness: 100%
              red: 0
              green: 1
              blue: 0
```

As far as I can tell everything matches the current documentation and is using the current versions of everything, but yet I have the exact issue described in the original thread (as I did back then too) - the wake word never triggers, either with micro_wake_work or with a streaming wake word.

joshuaboniface, 2025-05-22:
And just like that, without actually changing anything, it suddenly started working. Though there's still a bit more wonkiness going on that I'm working on sorting out regarding how `start_continuous` seems to work now (or not) with `micro_wake_word`. I'll blame ESP gremlins and come back with my final findings for posterity tomorrow.

jesserockz, 2025-05-22:
IIRC, setting `use_wake_word: true` means you want to use the streaming wake word (openwakeword add-on etc) and not micro_wake_word.

See the latest [Home Assistant Voice PE configuration](https://github.com/esphome/home-assistant-voice-pe/blob/dev/home-assistant-voice.yaml#L1633) that has been updated today and uses all components that are available inside ESPHome for voice related.

joshuaboniface, 2025-05-22:
One thing I noticed through just about every test I've been doing, is that I see a log message from `micro_wake_word`, but it never seems to "finish". Sometimes I see what we see above, other times I get the complete model and VAD sections like so

```
[11:10:03][C][api:170]: API Server:
[11:10:03][C][api:171]:   Address: supersensor-2860f0:6053
[11:10:03][C][api:173]:   Using noise encryption: YES
[11:10:03][C][micro_wake_word:064]: microWakeWord:
[11:10:03][C][micro_wake_word:065]:   models:
[11:10:03][C][micro_wake_word:014]:     - Wake Word: Hey Jarvis
[11:10:03][C][micro_wake_word:015]:       Probability cutoff: 0.83
[11:10:03][C][micro_wake_word:016]:       Sliding window size: 5
[11:10:03][C][micro_wake_word:020]:     - VAD Model
[11:10:03][C][micro_wake_word:021]:       Probability cutoff: 0.50
[11:10:03][C][micro_wake_word:022]:       Sliding window size: 5
[11:10:55][I][safe_mode:041]: Boot seems successful; resetting boot loop counter
[11:10:55][D][esp32.preferences:114]: Saving 1 preferences to flash...
[11:10:55][V][esp32.preferences:126]: sync: key: 233825507, len: 4
[11:10:55][D][esp32.preferences:142]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
```

but regardless that's always the last thing I see from that component. I wonder if it's just never starting correctly?

joshuaboniface, 2025-05-24:
And I believe I found the answer. And it was indeed a configuration issue.

>channel (Optional, enum): The channel of the microphone. One of left, right, or stereo. If stereo, the output data will be twice as big, with each right sample followed by a left sample. **Defaults to right.**

My config was missing `channel:` entirely, and always has been. I could have sworn, a long time back, that the default was `left`. So I wired up my INMP441's to ground the L/R channel, putting them permanently on the `left` channel. But (now?) it defaults to `right`. Thus, there was no signal coming from the microphone that the components could process. A simple

```
channel: left
```

in my microphone config seems to have solved this, and I'm getting reliable voice control. I'm curious when this changed, and why such a change in defaults would be made?

jesserockz, 2025-05-24:
Looking at the file history of the microphone config, it has been `right` since I first created the entry in 2023.6

joshuaboniface, 2025-05-26:
Very strange! I wonder if ADF was doing some sort of magic to switch the channels (or what channel was the default). In any case thanks for the pointers here!