esphome/issues #6583: RP2040 with lvgl (and Waveshare bwr) not working

swede89, 2024-12-19:
### The problem

Hi. I successfully tested the new Waveshare 7.5 bwr driver with the demo screen. Also, lambda rendering works fine. 
However, when I add lvgl the RP2040 does not boot (no logs, no home assistant, no network ping). In contrast, removing lvgl restores the system.
Please help me to troubleshoot this issue. Is it probably a memory issue? I read about users that solved issues with lvgl by adding PSRAM flags.

Cheers
Simon

### Which version of ESPHome has the issue?

2024.12

### What type of installation are you using?

Docker

### What platform are you using?

RP2040

### Board

RPi Pico W

### Component causing the issue

lvgl

### YAML Config

```yaml
esphome:
  name: pico-epaper

rp2040:
  board: rpipicow
 
# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "somebase64key"
  services:
    - service: refresh_display
      then:
        - component.update: epaper

ota:
  - platform: esphome
    password: "somepassword"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot in case wifi connection fails
  ap:
    ssid: "Pico-Epaper Fallback Hotspot"
    password: "somepassword"

color:
  - id: COLOR_RED
    red: 100%
    green: 0%
    blue: 0%

spi:
   clk_pin: 10
   mosi_pin: 11

lvgl:

display:
  - platform: waveshare_epaper
    id: epaper
    cs_pin: 9
    dc_pin: 8
    busy_pin: 13
    reset_pin: 12
    model: 7.50in-bV3-bwr
    update_interval: never
    auto_clear_enabled: false
```

### Anything in the logs that might be useful for us?

```txt
no logs, no home assistant, no network ping
```

clydebarrow, 2024-12-21:
It's not going to work - the RP2040 does not have enough RAM. ATM LVGL only supports 16 bit colour format, which is going to need a buffer bigger than your available RAM.

swede89, 2024-12-21:
Hi clydebarrow. Thanks for the explanation. That’s sad. I guess the issue should be closed then.

closed 2024-12-21 by clydebarrow