esphome/issues #6584: Mqtt topic name is wrong when name_add_mac_suffix is used

Hadatko, 2024-12-20:
### The problem

Imagine i have two different devices. I set name_add_mac_suffix = true. Both connect to same mqtt infrastracture. For HA config messages they are using correct topic name <name>_<mac>. But for data they are using <name>. As the name for data is using same topic both devices are pushing data to the same topics.

### Which version of ESPHome has the issue?

tested several from 2024.6.0 to the 2024.12.1

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

2024.10.3

### What platform are you using?

ESP32

### Component causing the issue

mqtt

### YAML Config

```yaml
esphome:
  name: ${name}
  friendly_name: ${friendly_name}
  min_version: 2024.11.0
  name_add_mac_suffix: true
  project:
    name: cervenka.lettis_invertor
    version: "1.0.0"
mqtt:
  broker: ${mqtt_ip}
  username: ${mqtt_user}
  password: ${mqtt_pass}
  # discovery_unique_id_gene
```

Hadatko, 2024-12-20:
Suggestion in 
[image]
[image]

closed 2025-01-13 by jesserockz

suredesigntshirts, 2025-01-15:
More issues. Can't post in #7988 because it's now locked.

TLDR: Firmware .bin files should add mac suffix to file name.

I have 4 devices using similar config. I recently started to use substitutions for IP address (manual ip address) and I wanted to generate 4 firmware .bin files for each of my devices.

So I ran `esphome compile s3-ed0c34.yaml s3-ed061c.yaml c6-64c660.yaml c6-649f5c.yaml` to compile all these, with the idea to go out in the field with my laptop and upload them (can't OTA due to long deep sleeps)

Problem is I noticed, that they'll all compile to the same .bin firmware. So I assume I won't be able to upload these alll they'll take the last compiled .bin 

A fix would be to add the mac suffix to the compiled .bin firmwares.

Here's the line near the end of the compile log which leads me to believe this.

```
Creating esp32c6 image...
Successfully created esp32c6 image.
esp32_create_combined_bin([".pioenvs/esp32c6-thermocouple/firmware.bin"], [".pioenvs/esp32c6-thermocouple/firmware.elf"])
```

Here's my .yaml
```
esphome:
  name: esp32c6-thermocouple
  name_add_mac_suffix: true
```

Hadatko, 2025-01-15:
You cannot add mac into filename as MAC is known at runtime (as it is specific per device).
But path for binary is unique per yaml (at least for me), because in every yaml i have different name and friendly name

Hadatko, 2025-01-19:
Hi @suredesigntshirts as you wanted log messages with fixed topic, could you take a look on this PR ? https://github.com/esphome/esphome/pull/8109