esphome/issues #6586: Fan speed can not be 0

Potyike, 2024-12-21:
### The problem

When I set the fan speed to 0 (turn the fan off) esphome reports the last used speed value (instead of 0) to home assistant.  On the fan's 'more info' dialog the fan is off, on the dashboard the state is 'off' but the speed is the last used value (ie. 100%) 

[image]
[image]

### Which version of ESPHome has the issue?

2024.11.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.12.2

### What platform are you using?

ESP8266

### Board

nodemcuv2

### Component causing the issue

fan

### YAML Config

```yaml
fan:
  - platform: template
    id: supply_fan
    speed_count: 2
    name: Supply Fan
    restore_mode: RESTORE_DEFAULT_OFF
    on_speed_set:
      then:
        - if:
            condition:
              lambda: return (id(supply_fan).speed == 1) ;
            then:
             - output.turn_on: supply_low_high
             - output.turn_on: supply_on_off
        - if:
            condition:
              lambda: return (id(supply_fan).speed == 2);
            then:
              - output.turn_off: supply_low_high
              - output.turn_on: supply_on_off
        - if:
            condition:
              lambda: return (id(supply_fan).state == 0);
            then:
              - output.turn_off: supply_low_high
              - output.turn_off: supply_on_off
   
    on_turn_off:
      - output.turn_off: supply_low_high
      - output.turn_off: supply_on_off
```

### Additional information

#3115  #4285

clydebarrow, 2024-12-21:
Seems more likely an HA problem than ESPHome.

clydebarrow, 2024-12-21:
Actually if you read the ESPHome Fan docs, that's the expected behaviour - the speeds are 1-based, not zero-based.

> A fan can be switched on or off, optionally has a speed between 1 and the maximum supported speed of the fan,

closed 2024-12-21 by clydebarrow

Potyike, 2024-12-21:
I've read that but the problem is when the fan is off HA displays the current speed as 100%. Wouldn't be more logical if the fan is off, the speed is 0? 
When I power cycle my board HA correctly displays the speed as 'off'. 
I only want to display 'off' as speed when the fan is off like on the screenshot, but without the ability to setting the speed to 0 is not possible. 
[image]

clydebarrow, 2024-12-21:
I can see your point, but that's an HA issue. You could try using a Speed Fan component with a template output instead of the template fan. That should presumably set the speed to 0 when its off, but I haven't tested it.

Potyike, 2024-12-21:
I don't want to be awkward but are you sure that is a HA problem? As You can see in the logs ESPHome reports "state: off speed: 2" and I assume HA just displays those values. 
I tried Your suggestion with the template output before I opened this issue but it didn't work. 
Anyway, thanks for trying to help.

clydebarrow, 2024-12-22:
> I don't want to be awkward but are you sure that is a HA problem?

Well, it's a display issue, and it's HA that's doing the display, so yes. In ESPHome the speed is never 0. That's not a bug, that's by design. If you don't like that design, raise a feature request.

I'm have no opinion on whether the current design is good or not, just that it's operating as intended so nothing is broken.

Potyike, 2024-12-22:
Ok, I understand. 
Thanks