esphome/issues #6588: Compile of esp32 btproxy fails in HA/ESPHome Builder

meliora, 2024-12-21:
### The problem

When compiling a btproxy for ESP32 in RPI4 with 64-bit kernel and 32-bit userspace the compile fails for missing "tasmota/tool-mklittlefs" v3.2.0 for system 'linux_armv7l'.

### Which version of ESPHome has the issue?

2024.12.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.12.5

### What platform are you using?

ESP32-IDF

### Board

Nodemcu 32S

### YAML Config

```yaml
esphome:
  name: esp32-btproxy
  friendly_name: esp32-btproxy

esp32:
  board: esp32dev
  framework:
    type: esp-idf  

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "..."

ota:
  - platform: esphome
    password: "..."

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esp32-Btproxy Fallback Hotspot"
    password: "..."

esp32_ble_tracker:
  scan_parameters: 
    interval: 1100ms
    window: 1100ms
    active: True

bluetooth_proxy:
  active: True
      
captive_portal:
```

### Anything in the logs that might be useful for us?

```txt
INFO ESPHome 2024.12.2
INFO Reading configuration /config/esphome/esp32-btproxy.yaml...
INFO Generating C++ source...
INFO Updating https://github.com/espressif/esp-protocols.git@mdns-v1.3.2
INFO Compiling app...
Processing esp32-btproxy (board: esp32dev; framework: espidf; platform: https://github.com/pioarduino/platform-espressif32.git#51.03.07)
--------------------------------------------------------------------------------
Tool Manager: Installing tasmota/tool-mklittlefs @ ^3.2.0
INFO Installing tasmota/tool-mklittlefs @ ^3.2.0
UnknownPackageError: Could not find the package with 'tasmota/tool-mklittlefs @ ^3.2.0' requirements for your system 'linux_armv7l'
```

### Additional information

Running on Raspberry Pi 4, Debian 11.11 (bullseye), aarch64 kernel, armv7 userspace.

perseus177, 2024-12-21:
Same problem

bradweingartner, 2024-12-22:
Experiencing a similar issue on Raspberry Pi 2 running Home Assistant.

I'd like to note that the error occurs on currently running and previously successfully compiled code.

`Tool Manager: Installing tasmota/tool-mklittlefs @ ^3.2.0
INFO Installing tasmota/tool-mklittlefs @ ^3.2.0
UnknownPackageError: Could not find the package with 'tasmota/tool-mklittlefs @ ^3.2.0' requirements for your system 'linux_armv7l'`

sca075, 2024-12-22:
Same problem updating the S3 box 3 for voice assistant. Downgraded ESPHome it was possible to update the device.

ssieb, 2024-12-22:
I don't know what would be pulling that in since esphome doesn't use it and why tasmota.  But support for ARMv7 is getting dropped very soon if not already and I can't imagine trying to compile on a Pi2 anyway.

sca075, 2024-12-22:
```
NOTICE: Current idf-component-manager default lock file version is 2.0.0, but found 1.0.0 in /data/build/esp32-s3-box-3-05ad68/dependencies.lock. Recreating lock file with the current version.
Manifest files have changed, solving dependencies.
.....NOTICE: Updating lock file at /data/build/esp32-s3-box-3-05ad68/dependencies.lock
Processing 1 dependencies:
[1/1] idf (5.1.5)
-- Configuring incomplete, errors occurred!
See also "/data/build/esp32-s3-box-3-05ad68/.pioenvs/esp32-s3-box-3-05ad68/CMakeFiles/CMakeOutput.log".

fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
CMake Error at /data/cache/platformio/packages/framework-espidf/tools/cmake/build.cmake:266 (message):
  Failed to resolve component 'jsmn'.
Call Stack (most recent call first):
  /data/cache/platformio/packages/framework-espidf/tools/cmake/build.cmake:308 (__build_resolve_and_add_req)
  /data/cache/platformio/packages/framework-espidf/tools/cmake/build.cmake:603 (__build_expand_requirements)
  /data/cache/platformio/packages/framework-espidf/tools/cmake/project.cmake:604 (idf_build_process)
  CMakeLists.txt:3 (project)
```
Should I open another issue for this?

ssieb, 2024-12-22:
That's a completely different issue that has been filed a few times already.  You have to set an earlier idf version.  There's a pending fix to the online package if you're using that.

sca075, 2024-12-23:
@ssieb thanks I did had this issue with the esphome.voice-asssistant..  
```
substitutions:
  name: esp32-s3-box-3-05ad68
  friendly_name: ESP32 S3 Box 3 05ad68

packages:
  esphome.voice-assistant: github://esphome/firmware/wake-word-voice-assistant/esp32-s3-box-3.yaml@main
```
and it happens with the last two releases of  ESPHome builders in HA, it looks I couldn't update the device since last November because of this and I'm sorry but I don't get from what side the lock is raised as you guys are not writing I guess this firmware.

It isn't important for me but.. just to let you guys know about that.. really appreciate your work and clarifications.. please relax a little for the coming days ;) I'm sure it will be fixed later on.

ssieb, 2024-12-23:
@sca075 see https://github.com/esphome/issues/issues/6578

TheRingland, 2025-01-01:
This may be helpful - I had the same compile issue on an RPI4, however it would only fail under the HomeAssistant add-on container. If I deployed ESPHOME on the same PI in a new VENV, it compiled just fine. I think the container is linux_armv7l even though the host system is AARCH64

meliora, 2025-01-07:
> I don't know what would be pulling that in since esphome doesn't use it and why tasmota. But support for ARMv7 is getting dropped very soon if not already and I can't imagine trying to compile on a Pi2 anyway.

@ssieb Thanks for your reply. As you say support for ARMv7 is getting dropped, from where actually? I would guess there are quite many RPi installations (me included) still running in 32-bit userspace as the upgrade path from 32-bit system to 64-bit system is a bit cumbersome. But anyway, as you mention dropping the ARMv7 support does this mean that ESPHome addon in Homeassistant will be deprecated on such systems?

jeeck-jan, 2025-01-09:
in esphome  on raspbery 4

esp32:
  board: esp32dev
  framework:
    type: **esp-idf**

log :

INFO Installing tasmota/tool-mklittlefs @ ^3.2.0
UnknownPackageError: Could not find the package with 'tasmota/tool-mklittlefs @ ^3.2.0' requirements for your system 'linux_armv7l'

so same problem as meliora

if I change framework: **type: arduino** 
no problem anymore, but less functionalitys, uses to much memory for esp32
so would prefer solution

ssieb, 2025-01-09:
@meliora at some point soon, we won't build container images for armv7.  So yes, the addon will no longer update on those systems.  I can't imagine that it works very well there anyway.
@jeeck-jan there is no other solution at this point.  The new idf framework we're using apparently has a requirement on that package and that package doesn't have a build for armv7.  The solution is to upgrade your OS install to 64-bit.

jeeck-jan, 2025-01-09:
thanks ssieb,

seems wrong someware,  my OS is arm64, that is 64-bit

Static hostname: raspberrypi
       Icon name: computer
      Machine ID: xxxx
         Boot ID: xxxx
Operating System: Debian GNU/Linux 12 (bookworm)
          Kernel: Linux 6.6.62+rpt-rpi-v8
    Architecture: **arm64**

jeeck-jan, 2025-01-09:
But sure, my esp32 is 32bits.
So I fear end of the story for esp-idf for Esp32 ?

ssieb, 2025-01-09:
The kernel may be 64-bit, but what is the userspace?  Try `ldd /bin/bash`.

jeeck-jan, 2025-01-09:
hi ssieb,  i made a mistake.  My raspberry is 64bits, but the esp is 32bits.(on witch i trie to use esp-idf)
as you wrote : no esp-idf anymore on 32bits esp32 (armv7)

ssieb, 2025-01-09:
The ESP is not ARM and that's not relevant.  We're not going to drop ESP32 support.  I didn't write anything like what you said there.

jeeck-jan, 2025-01-09:
Sorry ssieb,

I get this message installing esp-idf on **esp32**:

UnknownPackageError: Could not find the package with 'tasmota/tool-mklittlefs @ ^3.2.0' requirements for your system 'linux_armv7l'

you wrote : The new idf framework we're using apparently has a requirement on that package and that package doesn't have a build for armv7.

my raspberry is **arm64** not armv7
and you wrote esp is **not ARM**

So can you explain about your message: _there is no other solution at this point. The new idf framework we're using apparently has a requirement on that package and that package doesn't have a build for armv7. The solution is to upgrade your OS install to 64-bit._
Upgrading my raspberry OS to 64-bit : is already arm64
Upgraging my esp to 64-bits ???

ssieb, 2025-01-09:
Did you not see my earlier reply?  The kernel might be 64-bit, but not userspace.
What does `ldd /bin/bash` give you?

jeeck-jan, 2025-01-09:
ldd /bin/bash
        linux-vdso.so.1 (0x0000007f9fe53000)
        libtinfo.so.6 => /lib/aarch64-linux-gnu/libtinfo.so.6 (0x0000007f9fc50000)
        libc.so.6 => /lib/aarch64-linux-gnu/libc.so.6 (0x0000007f9faa0000)
        /lib/ld-linux-aarch64.so.1 (0x0000007f9fe16000)

ssieb, 2025-01-09:
Ok, that looks like the host userspace is 64-bit.  Then the question is how did you end up with the armv7 addon container?  Try removing the addon and installing it again.

jeeck-jan, 2025-01-09:
Addon removed and installed again : 
esp-idf reinstalled 
problem remains:

Platform Manager: espressif32@51.3.6+sha.ec17486 has been installed!
INFO espressif32@51.3.6+sha.ec17486 has been installed!
Tool Manager: Installing https://github.com/pioarduino/esp-idf/releases/download/v5.1.5/esp-idf-v5.1.5.zip
INFO Installing https://github.com/pioarduino/esp-idf/releases/download/v5.1.5/esp-idf-v5.1.5.zip
Downloading  [####################################]  100%          
Unpacking  [####################################]  100%          
Tool Manager: framework-espidf@3.50105.0 has been installed!
INFO framework-espidf@3.50105.0 has been installed!
Tool Manager: Installing espressif/toolchain-xtensa-esp32 @ 12.2.0+20230208
INFO Installing espressif/toolchain-xtensa-esp32 @ 12.2.0+20230208
Downloading  [####################################]  100%          
Unpacking  [####################################]  100%          
Tool Manager: toolchain-xtensa-esp32@12.2.0+20230208 has been installed!
INFO toolchain-xtensa-esp32@12.2.0+20230208 has been installed!
Tool Manager: Installing espressif/toolchain-esp32ulp @ 2.35.0-20220830
INFO Installing espressif/toolchain-esp32ulp @ 2.35.0-20220830
Downloading  [####################################]  100%          
Unpacking  [####################################]  100%
Tool Manager: toolchain-esp32ulp@2.35.0-20220830 has been installed!
INFO toolchain-esp32ulp@2.35.0-20220830 has been installed!
Tool Manager: Installing espressif/tool-xtensa-esp-elf-gdb @ ~12.1.0
INFO Installing espressif/tool-xtensa-esp-elf-gdb @ ~12.1.0
Downloading  [####################################]  100%          
Unpacking  [####################################]  100%          
Tool Manager: tool-xtensa-esp-elf-gdb@12.1.0+20221002 has been installed!
INFO tool-xtensa-esp-elf-gdb@12.1.0+20221002 has been installed!
Tool Manager: Installing espressif/tool-riscv32-esp-elf-gdb @ ~12.1.0
INFO Installing espressif/tool-riscv32-esp-elf-gdb @ ~12.1.0
Downloading  [####################################]  100%          
Unpacking  [####################################]  100%          
Tool Manager: tool-riscv32-esp-elf-gdb@12.1.0+20221002 has been installed!
INFO tool-riscv32-esp-elf-gdb@12.1.0+20221002 has been installed!
Tool Manager: Installing https://github.com/pioarduino/esptool/releases/download/v4.8.1.1/esptool.zip
INFO Installing https://github.com/pioarduino/esptool/releases/download/v4.8.1.1/esptool.zip
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Tool Manager: tool-esptoolpy@4.8.1 has been installed!
INFO tool-esptoolpy@4.8.1 has been installed!
Tool Manager: Installing tasmota/tool-mklittlefs @ ^3.2.0
INFO Installing tasmota/tool-mklittlefs @ ^3.2.0
**UnknownPackageError: Could not find the package with 'tasmota/tool-mklittlefs @ ^3.2.0' requirements for your system 'linux_armv7l'**

hSATAC, 2025-01-15:
Same here. Would love to have a solution for this without migrating my whole HA to another machine.

jeeck-jan, 2025-01-29:
On the moment no sollution for this problem.

Found a workaround.
Installed esphome for windows and installed bin from there.
A guide to do this :  https://www.youtube.com/watch?v=BX6tDsux_X4

p1ngb4ck, 2025-01-29:
Hey, the solution is quite neat and simple.
You "should" have your .platformio - directory at hand - check the subdir platforms - you should find the espressif32 directory there.
Edit the platform.json - file in that one, it should have a package-subarray for tasmota/tool-mklittlefs. Delete those lines and replace them with the "older" variant :
<code>
"tool-mklittlefs": {
      "type": "uploader",
      "optional": true,
      "owner": "platformio",
      "version": "~1.203.0"
    },
</code>
If you want to (optional, as littlefs is not required actually) - you can compile mklittlefs version 4.02 yourself locally on the pi

git clone https://github.com/earlephilhower/mklittlefs.git
cd mklittlefs
git submodule update --init
make dist

you should have the mklittlefs binary in that directory then - do a chmod+x on it, copy it to /usr/local/bin or similar and replace the tool-mklittlefs/mklittlefs binary inside the .platformio/packages/tool-mklittlefs dir with a symbolic link to it.
Enjoy =)

bradkeifer, 2025-02-04:
I also have this problem, and my `ldd /bin/bsh` output is
```
/usr/lib/arm-linux-gnueabihf/libarmmem-${PLATFORM}.so => /usr/lib/arm-linux-gnueabihf/libarmmem-v8l.so (0xf78fe000)
libtinfo.so.6 => /lib/arm-linux-gnueabihf/libtinfo.so.6 (0xf78b6000)
libdl.so.2 => /lib/arm-linux-gnueabihf/libdl.so.2 (0xf78a2000)
libc.so.6 => /lib/arm-linux-gnueabihf/libc.so.6 (0xf774f000)
/lib/ld-linux-armhf.so.3 (0xf7913000)
```
Is anyone working on a proper fix for this that will be in a future release of the esphome docker container?
Or is it possible for me to see if someone is working on a fix for this?

ssieb, 2025-02-05:
@bradkeifer that's armv7.  The armv7 container is currently being removed.  You will need to upgrade your system to aarch64.

bradkeifer, 2025-02-05:
> [@bradkeifer](https://github.com/bradkeifer) that's armv7. The armv7 container is currently being removed. You will need to upgrade your system to aarch64.

Thank you @ssieb, but the output from `uname -m` on my machine (and when I enter a shell in the esphome docker container) is `aarch64`. I installed esphome using docker, with the command `docker pull ghcr.io/esphome/esphome:stable`.

ssieb, 2025-02-05:
That just says that the kernel is 64-bit.  That was a common configuration from a while back.  64-bit kernel and 32-bit userspace.  That `ldd` output shows that the userspace is 32-bit.

jeeck-jan, 2025-02-19:
new update esphome builder installed today,
problem solved on raspberry pi 4b
thanks

sca075, 2025-02-19:
yep but all of my (old code style) can't be updated anymore.. uffff.

jeeck-jan, 2025-02-19:
layout has changed,
new  layout something like this :

esphome:
  name: camera
  friendly_name: camera
esp8266:
  board: esp01_1m

jeeck-jan, 2025-02-19:
sorry, like this

esphome:
    name: camera
   friendly_name: camera
esp8266:
    board: esp01_1m

jeeck-jan, 2025-02-19:
cant get it written : ident for name, friendly_name and board: