esphome/issues #6589: (how) Does  one_wire work with the esp32_rmt.use_new_rmt_driver (remote_transmitter component)?

juanboro, 2024-12-21:
### The problem

Hi (@swoboda1337),
Is there documentation/an example/working case of how to use the one_wire with remote_transmitter with the new ESP-IDF platform for ESP32?  On the ESP8266, and ESP-32 with Arduino  and older RMT in older IDF, I have been able to use a single GPIO to transmit and receive on the TI-CC1101 using Radiolib (like [here](https://github.com/juanboro/esphome-radiolib-cc1101/blob/main/esp32-cc1101.yaml).  With the latest ESP-IDF platform release included with esphome -I can only get receive or transmit alone - not both.   With both enabled, and prioritizing receive component ahead of transmit, they can co-exist with receive working, but transmit will only transmit an idle carrier.  

### Which version of ESPHome has the issue?

2025.1.0-dev

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

NA

### What platform are you using?

ESP32-IDF

### Board

esp32dev

### Component causing the issue

remote_receiver

### YAML Config

```yaml
https://github.com/juanboro/esphome-radiolib-cc1101/blob/main/esp32-cc1101.yaml

I test with: https://github.com/juanboro/esphome-radiolib-cc1101/blob/main/esp32-cc1101-udp-example.yaml
```

swoboda1337, 2024-12-21:
Try removing this:

```
id(rf_transmitter).setup();
```

That hack was needed with the old driver. You also shouldn't need to adjust setup priority (my pr had that change).

juanboro, 2024-12-21:
I tried that (removing the component setup) earlier (I'll check again just to make sure) --- it didn't seem to impact it one way or the other last I checked.  I also removed setting tx/rx of the gpio pins (was necessary on all platforms for transmit before)  and also experimented with setting output during xmit to open collector - with no effect.

swoboda1337, 2024-12-21:
Can you try using a random gpio with nothing connected to it? If you transmit something you should see in on the receive side.

swoboda1337, 2024-12-21:
Dont change the gpio mode either let rmt control it

juanboro, 2024-12-21:
Ok --- good idea :-).  I tried [that](https://github.com/juanboro/esphome-radiolib-cc1101/blob/main/esp32-cc1101-udp-example-loopgpio.yaml)  --- and am seeing the transmit data loop back to the receive correctly as it should.  I'll try seeing if I can remove all the other references and use the real pin next.

Thanks!

juanboro, 2024-12-21:
Ok - I removed all the additional pin references.  
- With [both](https://github.com/juanboro/esphome-radiolib-cc1101/blob/main/esp32-cc1101.yaml) receive and transmit components, receive (from CC1101) does NOT work.  Transmit does work, and also does correctly loop back to receive.  But nothing else is received.
- With [just](https://github.com/juanboro/esphome-radiolib-cc1101/blob/main/esp32-cc1101-udp-example-noxmit.yaml) receive and no transmit component, receive does work.

swoboda1337, 2024-12-21:
Ok I will take a look tonight. I also have some cc1101s in the mail, should be here soon.

ssieb, 2024-12-22:
How does this involve `one_wire` and this is about an external component anyway, so this is the wrong place.

juanboro, 2024-12-22:
Hi - this is NOT about an external component.  It is also not about the one_wire component.  It about the one_wire option in the remote_receiver component.  In: https://github.com/esphome/esphome/blob/dev/esphome/components/remote_transmitter/__init__.py -- see CONF_ONE_WIRE

Can you please re-open this?  Thanks!

swoboda1337, 2024-12-22:
> How does this involve `one_wire` and this is about an external component anyway, so this is the wrong place.

There is a feature in rmt to allow both remote transmitter and remote receiver to use the same gpio. It is also called one_wire as it is one wire. It is very different from the one_wire component. We should reopen this as it should be working.

ssieb, 2024-12-22:
Ok, I just saw there was an external component involved and I hadn't heard of the `one_wire` option since it's only in dev.

swoboda1337, 2024-12-22:
I tried again on my sx1278 and I am able to to both rx and tx without any issues:

```
remote_receiver:
  id: rx_id
  pin:
    number: GPIO32
    allow_other_uses: true
  filter: 100us
  idle: 800us
  filter_symbols: 50
  receive_symbols: 256

remote_transmitter:
  id: tx_id
  pin:
    number: GPIO32
    allow_other_uses: true
  one_wire: true
  carrier_duty_percent: 100%
  on_transmit:
    then:
      - sx127x.set_mode_tx
  on_complete:
    then:
      - sx127x.set_mode_rx
```

The only thing I can think to try is setting io_loop_back to 1 here: https://github.com/esphome/esphome/blob/dev/esphome/components/remote_receiver/remote_receiver_esp32.cpp#L53

Not sure if this will help or not but it doesn't break my stuff. Give that a try?

I guess instead of radio lib you could also try this: https://github.com/esphome/esphome/pull/6300 I thought those guys had cc1101 working with one_wire: true

juanboro, 2024-12-22:
Yeah - I'll give the io_loop_back a try on the remote_receiver tonight --- and also review again to see if I'm missing something obvious (as far as I know - I have latest dev env on a couple of machines).   I was also going to try the standalone cc1101 component PR as well at some point (...I also have some sx127x that will arrive in the mail I was going to try yours with when I get them :-)).

swoboda1337, 2024-12-22:
Also noticed this should be removed from remote_receiver as we don't actually call setup anymore on the pin, try with this removed too:
```
  if (this->pin_->digital_read()) {
    ESP_LOGW(TAG, "Remote Receiver Signal starts with a HIGH value. Usually this means you have to "
                  "invert the signal using 'inverted: True' in the pin schema!");
  }
```
Ideally we just pass in the number and don't create a gpio object. Might be messy though as non esp32 builds need it.

juanboro, 2024-12-22:
Ok - I tried setting io_loop_back as well as removing the digital_read in remote_receiver --- still no luck.  I'm trying a couple more things as well tonight around the transmitter (I tried resetting the pin to open drain --- if I don't run out of time tomorrow I'll try to see if something going on with the scope on that pin).

swoboda1337, 2024-12-22:
One more thing, it doesnt break my device, remove the call to rmt_enable in remote_transmitter and add only enable it around the actual transmit:

```
#if ESP_IDF_VERSION_MAJOR >= 5
  esp_err_t error = rmt_enable(this->channel_);
  if (error != ESP_OK) {
    this->error_code_ = error;
    this->error_string_ = "in rmt_enable";
    this->mark_failed();
    return;
  }
  for (uint32_t i = 0; i < send_times; i++) {
    rmt_transmit_config_t config;
    memset(&config, 0, sizeof(config));
    config.loop_count = 0;
    config.flags.eot_level = this->inverted_;
    esp_err_t error = rmt_transmit(this->channel_, this->encoder_, this->rmt_temp_.data(),
                                   this->rmt_temp_.size() * sizeof(rmt_symbol_word_t), &config);
    if (error != ESP_OK) {
      ESP_LOGW(TAG, "rmt_transmit failed: %s", esp_err_to_name(error));
      this->status_set_warning();
    } else {
      this->status_clear_warning();
    }
    error = rmt_tx_wait_all_done(this->channel_, -1);
    if (error != ESP_OK) {
      ESP_LOGW(TAG, "rmt_tx_wait_all_done failed: %s", esp_err_to_name(error));
      this->status_set_warning();
    } else {
      this->status_clear_warning();
    }
    if (i + 1 < send_times)
      delayMicroseconds(send_wait);
  }
  error = rmt_disable(this->channel_);
  if (error != ESP_OK) {
    this->error_code_ = error;
    this->error_string_ = "in rmt_disable";
    this->mark_failed();
    return;
  }
#else
```

swoboda1337, 2024-12-22:
The cc1101 docs also have this: 
```
5. The interface can connect directly to the MCU which use 3.3V power supply, there is no need serial resistor. If the I/O is open-drain, the pull-up resistor will be added. When connect to the MCU which use 5V power supply, the MCU's I/O output current beyond 10 mA, a 2~5k ohm resistor will be series to divide the voltage, or it will damage the module.
```
Seems you need to configure the cc1101 to work in open drain mode or add a resistor yourself?

juanboro, 2024-12-22:
>Seems you need to configure the cc1101 to work in open drain mode or add a resistor yourself?
That's a definite possibility.    I did the changes around rmt_enable/disable above... the transmit still works well... but still no receive... I wonder if the issue is both boards are pulling up?  I'll dig more early tomorrow morning.  Thanks for all the help!

swoboda1337, 2024-12-22:
I thought about this more and I don't think a pull up is the problem as transmit works. One last thing to try, change io_loop_back to 1 and io_od_mode to 0 in remote transmitter:
```
    channel.flags.io_loop_back = this->one_wire_;
    channel.flags.io_od_mode = 0;
```
If it works I guess we should make two configs in remote transmitter or just not set OD mode. It works for me this way as well.

juanboro, 2024-12-22:
ok - will give that a try right now.  I've been reading through the cc1101 datasheet again this morning around async direct receive pin config settings.  Like you say - if anything - I'd expect reliably getting TX keying to it would be the issue and not with getting the RX signaling from it.

juanboro, 2024-12-22:
Ok - setting io_od_mode=0 didn't seem to change anything.  But I made another interesting observation -- I commented out all the pin references again in my cc1101 code --- and then added in just the yaml generating the rx pin:
```yaml
radiolib_cc1101:
  id: mycc1101
  cs_pin: 5 # CC1101 pin 4
  rx_pin:
    number: 32 # This is CC1101 GDO0 pin 3
    allow_other_uses: true
```

Which the code generation turns into GPIO set flags.  Doing JUST that breaks the transmit functionality for me (and leaves the RX not working).  Then I added this back into my standby method:
```c++
int RadiolibCC1101Component::standby() {
  // standby state: gd0 is input, radio in standby
  if ((_gd0_rx!=nullptr)&&(state==CC1101_XMIT)) _gd0_rx->setup();
```

When I do that --- after I hit the transmit button --- the receive begins to start working.  The transmit doesn't transmit anything, but after that I hit the transmit button,  RX starts working.  It seems like the ESP rmt might be very sensitive/unpredictable with anything messing with the GPIO modes after it is setup or still doing its own thing with the gpios.

I also created a tx_pin similar to the RX pin (difference is where gpio flag is output versus input) --- same effect with breaking transmit functionality.

juanboro, 2024-12-22:
I'm going to close this issue for now since it can't be reproduced outside my own boards currently (and I don't want to add any more noise to the long issue list backlog for esphome).  I'll work on collecting more details and data after the holidays and re-open if necessary on what I find out.   I think something odd is going on with GPIO interactions between multiple components - but right now it isn't 100% clear if it isn't something specifically unique to my setup.

juanboro, 2024-12-22:
Ok - one more set of observations... I added some logs to the remote_receiver and transmitter... in both rmt_new_rx_channel and rmt_new_tx_channel --- they are both setting pullup on the GPIO:
```
[17:41:14][I][esp-idf:000]: I (1079) gpio: GPIO[32]| InputEn: 1| OutputEn: 1| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
[17:41:14]
[17:41:14][I][remote_receiver.esp32:098]: RMT receive created
[17:41:14][I][remote_transmitter:047]: RMT tx channel begin init
[17:41:14][I][esp-idf:000]: I (1088) gpio: GPIO[32]| InputEn: 1| OutputEn: 1| OpenDrain: 1| Pullup: 1| Pulldown: 0| Intr:0
```

I believe this is why the receive is having the problem (the CC1101 driver likely not strong enough  pulldown to fight the pullup).  The interesting part is if I do _anything_ to the GPIOs including setting the exact state above with the pullup enabled, i.e:
```c+++
    this->pin_->pin_mode(gpio::Flags::FLAG_INPUT | gpio::Flags::FLAG_OUTPUT | gpio::Flags::FLAG_PULLUP | gpio::Flags::FLAG_OPEN_DRAIN);
```
 --- TX is then not working from that point forward.  But if I turn off the pullup on the GPIO --- the RX starts to work.

swoboda1337, 2024-12-22:
If you do rx only is there not a pull up as well?

juanboro, 2024-12-22:
Actually - yes --- and it gets more interesting (just finished compile now of rx only).  I still had ```channel.flags.io_loop_back = 1;``` in receive - so got this config:
```
[18:05:51][I][esp-idf:000]: I (1069) gpio: GPIO[32]| InputEn: 1| OutputEn: 1| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
[18:05:51]
[18:05:51][I][remote_receiver.esp32:098]: RMT receive created
```

and the RX wasn't working.  So then I turned off loop_back... and get:
```
[18:07:16][I][esp-idf:000]: I (1071) gpio: GPIO[32]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
[18:07:16]
[18:07:16][I][remote_receiver.esp32:096]: RMT receive created
```

yet - the RX _is_ apparently now working in that state.  So there goes my theory about the pullup :-(  (I think i was calling the RX pin setup to get RX working earlier - so that also would have turned off output and opendrain).

swoboda1337, 2024-12-22:
Or TX still drives the pin as well even when off. However not sure how that works as it would fight the input/receive side.

swoboda1337, 2024-12-23:
Ok @juanboro I think this has a good chance at working. In the transmitter try changing eot_level to 1:
```
    config.flags.eot_level = 1;
```
The reason is open drain has two mode. Its actively driven low or it high impedance and pulled high by the pullup. So by setting the eot level to 1 the transmitter isn't do anything to the pin. I am almost confident this work for you. 

Revert all other changes to the transmitter and receiver code.

swoboda1337, 2024-12-23:
Receive also wont work till you send the first transmit. I will have send a dummy transmit in setup to release the pin.

juanboro, 2024-12-23:
So force it to 1 [here](https://github.com/juanboro/esphome/blob/dev/esphome/components/remote_transmitter/remote_transmitter_esp32.cpp#L213)?   

It still seems like RX doesn't activate with that after a transmit either.
```c++
    config.flags.eot_level = this->inverted_;
    config.flags.eot_level = 1;
```

swoboda1337, 2024-12-23:
Yes set that to one. You removed all the other changes we tried?

juanboro, 2024-12-23:
Yes --- reverted everything back to dev.  Although i did notice i also reverted one_wire setting on my config side - so I put that back and it re-enabled the loopback working.  But still no incoming RX data.

swoboda1337, 2024-12-23:
> Yes --- reverted everything back to dev. Although i did notice i also reverted one_wire setting on my config side - so I put that back and it re-enabled the loopback working. But still no incoming RX data.
> 
> --- Wait a sec --- I think it is working!

It has to work, it makes sense why it was broken :)

Not sure why it took me so long to realize why though.

juanboro, 2024-12-23:
Awesome job --- thanks!  That definitely is working - there is no way I would ever have thought setting the end of transmit like that would work.  But thinking about it now -- it makes absolute sense for operating in single wire remote mode with TX and RX both enabled.  Thanks again!

swoboda1337, 2024-12-23:
Please reopen this I will get a PR up today. Thanks for finding this before it went out next release.

swoboda1337, 2024-12-23:
Not sure how this worked with my sx1278 it really shouldn't have.

swoboda1337, 2024-12-23:
Can you give this a try sometime: https://github.com/esphome/esphome/pull/8002

Make sure receive works without having to transmit first and still works after transmitting. Thanks

swoboda1337, 2024-12-23:
I think I may need to rethink the fix bit. It annoying to start tx mode with the pin high and the end with the pin high. If you transmit multiple times with a wait in between the pin will also be high. I think we need to add an option for eot_level and the ability to set the level. Could do something like:

```
remote_transmitter:
  id: tx_id
  pin:
    number: GPIO32
    allow_other_uses: true
  one_wire: true
  eot_level: false
  carrier_duty_percent: 100%
  on_transmit:
    then:
      - sx127x.set_mode_standby
      - remote_transmitter.set_level_low
      - sx127x.set_mode_tx
  on_complete:
    then:
      - sx127x.set_mode_standby
      - remote_transmitter.set_level_high
      - sx127x.set_mode_rx
```

This way the pin is always low during tx by default but is released before rx. Thoughts?

juanboro, 2024-12-23:
I can take a look in more detail this evening (have holiday driving to do).  Agree eot_level probably needs to be exposed to the yaml.  I'm going to be away from direct access to the hardware for awhile so I might not be able to test things well.  I did some very basic [testing](https://github.com/juanboro/esphome/tree/mymaster) of the PR this morning and it seems to work ok in that form - although I saw a panic on the first boot with rmt overruns (but that may just be how I'm bringing up that particular board receiving a flood of noise).,

swoboda1337, 2024-12-23:
I updated the PR, should be flexible now, basically what I described above.

closed 2024-12-24 by kbx81

juanboro, 2024-12-24:
> I updated the PR, should be flexible now, basically what I described above.
The updates look good.  I finally got a chance to test this morning and tx and rx look to be working fine.