esphome/issues #6590: ESP32-S2 Feather w/ MBE280 doesn’t work beyond 2024.8.1

p2-3d, 2024-12-21:
### The problem

I am using the Adafruit ESP32-S2 Feather with integrated MBE280 sensor to measure pressure and temperature.
Everything works fine with ESPHome 2024.8.1.

But when I tried to update to 2024.10.? I would get no data back. The device is still on-line and I can control the on board GPIO LED, but there is apparently no I2C data.

I restored the 2024.8.1 and tried again with 2024.11.3 and 2024.12.2.  The same problem, no sensor data.

### Which version of ESPHome has the issue?

Apparently anything past 2024.8.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

Any recent version

### What platform are you using?

ESP32-IDF

### Board

Adafruit ESP32-S2 Feather with integrated MBE280

### Component causing the issue

No I2C data

### YAML Config

```yaml
[quote="p2-3d, post:1, topic:807577"]

esphome:
  name: feather-bme280
  friendly_name: Feather_BME280

  platformio_options:
    board_build.variant: um_feathers2

esp32:
  board: featheresp32-s2
  variant: esp32s2
  framework:
    type: esp-idf
    version: recommended

# Enable logging

# Enable Home Assistant API

ota:

wifi:
 
captive_portal:
    
switch:
  - platform: gpio
    id: power_on
    restore_mode: ALWAYS_ON
    pin: 7
    setup_priority: 801
  - platform: gpio
    pin: 13
    name: "LED"

i2c:
  sda: GPIO3
  scl: GPIO4
  # scan: true
  id: stemma
  frequency: 200kHz
  setup_priority: 700

sensor:
  - platform: uptime
    update_interval: 10s
    id: up
  - platform: bme280_i2c
    i2c_id: stemma
    temperature:
      name: "BME280 Temperature"
      id: bme280_temperature
      oversampling: 16x
      filters:
        - offset: -6.2
        - sliding_window_moving_average:
            window_size: 12
            send_every: 6
    pressure:
      name: "BME280 Pressure"
      id: bme280_pressure
      oversampling: 16x
      unit_of_measurement: "hPa"
      accuracy_decimals: 1
      filters:
        - offset: 26.7
        - sliding_window_moving_average:
            window_size: 6
            send_every: 6
    humidity:
      name: "BME280 Humidity"
      id: bme280_humidity
      oversampling: 16x
      filters:
        - multiply: 1.19
        - offset: 6.0
        - sliding_window_moving_average:
            window_size: 6
            send_every: 6
    address: 0x77
    update_interval: 60s
```
[/quote]
```

### Anything in the logs that might be useful for us?

```txt
I do not see any error messages, the data just becomes unknown.
```

### Additional information

I have posted a question here:
https://community.home-assistant.io/t/esp32-s2-feather-w-mbe280-doesnt-work-beyond-2024-8-1/807577

sbrk-r, 2025-01-10:
I2C is broken completely on S2 after 2024.8.1, afaict. 
See #6445

p2-3d, 2025-03-20:
Just tried with version 2025.3.0.  Still no luck.

Tried to bump up the clock speed as suggested by https://github.com/esphome/issues/issues/6445#issuecomment-2558460250 but it didn't help.