esphome/issues #6593: esp32-c6-devkitm-1 board not recognized in ESPHome 2024.12.2

jaydeethree, 2024-12-22:
### The problem

I just started upgrading my ESPHome setup to the 2024.12.x versions, but builds are failing for my ESP32-C6 boards because their board type (esp32-c6-devkitm-1) is not recognized. I have confirmed that this board type exists in the platformio fork:

https://github.com/pioarduino/platform-espressif32/blob/main/boards/esp32-c6-devkitm-1.json

I also tried cleaning my build files but that did not help.

### Which version of ESPHome has the issue?

2024.12.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.12.5

### What platform are you using?

ESP32

### Board

esp32-c6-devkitm-1

### YAML Config

```yaml
esphome:
  name: test-01
  friendly_name: test-01

esp32:
  board: esp32-c6-devkitm-1
  variant: esp32c6
  flash_size: 4MB

# Enable logging
logger:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

captive_portal:

uart:
  rx_pin: GPIO4
  tx_pin: GPIO5
  baud_rate: 9600

sensor:
  - platform: senseair
    co2:
      name: "Carbon Dioxide"
    update_interval: 60s
    id: senseair_s8
```

### Anything in the logs that might be useful for us?

```txt
INFO ESPHome 2024.12.2
INFO Reading configuration /config/esphome/co2-sensor-02.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing test-01 (board: esp32-c6-devkitm-1; framework: arduino; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
UnknownBoard: Unknown board ID 'esp32-c6-devkitm-1'
```

jaydeethree, 2024-12-22:
Shortly after opening this issue, I noticed this in the logs:

>framework: arduino

That didn't look right so I added this to the `esp32` section of my YAML:

```
  framework:
    type: esp-idf
```

And that immediately fixed it. I'm not sure why it's defaulting to the arduino framework - is that a bug?

ssieb, 2024-12-22:
The default framework for ESP32 is still Arduino.

closed 2024-12-22 by ssieb

jaydeethree, 2024-12-22:
Strange, somehow I missed that. Thanks for the quick response, and sorry for the noise!

ncp1113, 2025-01-04:
I changed the framework to esp-idf and it worked using esphome on the command line. It worked! Thanks for the tip, jaydeethree.

attila-lendvai, 2025-03-19:
same here: when trying to compile my first project i arrived here from a websearch. i changed `arduino` to `esp-idf` and the error was gone.

```
$ esphome --version
Version: 2025.2.2
```

maybe retitle and reopen?

ssieb, 2025-03-20:
There's no issue.  The arduino framework used is older and doesn't have that board definition.