esphome/issues #6594: Display qspi_dbi together with LED spi_led_strip  => crash

chefbauer, 2024-12-22:
### The problem

Display qspi_dbi together with spi_led_strip  => crash, reproducable

The LEDs alone are ok, the Display alone too.
Together there is a crash when the Display initializes (LEDs flush before).
Maybe a SPI Problem?

It worked 2 or three months ago.
Sorry i can not say if it´s with the change qspi_amoled => qspi_dbi.

No debug, sorry

### Which version of ESPHome has the issue?

2024.12.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.12.2

### What platform are you using?

ESP32-IDF

### Board

LilyGo T-Display S3 AMOLED Touch

### Component causing the issue

spi_led_strip qspi_dbi

### YAML Config

```yaml
substitutions:
  name: esphome-web-877f7c
  friendly_name: TEST

esphome:
  platformio_options:
    build_flags: "-DBOARD_HAS_PSRAM"
    board_build.arduino.memory_type: qio_opi
    board_build.flash_mode: dio

  name: ${name}
  friendly_name: ${friendly_name}
  min_version: 2024.6.0
  name_add_mac_suffix: false

  project:
    name: esphome.web
    version: dev
  includes: 
    - includes/functions.h
    - includes/std_includes.h
   
esp32:
  board: esp32-s3-devkitc-1
  framework:
    type: esp-idf
  flash_size: 16MB
  partitions: "partitions/default_16MB.csv"
  

dashboard_import:
  package_import_url: github://esphome/firmware/esphome-web/esp32s3.yaml@main
  import_full_config: true

# Enable logging
logger:
  #level: VERY_VERBOSE

# Enable Home Assistant API
api:
  reboot_timeout: 0s 

# Allow Over-The-Air updates
ota:
- platform: esphome

# Allow provisioning Wi-Fi via serial
improv_serial:

wifi:
  # Set up a wifi access point
  ap: {}
  networks:
   - ssid: !secret wifi_ssid
     password: !secret wifi_password

# In combination with the `ap` this allows the user
# to provision wifi credentials to the device via WiFi AP.
captive_portal:

# Sets up Bluetooth LE (Only on ESP32) to allow the user
# to provision wifi credentials to the device.
#esp32_improv:
#  authorizer: none

# To have a "next url" for improv serial
web_server:
  version: 3

#TEST
spi:
  - #id: spi_led_ring
    type: single
    clk_pin: GPIO41
    mosi_pin: GPIO40
    interface: any
  - id: quad_spi
    type: quad
    clk_pin: 47
    data_pins: [18,7,48,5]

i2c:
  sda: GPIO3
  scl: GPIO2
  scan: True
  frequency: 400kHz

light:
  - platform: spi_led_strip
    id: led_ring_front
    name: "LED Ring Front"
    num_leds: 20 #20
    data_rate: 1MHz

display:
  - platform: qspi_dbi
    model: RM67162
    data_rate: 80MHz
    spi_mode: mode0
    id: main_lcd
    dimensions:
      height: 240
      width: 536
    transform:
      mirror_x: true
      swap_xy: true
    color_order: rgb
    brightness: 180
    cs_pin: 6
    reset_pin: 17
    enable_pin: 38
```

clydebarrow, 2024-12-22:
It's running out of RAM - you need to enable PSRAM:

```
psram:
  mode: octal
  speed: 80MHz
```

chefbauer, 2024-12-22:
i forgot in my examble, you are right, sorry.
the example is running now.

but my big system (with psram config) with about 3000 lines (lvgl and so on parted in many files) - 
as soon as i add spi_led_strip => crash.

i will look into free-memory now.
thank you for examining.

chefbauer, 2024-12-26:
This is the Debug Output of the Crash:
`[0;37m[V][spi_led_strip:066]: write_state: buf = 00 00 00 00 FF FF FF FF FF FF FF FF [0Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x42026af1  PS      : 0x00060e30  A0      : 0x82015a0e  A1      : 0x3fcaad70  
A2      : 0x3fcac738  A3      : 0x0000000c  A4      : 0x0000000c  A5      : 0x3fcac5ac  
A6      : 0x3f800000  A7      : 0x3f800000  A8      : 0x82026aee  A9      : 0x3fcaad10  
A10     : 0x00000000  A11     : 0x3c0f934f  A12     : 0x00000042  A13     : 0x3c0f9455  
A14     : 0x3fcaad70  A15     : 0x00000003  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x40056f08  LEND    : 0x40056f12  LCOUNT  : 0x00000000  

Backtrace: 0x42026aee:0x3fcaad70 0x42015a0b:0x3fcaadd0 0x42015a16:0x3fcaae20 0x420d8a45:0x3fcaae40 0x420d8b0f:0x3fcaae60 0x420245b7:0x3fcaae80 0x42030d73:0x3fcaaeb0 0x4200dfef:0x3fcab230
`

chefbauer, 2024-12-26:
i found the problem! and it is a bug.

This works:
```
light:
  - platform: spi_led_strip
    
display:
  - platform: qspi_dbi
```

This NOT:
```
display:
  - platform: qspi_dbi
  
  light:
  - platform: spi_led_strip
```

clydebarrow, 2024-12-27:
> there is a small bug i think

Yes, it's a priority issue between the light state and the spi_led component. PR raised to fix.

closed 2025-01-12 by jesserockz